espressif/esp-idf #15390: ble_spp_server例程运行错误

EwaneGao, 2025-02-13:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4.0

### Espressif SoC revision.

ESP32-S3-N16R8

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

成功编译运行ble_spp_server

### What is the actual behavior?

成功编译，但在运行时出现如下错误

[image]

### Steps to reproduce.

1. 打开例程
2. 编译例程
3. 下载
4. 打开串口监控

### Debug Logs.

```plain
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2810,len:0x15a0
load:0x403c8700,len:0x4
load:0x403c8704,len:0xd20
load:0x403cb700,len:0x2ee4
entry 0x403c8928
I (27) boot: ESP-IDF v5.4 2nd stage bootloader
I (27) boot: compile time Feb 13 2025 15:46:34
I (27) boot: Multicore bootloader
I (27) boot: chip revision: v0.2
I (30) boot: efuse block revision: v1.3
I (33) boot.esp32s3: Boot SPI Speed : 80MHz
I (37) boot.esp32s3: SPI Mode       : DIO
I (41) boot.esp32s3: SPI Flash Size : 16MB
I (45) boot: Enabling RNG early entropy source...
I (49) boot: Partition Table:
I (52) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (65) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (71) boot:  2 factory          factory app      00 00 00010000 00100000
I (78) boot: End of partition table
I (81) esp_image: segment 0: paddr=00010020 vaddr=3c090020 size=21620h (136736) map
I (112) esp_image: segment 1: paddr=00031648 vaddr=3fc99900 size=040cch ( 16588) load
I (116) esp_image: segment 2: paddr=0003571c vaddr=40374000 size=0a8fch ( 43260) load
I (126) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=81cc0h (531648) map
I (220) esp_image: segment 4: paddr=000c1ce8 vaddr=4037e8fc size=0af30h ( 44848) load
I (230) esp_image: segment 5: paddr=000ccc20 vaddr=600fe100 size=0001ch (    28) load
I (238) boot: Loaded app from partition at offset 0x10000
I (239) boot: Disabling RNG early entropy source...
I (249) cpu_start: Multicore app
I (258) cpu_start: Pro cpu start user code
I (258) cpu_start: cpu freq: 160000000 Hz
I (258) app_init: Application information:
I (258) app_init: Project name:     ble_spp_server_demo
I (263) app_init: App version:      1
I (266) app_init: Compile time:     Feb 13 2025 15:45:53
I (271) app_init: ELF file SHA256:  c067abd8c...
I (276) app_init: ESP-IDF:          v5.4
I (279) efuse_init: Min chip rev:     v0.0
I (283) efuse_init: Max chip rev:     v0.99 
I (287) efuse_init: Chip rev:         v0.2
I (291) heap_init: Initializing. RAM available for dynamic allocation:
I (297) heap_init: At 3FCA1AF8 len 00047C18 (287 KiB): RAM
I (302) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (307) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (313) heap_init: At 600FE11C len 00001ECC (7 KiB): RTCRAM
I (319) spi_flash: detected chip: boya
I (321) spi_flash: flash io: dio
I (325) sleep_gpio: Configure to isolate all GPIO pins in sleep state
I (331) sleep_gpio: Enable automatic switching of GPIO sleep configuration
I (337) coexist: coex firmware version: 49a8cdc
I (368) coexist: coexist rom version e7ae62f
I (369) main_task: Started on CPU0
I (379) main_task: Calling app_main()
I (389) BLE_INIT: BT controller compile version [d752dea]
I (389) BLE_INIT: Bluetooth MAC: 94:a9:90:1a:21:86
I (389) phy_init: phy_version 680,a6008b2,Jun  4 2024,16:41:10
W (429) phy_init: saving new calibration data because of checksum failure, mode(0)
I (489) GATTS_SPP_DEMO: app_main init bluetooth
I (509) GATTS_SPP_DEMO: EVT 0, gatts if 3
I (509) GATTS_SPP_DEMO: event = 0
I (509) GATTS_SPP_DEMO: gatts_profile_event_handler 502
I (509) GATTS_SPP_DEMO: gatts_profile_event_handler 505
I (519) GATTS_SPP_DEMO: gatts_profile_event_handler 508
I (529) GATTS_SPP_DEMO: EVT 22, gatts if 3
I (529) GATTS_SPP_DEMO: event = 16
I (529) GATTS_SPP_DEMO: The number handle =b
I (539) GATTS_SPP_DEMO: GAP_EVT, event 4
I (539) GATTS_SPP_DEMO: EVT 12, ga��I (549) main_task: Returned from app_main()
```

EwaneGao, 2025-02-14:
为什么每次重新下载/复位均会在此处出外问题。
每次都会打印出同样的内容
`I (469) GATTS_SPP_DEMO: The number handle =b
I (469) GATTS_SPP_DEMO: GAP_EVT, event 4
I (479) GATTS_SPP_DEMO: EVT 12, ga��I (489) main_task: Returned from app_main()`

esp-zhp, 2025-02-14:
您的情况听起来有些异常，与我本地的测试结果不一致。建议您尝试删除以下日志代码，看看是否会影响问题的复现：  

```c
ESP_LOGI(GATTS_TABLE_TAG, "EVT %d, gatts if %d", event, gatts_if);
```

我本地的测试日志如下，运行情况正常：  

```
（省略冗长日志，仅保留关键信息）
I (828) GATTS_SPP_DEMO: app_main init bluetooth
I (908) GATTS_SPP_DEMO: EVT 0, gatts if 3
I (918) GATTS_SPP_DEMO: EVT 22, gatts if 3
I (928) GATTS_SPP_DEMO: EVT 12, gatts if 3
I (948) main_task: Returned from app_main()
```

EwaneGao, 2025-02-14:
将代码639行
`     //ESP_LOGI(GATTS_TABLE_TAG, "EVT %d, gatts if %d", event, gatts_if);`
注释后，问题依旧
`I (389) phy_init: phy_version 680,a6008b2,Jun  4 2024,16:41:10
I (429) GATTS_SPP_DEMO: app_main init bluetooth
I (449) GATTS_SPP_DEMO: event = 0
I (449) GATTS_SPP_DEMO: gatts_profile_event_handler 502
I (449) GATTS_SPP_DEMO: gatts_profile_event_handler 505
I (449) GATTS_SPP_DEMO: gatts_profile_event_handler 508
I (459) GATTS_SPP_DEMO: event = 16
I (459) GATTS_SPP_DEMO: The number handle =b
I (469) GATTS_SPP_DEMO: GAP_EVT,��I (479) main_task: Returned from app_main()`

esp-zhp, 2025-02-14:
你修改代码了吗？
怎么从 I (539) GATTS_SPP_DEMO: EVT 12, ga��I (549) main_task: Returned from app_main() 
变成
(469) GATTS_SPP_DEMO: GAP_EVT,��I (479) main_task: Returned from app_main()

看起来不是一个位置
你能找到在哪里打印的这个 LOG 吗？

esp-zhp, 2025-02-20:
6c10a7f60d5241dedbbda1a89d77f6726d829d46

esp-zhp, 2025-03-10:
Master Branch: b4e3c5c1e260a05d072a2a9eb0f1c53e478b8bab
v5.4 Branch: 28a78ba1fb20ce875730afec9116f1395d519ebc
v5.3 Branch: dda01d886f2430f6e217c309ca06d5030424ac86
v5.2 Branch: 0fcca4ec3ce99866110a91140baa3a39d35091ce
v5.1 Branch: e71af692b96154f403b827bc4f3787b765b8e6b6
v5.0 Branch: 770f713e655969c0412a113481b994d4e0f731d2

Alvin1Zhang, 2025-03-27:
@EwaneGao 您好，可以按照 https://github.com/espressif/esp-idf/issues/15390#issuecomment-2658481818 建议提供更多反馈么？谢谢

closed 2025-04-17 by Alvin1Zhang