espressif/esp-idf #15392: After ESP32C3 started the wifi manager, when I connected to the device's wifi, it did not automatically redirect to the authentication page

q127981, 2025-02-13:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Open the wifi manager of ESP32C3 and connect to the default AP of the device using my phone. The wifi authentication webpage (192.168.4.1) will automatically pop up, allowing me to configure the wifi network.
After configuring the network for the device, when I opened the WiFi Manager again and connected to the device AP, my phone no longer automatically redirected to the WiFi authentication page (192.168.4.1).
That is to say, it will only automatically redirect to the authentication page when there is no wifi connection information.

uqfus, 2025-02-23:
Android smartphones do not popup captive portal page if AP assigns them IP in private range. Reconfigure to any other. Sample below.
// Reconfigure DHCPd. With IP 192.168.*.* 10.*.*.* 172.16-31.*.* captive portal page do not popup automatically.
https://github.com/uqfus/esp32-wifi-provision-care/blob/main/esp32-wifi-provision-care.c

hrushikesh430, 2025-02-26:
Hi @q127981, could you please send me your minimal application and details like which phone are you using, sdkconfig and IDF version? so that I can reproduce you issue easily.

uqfus, 2025-02-26:
I am using Samsung S10 smartphone. And it seems problem in smartphone not in code. Later will try to add code "ESP_ENABLE_DHCP_CAPTIVEPORTAL Enables more modern DHCP-based Option 114 to provide clients with the captive portal URI" from examples. Now i am not using it.

hrushikesh430, 2025-02-26:
Hi @uqfus, seems that your problem is solved, feel free to reopen the issue, if any issue arrives. Thank you.

hrushikesh430, 2025-03-10:
Hi @q127981, closing this issue due to response delay. Please feel free to reopen the issue for further queries.

closed 2025-03-10 by hrushikesh430