espressif/esp-idf #15393: Scan devices with advertise in BLE Phy Coded S8

yanmartins, 2025-02-13:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello,

I am using ESP32-S3 with ESP-IDF 5.3.1 to scan for BLE devices. One of these BLE devices is configured to warn in Coded S8 phy mode.

I am not able to find it in the ESP scan.

I set this on my init code: `esp_ble_gap_set_preferred_default_phy(ESP_BLE_GAP_PHY_CODED_PREF_MASK, ESP_BLE_GAP_PHY_CODED_PREF_MASK);`

Any suggestions?

Thanks!

yanmartins, 2025-02-21:
this is my init:

```c
#define SCAN_INTERVAL                 0X50
#define SCAN_WINDOW                   0X50
esp_ble_scan_params_t             ble_scan_params = {.scan_type          = BLE_SCAN_TYPE_ACTIVE,
                                                                       .own_addr_type      = BLE_ADDR_TYPE_PUBLIC,
                                                                       .scan_filter_policy = BLE_SCAN_FILTER_ALLOW_ALL,
                                                                       .scan_interval      = SCAN_INTERVAL,
                                                                       .scan_window        = SCAN_WINDOW,
                                                                       .scan_duplicate     = BLE_SCAN_DUPLICATE_DISABLE};

void srv_ble_gap_init(void)
{
    esp_ble_gap_set_preferred_default_phy(ESP_BLE_GAP_PHY_CODED_PREF_MASK, ESP_BLE_GAP_PHY_CODED_PREF_MASK);
    esp_ble_gap_set_scan_params(&ble_scan_params);

    static esp_ble_ext_scan_params_t ext_scan_params = {
        .own_addr_type = BLE_ADDR_TYPE_PUBLIC,
        .filter_policy = BLE_SCAN_FILTER_ALLOW_ALL,
        .scan_duplicate = BLE_SCAN_DUPLICATE_DISABLE,
        .cfg_mask = ESP_BLE_GAP_EXT_SCAN_CFG_UNCODE_MASK | ESP_BLE_GAP_EXT_SCAN_CFG_CODE_MASK,
        .uncoded_cfg = {BLE_SCAN_TYPE_ACTIVE, 40, 40},
        .coded_cfg = {BLE_SCAN_TYPE_ACTIVE, 80, 80},
    };

    esp_ble_gap_set_ext_scan_params(&ext_scan_params);
    esp_ble_gap_register_callback(esp_gap_cb);
}
```

Any suggestions?

yanmartins, 2025-05-10:
Hello, @esp-zhp. Do you have any suggention about this?

esp-zhp, 2025-05-13:
Please avoid mixing BLE 4.2 and BLE 5.0 scanning APIs, as this may lead to unexpected behavior.

```c
#define SCAN_INTERVAL_CODED     0x50
#define SCAN_WINDOW_CODED       0x50
#define SCAN_INTERVAL_UNCODED   0x28
#define SCAN_WINDOW_UNCODED     0x28

void srv_ble_gap_init(void)
{
    // Set preferred PHY to CODED (can be adjusted based on your requirements)
    esp_ble_gap_set_preferred_default_phy(ESP_BLE_GAP_PHY_CODED_PREF_MASK, ESP_BLE_GAP_PHY_CODED_PREF_MASK);

    // BLE 5.0 extended scan parameters
    static esp_ble_ext_scan_params_t ext_scan_params = {
        .own_addr_type   = BLE_ADDR_TYPE_PUBLIC,
        .filter_policy   = BLE_SCAN_FILTER_ALLOW_ALL,
        .scan_duplicate  = BLE_SCAN_DUPLICATE_DISABLE,
        .cfg_mask        = ESP_BLE_GAP_EXT_SCAN_CFG_UNCODE_MASK | ESP_BLE_GAP_EXT_SCAN_CFG_CODE_MASK,
        .uncoded_cfg     = {
            .scan_type     = BLE_SCAN_TYPE_ACTIVE,
            .scan_interval = SCAN_INTERVAL_UNCODED,
            .scan_window   = SCAN_WINDOW_UNCODED,
        },
        .coded_cfg       = {
            .scan_type     = BLE_SCAN_TYPE_ACTIVE,
            .scan_interval = SCAN_INTERVAL_CODED,
            .scan_window   = SCAN_WINDOW_CODED,
        },
    };

    esp_ble_gap_set_ext_scan_params(&ext_scan_params);
    esp_ble_gap_register_callback(esp_gap_cb);
}
```

### Notes for you:

* `esp_ble_gap_set_scan_params()` is **only for BLE 4.2** and should not be used in BLE 5.0 applications.
* `esp_ble_gap_set_ext_scan_params()` is designed for **extended scanning introduced in BLE 5.0**.
* Make sure your target device and configuration support BLE 5.0 features.