espressif/esp-idf #15397: OTBR Linux with esp32h2 as ot_rcp

donaldkohcw, 2025-02-14:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi,

I previously managed to run the physical hardware (ESP32-S3 and ESP32-H2) on the ESP Thread Border Router. However, I am now encountering an issue getting OTBR in Linux to communicate with ot_rcp, as shown in the following running log.

It seems that the serial connection is not being established correctly at the moment, and I am continuing my troubleshooting.

Are there any features I need to enable or disable in the ot_rcp build that I may have missed? The USB is connected to J2 (UART) of the ESP32-H2-DevKitM-1.

Thank you.

donaldkoh@donaldkoh-VMware-Virtual-Platform:~$ journalctl -u otbr-agent --no-pager | tail -n 30
Feb 14 16:34:47 donaldkoh-VMware-Virtual-Platform otbr-agent[25009]: [NOTE]-AGENT---: Backbone interface: eth0
Feb 14 16:34:47 donaldkoh-VMware-Virtual-Platform otbr-agent[25009]: [NOTE]-AGENT---: Running 0.3.0-thread-reference-20230710-542-g3f28d4c990c
Feb 14 16:34:47 donaldkoh-VMware-Virtual-Platform otbr-agent[25009]: [NOTE]-AGENT---: Thread version: 1.4.0
Feb 14 16:34:47 donaldkoh-VMware-Virtual-Platform otbr-agent[25009]: [NOTE]-AGENT---: Thread interface: wpan0
Feb 14 16:34:47 donaldkoh-VMware-Virtual-Platform otbr-agent[25009]: [NOTE]-AGENT---: Radio URL: spinel+hdlc+uart:///dev/ttyUSB0?baudrate=460800
Feb 14 16:34:47 donaldkoh-VMware-Virtual-Platform otbr-agent[25009]: [NOTE]-AGENT---: Radio URL: trel://eth0
Feb 14 16:34:47 donaldkoh-VMware-Virtual-Platform otbr-agent[25009]: [NOTE]-ILS-----: Infra link selected: eth0
Feb 14 16:34:47 donaldkoh-VMware-Virtual-Platform otbr-agent[25009]: [INFO]-RCP_HOS-: OpenThread log level changed to 4
Feb 14 16:34:49 donaldkoh-VMware-Virtual-Platform otbr-agent[25009]: 49d.19:22:55.292 [W] P-SpinelDrive-: Wait for response timeout
Feb 14 16:34:49 donaldkoh-VMware-Virtual-Platform otbr-agent[25009]: 49d.19:22:55.292 [I] P-SpinelDrive-: co-processor self reset successfully
Feb 14 16:34:51 donaldkoh-VMware-Virtual-Platform otbr-agent[25009]: 49d.19:22:57.294 [W] P-SpinelDrive-: Wait for response timeout
Feb 14 16:34:53 donaldkoh-VMware-Virtual-Platform otbr-agent[25009]: 49d.19:22:59.296 [W] P-SpinelDrive-: Wait for response timeout
Feb 14 16:34:53 donaldkoh-VMware-Virtual-Platform otbr-agent[25009]: 49d.19:22:59.296 [C] Platform------: Init() at spinel_driver.cpp:83: Failure
Feb 14 16:34:55 donaldkoh-VMware-Virtual-Platform otbr-agent[25009]: 49d.19:23:01.298 [W] P-SpinelDrive-: Wait for response timeout
Feb 14 16:34:55 donaldkoh-VMware-Virtual-Platform systemd[1]: otbr-agent.service: Main process exited, code=exited, status=1/FAILURE
Feb 14 16:34:55 donaldkoh-VMware-Virtual-Platform systemd[1]: otbr-agent.service: Failed with result 'exit-code'.
Feb 14 16:35:01 donaldkoh-VMware-Virtual-Platform systemd[1]: otbr-agent.service: Scheduled restart job, restart counter is at 754.
Feb 14 16:35:01 donaldkoh-VMware-Virtual-Platform systemd[1]: Starting otbr-agent.service - OpenThread Border Router Agent...

zwx1995esp, 2025-02-17:
Hi, @donaldkohcw Sorry for the delay. Could you please share the code base you used? Which version of IDF did you used? Did you modify any configuations for H2 ot_rcp?

donaldkohcw, 2025-02-18:
Hi @zwx1995esp, 

Thanks for following up. Please advise.

On the hardware side, the FTDI TTL is connected to the ESP32H2’s TX (yellow FTDI) and RX (orange FTDI) pins. Could you also explain the function of “[] Configure RCP UART pin manually”? How is the pin number assignment calculated? Do I need to configure this since I am using Ubuntu as the host

(4) The number of RX pin (NEW)

(5) The number of TX pin (NEW)

__________________________________________________________________
Code 'origin/release/v5.4' is running on 'esp-idf-v5.3.2' for ot_rcp.
____________________________________________________________________
In Ubuntu 24.04.1 running with VM:

ls /dev/ttyUSB* ->/dev/ttyUSB0
picocom -b 460800 /dev/ttyUSB0
"sudo otbr-agent -d 7 -I wpan0 -B eth0 spinel+hdlc+uart:///dev/ttyUSB0?baud=460800 trel://eth0"

____________________________________________________________________otbr-agent is having issues communicating with its radio co-processor via the Spinel interface over UART

journalctl -u otbr-agent --no-pager --lines=20
Feb 18 09:12:46 donaldkoh-VMware-Virtual-Platform otbr-agent[4226]: 49d.17:17:26.459 [W] P-SpinelDrive-: Wait for response timeout
Feb 18 09:13:16 donaldkoh-VMware-Virtual-Platform systemd[1]: otbr-agent.service: Main process exited, code=exited, status=1/FAILURE
Feb 18 09:13:16 donaldkoh-VMware-Virtual-Platform systemd[1]: otbr-agent.service: Failed with result 'exit-code'.
Feb 18 09:13:21 donaldkoh-VMware-Virtual-Platform systemd[1]: otbr-agent.service: Scheduled restart job, restart counter is at 50.
Feb 18 09:13:21 donaldkoh-VMware-Virtual-Platform systemd[1]: Starting otbr-agent.service - OpenThread Border Router Agent...
Feb 18 09:13:22 donaldkoh-VMware-Virtual-Platform systemd[1]: Started otbr-agent.service - OpenThread Border Router Agent.
Feb 18 09:13:22 donaldkoh-VMware-Virtual-Platform otbr-agent[4545]: [NOTE]-AGENT---: Backbone interface: eth0
Feb 18 09:13:22 donaldkoh-VMware-Virtual-Platform otbr-agent[4545]: [NOTE]-AGENT---: Running 0.3.0-thread-reference-20230710-542-g3f28d4c990c
Feb 18 09:13:22 donaldkoh-VMware-Virtual-Platform otbr-agent[4545]: [NOTE]-AGENT---: Thread version: 1.4.0
Feb 18 09:13:22 donaldkoh-VMware-Virtual-Platform otbr-agent[4545]: [NOTE]-AGENT---: Thread interface: wpan0
Feb 18 09:13:22 donaldkoh-VMware-Virtual-Platform otbr-agent[4545]: [NOTE]-AGENT---: Radio URL: spinel+hdlc+uart:///dev/ttyUSB0?baud=460800&uart-flow-control=1
Feb 18 09:13:22 donaldkoh-VMware-Virtual-Platform otbr-agent[4545]: [NOTE]-AGENT---: Radio URL: trel://eth0
Feb 18 09:13:22 donaldkoh-VMware-Virtual-Platform otbr-agent[4545]: [NOTE]-ILS-----: Infra link selected: eth0
Feb 18 09:13:22 donaldkoh-VMware-Virtual-Platform otbr-agent[4545]: [INFO]-RCP_HOS-: OpenThread log level changed to 4
Feb 18 09:13:25 donaldkoh-VMware-Virtual-Platform otbr-agent[4545]: 49d.17:18:05.521 [W] P-SpinelDrive-: Wait for response timeout
Feb 18 09:13:25 donaldkoh-VMware-Virtual-Platform otbr-agent[4545]: 49d.17:18:05.521 [I] P-SpinelDrive-: co-processor self reset successfully
Feb 18 09:13:27 donaldkoh-VMware-Virtual-Platform otbr-agent[4545]: 49d.17:18:07.523 [W] P-SpinelDrive-: Wait for response timeout
Feb 18 09:13:29 donaldkoh-VMware-Virtual-Platform otbr-agent[4545]: 49d.17:18:09.525 [W] P-SpinelDrive-: Wait for response timeout
Feb 18 09:13:29 donaldkoh-VMware-Virtual-Platform otbr-agent[4545]: 49d.17:18:09.525 [C] Platform------: Init() at spinel_driver.cpp:83: Failure
Feb 18 09:13:31 donaldkoh-VMware-Virtual-Platform otbr-agent[4545]: 49d.17:18:11.527 [W] P-SpinelDrive-: Wait for response timeout

Regards,
Donald Koh

zwx1995esp, 2025-02-18:
If you use the H2 UART port to connect to the Host, please do not set OPENTHREAD_UART_PIN_MANUAL. This configuration is meant for specifying UART Tx/Rx pins when manually connecting the Host to the RCP. However, on the H2 devkit board, the hardware is already wired to the default Tx/Rx pins. Please refer to [this code](https://github.com/espressif/esp-idf/blob/v5.3.2/examples/openthread/ot_rcp/main/esp_ot_config.h#L28-L34)—using the default pins should work.

dk77148, 2025-03-04:
Hi @zwx1995esp, thank you for your guidance and clarification. I was able to get this running. Thanks!

closed 2025-03-14 by Alvin1Zhang