espressif/esp-idf #15398: NimBLE Stack smashing detected

florianhumblot, 2025-02-14:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.4-674-gd38afc77db

### Espressif SoC revision.

ESP32-C3 (QFN32) (revision v0.4)

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

When connecting to a NimBLE application on the ESP32C3 I expect to be able to connect without a crash.

### What is the actual behavior?

The device crashes with a instruction access fault, unless I enable stack smashing detection, which then shows up in nimble

### Steps to reproduce.

1. Set up NimBLE 
```c++
  void NimBleDriver::Init() const {
  ESP_LOGI(LOG_TAG, " initialize BT Controller and nimble stack");
  if (const esp_err_t resultCode = nimble_port_init(); resultCode != ESP_OK) {
    ESP_LOGE(LOG_TAG, "Failed nimble_port_init, error: %d", resultCode);
    return;
  }
  ble_hs_cfg.sync_cb = OnSync;
  ble_hs_cfg.reset_cb = OnReset;
  ble_svc_gap_init();
  ble_svc_gatt_init();
  if (ble_gatts_count_cfg(m_gattSvr) != 0) {
    isl_assert(false, "m_gattSvr contains at least one invalid resource definition");
  }
  if (ble_gatts_add_svcs(m_gattSvr) != 0) {
    isl_assert(false, "Couldn't add all resource definitions, out of ram!");
  }
  etl::string<MAX_BLE_DEVICE_NAME_LENG> deviceId{};
  if (!m_settingsHelper.GetSetting(Setting_e::DEVICE_ID, deviceId)) {
    ESP_LOGE(LOG_TAG, "Device ID not found!!");
  }
  instance->m_deviceName.append(deviceId);
  ESP_LOGI(LOG_TAG, "BLE device name '%s' set", instance->m_deviceName.c_str());
  isl_assert(ble_svc_gap_device_name_set(instance->m_deviceName.c_str()) == 0, "ble_svc_gap_device_name_set failed");
  nimble_port_freertos_init(HostTask);
[... 2001 lines cut ...]
D (9481) NimBLE: 0x0e
D (9481) NimBLE: 0x01
D (9491) NimBLE: 0x00
D (9491) NimBLE: 0x24
D (9491) NimBLE: 0x00
D (9491) NimBLE: 0x24
D (9501) NimBLE: 0x00
D (9501) NimBLE: 0x00
D (9501) NimBLE: 0x00
D (9501) NimBLE: 0xf4
D (9511) NimBLE: 0x01
D (9511) NimBLE: 0x00
D (9511) NimBLE: 0x00
D (9511) NimBLE: 0x00
D (9521) NimBLE: 0x00
D (9521) NimBLE:

D (9621) NimBLE: ble_hs_event_rx_hci_ev; opcode=0x3e
D (9621) NimBLE: subevent: 0x3
D (9621) NimBLE:

I (9621) NimBLE_DRIVER: unknown gap event: 3
```

i.e. there is no apparent crash :)

strange969, 2025-02-18:
Hi @florianhumblot, 

I tested this scenario using the IDF example bleprph and attempted to connect using python Bleak, but I did not encounter any stack smashing on my side.
 
Could you kindly share a minimal example code based on IDF examples that reproduces this issue? This would help us run it on our local device and analyse the behaviour more effectively. Additionally, it would be helpful if you could provide a detailed step-by-step description of how to reproduce the issue, including If you made any changes to sdkconfig , please include those as well so we can replicate the setup accurately.

florianhumblot, 2025-02-19:
@strange969 I'll try to build a minimal example.
In the meantime, here is a minimal sdkconfig of my setup:
```
CONFIG_APP_REPRODUCIBLE_BUILD=y
CONFIG_BOOTLOADER_APP_ROLLBACK_ENABLE=y
CONFIG_SECURE_BOOT=y
CONFIG_SECURE_BOOT_SIGNING_KEY="development_key.pem"
CONFIG_SECURE_BOOT_INSECURE=y
CONFIG_SECURE_BOOT_ALLOW_UNUSED_DIGEST_SLOTS=y
CONFIG_SECURE_INSECURE_ALLOW_DL_MODE=y
CONFIG_ESPTOOLPY_FLASHSIZE_4MB=y
CONFIG_PARTITION_TABLE_CUSTOM=y
CONFIG_PARTITION_TABLE_OFFSET=0x10000
CONFIG_COMPILER_OPTIMIZATION_SIZE=y
CONFIG_BT_ENABLED=y
CONFIG_BT_NIMBLE_ENABLED=y
CONFIG_BT_NIMBLE_LOG_LEVEL_DEBUG=y
CONFIG_BT_NIMBLE_MAX_CONNECTIONS=1
CONFIG_BT_NIMBLE_MAX_BONDS=2
CONFIG_BT_NIMBLE_MAX_CCCDS=1
CONFIG_BT_NIMBLE_HOST_TASK_STACK_SIZE=4096
# CONFIG_BT_NIMBLE_ROLE_CENTRAL is not set
# CONFIG_BT_NIMBLE_ROLE_BROADCASTER is not set
# CONFIG_BT_NIMBLE_ROLE_OBSERVER is not set
CONFIG_BT_NIMBLE_DEBUG=y
CONFIG_BT_NIMBLE_MSYS_1_BLOCK_SIZE=100
CONFIG_BT_NIMBLE_MSYS_2_BLOCK_COUNT=4
CONFIG_BT_NIMBLE_MSYS_2_BLOCK_SIZE=100
CONFIG_BT_NIMBLE_TRANSPORT_ACL_FROM_LL_COUNT=12
CONFIG_BT_NIMBLE_TRANSPORT_ACL_SIZE=100
CONFIG_BT_NIMBLE_TRANSPORT_EVT_SIZE=50
CONFIG_BT_NIMBLE_TRANSPORT_EVT_COUNT=5
CONFIG_BT_NIMBLE_TRANSPORT_EVT_DISCARD_COUNT=3
CONFIG_BT_NIMBLE_GATT_MAX_PROCS=3
CONFIG_BT_NIMBLE_WHITELIST_SIZE=5
CONFIG_BT_CTRL_BLE_MAX_ACT=2
CONFIG_BT_CTRL_ADV_DUP_FILT_MAX=15
CONFIG_BT_CTRL_HW_CCA_VAL=30
# CONFIG_BT_CTRL_BLE_ADV_REPORT_FLOW_CTRL_SUPP is not set
CONFIG_BT_CTRL_COEX_PHY_CODED_TX_RX_TLIM_EN=y
CONFIG_BLE_MESH=y
CONFIG_ETH_SPI_ETHERNET_W5500=y
CONFIG_ESP_SYSTEM_EVENT_QUEUE_SIZE=16
CONFIG_ESP_SYSTEM_EVENT_TASK_STACK_SIZE=2048
CONFIG_ESP_MAIN_TASK_STACK_SIZE=4096
CONFIG_ESP_CONSOLE_SECONDARY_NONE=y
CONFIG_ESP_TASK_WDT_TIMEOUT_S=15
CONFIG_ESP_WIFI_STATIC_RX_BUFFER_NUM=8
CONFIG_ESP_WIFI_DYNAMIC_RX_BUFFER_NUM=16
CONFIG_ESP_WIFI_DYNAMIC_TX_BUFFER_NUM=16
CONFIG_ESP_WIFI_DYNAMIC_RX_MGMT_BUFFER=y
CONFIG_ESP_WIFI_MGMT_SBUF_NUM=16
# CONFIG_ESP_WIFI_IRAM_OPT is not set
# CONFIG_ESP_WIFI_RX_IRAM_OPT is not set
# CONFIG_ESP_WIFI_GMAC_SUPPORT is not set
# CONFIG_ESP_WIFI_SOFTAP_SUPPORT is not set
CONFIG_HEAP_TRACING_STANDALONE=y
CONFIG_HEAP_USE_HOOKS=y
CONFIG_HEAP_ABORT_WHEN_ALLOCATION_FAILS=y
CONFIG_LWIP_MAX_SOCKETS=5
CONFIG_LWIP_TCP_SND_BUF_DEFAULT=4096
CONFIG_LWIP_TCP_WND_DEFAULT=4096
CONFIG_MBEDTLS_DYNAMIC_BUFFER=y
CONFIG_MBEDTLS_DYNAMIC_FREE_CONFIG_DATA=y
CONFIG_MBEDTLS_ECP_RESTARTABLE=y
CONFIG_MBEDTLS_TLS_CLIENT_ONLY=y
```

florianhumblot, 2025-02-26:
It seems to have to do with one or a combination of these settings:
```
CONFIG_BT_NIMBLE_MAX_CONNECTIONS=1
CONFIG_BT_NIMBLE_MAX_BONDS=2
CONFIG_BT_NIMBLE_MAX_CCCDS=1
CONFIG_BT_NIMBLE_MSYS_1_BLOCK_SIZE=100
CONFIG_BT_NIMBLE_MSYS_2_BLOCK_COUNT=4
CONFIG_BT_NIMBLE_MSYS_2_BLOCK_SIZE=100
CONFIG_BT_NIMBLE_TRANSPORT_ACL_FROM_LL_COUNT=12
CONFIG_BT_NIMBLE_TRANSPORT_ACL_SIZE=100
CONFIG_BT_NIMBLE_TRANSPORT_EVT_SIZE=50
CONFIG_BT_NIMBLE_TRANSPORT_EVT_COUNT=5
CONFIG_BT_NIMBLE_TRANSPORT_EVT_DISCARD_COUNT=3
CONFIG_BT_NIMBLE_GATT_MAX_PROCS=3
CONFIG_BT_NIMBLE_WHITELIST_SIZE=5
CONFIG_BT_CTRL_BLE_MAX_ACT=2
CONFIG_BT_CTRL_ADV_DUP_FILT_MAX=15
CONFIG_BT_CTRL_HW_CCA_VAL=30
```

By restoring them to default values, nimble works just fine. I'll try to reduce the ram usage of other components instead of this for now.

closed 2025-02-26 by florianhumblot