espressif/esp-idf #15402: esp32c5-wroom can not be work

skylin008, 2025-02-15:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I had designed PCB for esp32c5-wroom(module show v1.1 character) with eco0 version, it work well. But replace esp32c5-wroom with eco1(module show v1.2 character) version, it can not be work not only flash download, but also flash_id erase_flash so on, it only show "BESP-ROM:esp32c5-eco1-20240726" continuely. How to solve this issue? Thanks!

suda-morris, 2025-02-17:
Which branch are you working on?

I just tried on the master branch, and I can flash a binary to my C5 board

```text
ESP-ROM:esp32c5-eco1-20240726
Build:Jul 26 2024
rst:0x1 (POWERON),boot:0x18 (SPI_FAST_FLASH_BOOT)
SPI mode:DIO, clock div:1
load:0x40855ab0,len:0x176c
load:0x4084bfa0,len:0xcfc
load:0x4084e9a0,len:0x3050
entry 0x4084bfaa
I (23) boot: ESP-IDF v5.5-dev-1919-g3ec476405f5 2nd stage bootloader
I (23) boot: compile time Feb 17 2025 10:11:31
I (24) boot: chip revision: v0.2
```

skylin008, 2025-02-17:
I used esptool v4.8 tool to download， it can not enter flash download mode.

dobairoland, 2025-02-17:
https://github.com/espressif/esptool/commit/6cc002c4bd0de6a5b1afa630a59c7cbaac44ab86 - The ECO1 support of ESP32-C5 is in the 4.9.dev2 or newer development releases. 4.8 doesn't have it. This development release is available only in the ESP-IDF master branch for now. 

You can install esptool separately from ESP-IDF with this command:
```sh
pip install -U "esptool~=4.9.dev4"
```
and based on these information: https://docs.espressif.com/projects/esptool/en/latest/esp32/installation.html#virtual-environment-installation.

radimkarnis, 2025-02-24:
Hi @skylin008,

you mention that `it can not enter flash download mode` and `it only show "BESP-ROM:esp32c5-eco1-20240726" continuely`. This probably points to a HW issue. Please check that the different module versions don't introduce any changes incompatible with your custom PCB design. 

Entering the download mode is crucial for esptool to work properly, it is also highly dependent on a correct implementation of the resetting circuit. If your chip is getting stuck in a boot loop or is unable to enter the download mode using the `DTR` and `RTS` lines of the USB-to-UART bridge, it's a PCB design issue.

dobairoland, 2025-03-03:
I'm closing this for inactivity.

closed 2025-03-03 by dobairoland

Lucia-Rubio-E, 2025-05-21:
Hi!
Where did you buy ESP32-C5 v1.2??
Thanks