espressif/esp-idf #15403: ESP-IDF upgrade 5.3 to 5.4 breaks deep sleep wake from touch for stable code ESP32S3

wreyford, 2025-02-15:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.4

### Espressif SoC revision.

ESP32S3 v0.2

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

Custom PCB usinf ESP32S3 F16 R8

### Power Supply used.

USB

### What is the expected behavior?

After **Flashing** the device from VSCode IDE using Espressif Extension, I place the device in deep sleep by pressing power down button on home screen.
When touching the touch sensor, I expect the device to wake from deep sleep.

After restarting the device with **Monitor** from VSCode IDE using Espressif Extension, I place the device in deep sleep by pressing power down button on home screen.
When touching the touch sensor, I expect the device to wake from deep sleep.

After purposefully calling abort() - when I browse to my about screen, to simulate a crash, I expect that after the device wakes up, and I put it to sleep by pressing the power down button, that it should wake from deep sleep.

### What is the actual behavior?

In all three of the above situations it fails to wake from deep sleep.
I try to monitor the wake stub over USB form a terminal app, but get no output.

When I power-cycle the device, then put it into deep sleep by going to the home screen and pressing the power down button, it correctly wakes from deep sleep, with logs from the wake stub correctly showing up on a terminal app.

Thereafter all subsequent deep sleep - wake on touch cycles work perfectly, until monitor, flash or abort().

We have not changed out rtc_wake_stub's code.
We have not altered our touch sensor code.

This just happened after updating from 5.3 to 5.4

### Steps to reproduce.

No code, as nothing changed between 5.3 to 5.4

### Debug Logs.

```plain
No logs
```

esp-wzh, 2025-02-17:
>  I try to monitor the wake stub over USB form a terminal app, but get no output.

When using monitor or do flash, are you using the USB-UART converter module or the built-in USB serial on esp32s3? @wreyford

wreyford, 2025-02-17:
@esp-wzh, I monitor and flash over OTG using the built in USB on the esp32s3.

esp-wzh, 2025-02-18:
When wake up the chip by touch, I think we need to distinguish whether the built-in USB is not working cause you can't see the log, or the chip is really not woken up. Can you use UART monitor to reproduce the above problem? @wreyford

wreyford, 2025-02-18:
In Vscode Insiders, i flash and monitor UART, using built in otg of esp32s3 chip. 
The device does not come on at all. It does not power up the lcd or other peripherals. I need to remove battery from it (power cycle) it, then all is back to normal. Thx

esp-wzh, 2025-02-26:
Sorry I can't reproduce your problem, could you provide a minimal reproducible example?

wreyford, 2025-02-26:
Dear @esp-wzh esp-wzh, I have spent the last few days doing just that. Only to find that it works as expected.
I re-iterate though. We have quite a stable base of code, and we updated to 5.4 , but before that we did not experience this issue.
I'm gradually adding pieces of our code back, but as I'm sure you will understand this is a major undertaking.
As soon as I narrow down the issue, I will let you know. I'm trying my utmost best to reproduce it, in an example of code.
I have even gone as far as to use our specific partition table, sdkconfig, hoping that that would perhaps draw out the problem, but not yet.

esp-wzh, 2025-03-07:
```
    esp_rom_printf("sleep_flags:0x%08x\n",sleep_flags);
    esp_rom_output_tx_wait_idle(0);
```

Hi, @wreyford , thanks for feedback, could you please help to add the above code [here](https://github.com/espressif/esp-idf/blob/v5.4/components/esp_hw_support/sleep_modes.c#L778) to print the sleep power domain configuration information? I need to know this info when the touch wakeup problem occurred.

wreyford, 2025-03-23:
Hi, @esp-wzh , sorry for the delayed answer. Cannot compile when adding the two lines of code to my wake stub.
Looks like you may be using a unity? component. Where is pd_flags defined?
If you can give me some pointers I will give you the required feedback.

esp-wzh, 2025-03-23:
`pd_flags` is the first parameter passed to the `esp_sleep_start` function in the `esp-idf/components/esp_hw_support/sleep_modes.c` file. In the newer version of IDF, it is renamed to `sleep_flags`.

wreyford, 2025-03-23:
Hello @esp-wzh,
I'm using ESP-IDF 5.4
I tried to compile after changing to `sleep_flags`.
I get the following errors:
`C:/ESP32/Projects/deep_sleep_wake_stub/main/rtc_wake_stub_example.c:79:39: error: 'sleep_flags' undeclared (first use in this function)
   79 |   esp_rom_printf("pd_flags:0x%08x\n", sleep_flags);
      |                                       ^~~~~~~~~~~
C:/ESP32/Projects/deep_sleep_wake_stub/main/rtc_wake_stub_example.c:79:39: note: each undeclared identifier is reported only once for each function it appears in
C:/ESP32/Projects/deep_sleep_wake_stub/main/rtc_wake_stub_example.c:80:3: error: implicit declaration of function 'esp_rom_output_tx_wait_idle'; did you mean 'esp_rom_output_to_channels'? [-Wimplicit-function-declaration]
   80 |   esp_rom_output_tx_wait_idle(0);
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~
      |   esp_rom_output_to_channels
ninja: build stopped: subcommand failed.`

I'm including the code in the beginning of the wake_stub.
I searched for : ` esp_rom_output_tx_wait_idle` in ESP-IDF programming guide and found no reference to it?

I tried including: `#include "esp_rom_sys.h"`, but did not help.

Sorry will need some help from you to get it to compile.

esp-wzh, 2025-03-24:
@wreyford  Oh, sorry I didn't make it clear, I want to add it directly to the function [`esp_sleep_start`](https://github.com/espressif/esp-idf/blob/v5.4/components/esp_hw_support/sleep_modes.c#L778) `in esp-idf/components/esp_hw_support/sleep_modes.c`, not in the stub code, `esp_rom_output_tx_wait_idle` is declared in `esp_rom_uart.h` and `esp_rom_printf` is declared in `esp_rom_sys.h.`

0xPiyush, 2025-05-30:
Hello, I have encounterd the same issue my ESP32 S3 will not wake up from deep sleep (using deep sleep timer) when using the internal 17.5 MHz oscillator. I am on ESP IDF v5.4.1

esp-wzh, 2025-07-02:
@dvosully Sorry, I didn't reproduce your problem, but I think GPIO wakeup failure is not the same problem as described in this tread. Can you create a new issue and describe the complete reproduction steps?