espressif/esp-idf #15404: dm9051: add compatible mode for alternatives

AlanCui4080, 2025-02-16:
### Is your feature request related to a problem?

Yes, we are using CH390 to replace DM9051, this product is binary compatible with DM9051, but have different VID&PID also OUI&Model, so it failed at the id checking.

### Describe the solution you'd like.

Add an option to skip VID&PID,OUI&Model checking or custom VID&PID,OUI&Model in the component configuration

suda-morris, 2025-02-17:
Hi @AlanCui4080 Have you ever tried the CH390 driver [here](https://components.espressif.com/components/espressif/ch390/versions/0.2.1)?

AlanCui4080, 2025-02-17:
> Hi [@AlanCui4080](https://github.com/AlanCui4080) Have you ever tried the CH390 driver [here](https://components.espressif.com/components/espressif/ch390/versions/0.2.1)?

Thank you, This driver works well.

closed 2025-02-17 by AlanCui4080