espressif/esp-idf #15406: [ble-mesh] oob event enums are wrong

chegewara, 2025-02-16:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi team,
sorry, im not complaining because ble-mesh is a good piece of work and probably i am dumb, but naming convention in here is confusing a bit:
```
    /**
     * @brief ESP_BLE_MESH_PROVISIONER_PROV_INPUT_EVT
     */
    struct ble_mesh_provisioner_prov_input_evt_param {
        esp_ble_mesh_oob_method_t method;       /*!< Method of device Output OOB Authentication */
        esp_ble_mesh_output_action_t action;    /*!< Action of device Output OOB Authentication */
        uint8_t size;                           /*!< Size of device Output OOB Authentication */
        uint8_t link_idx;                       /*!< Index of the provisioning link */
    } provisioner_prov_input;                   /*!< Event parameter of ESP_BLE_MESH_PROVISIONER_PROV_INPUT_EVT */
    /**
     * @brief ESP_BLE_MESH_PROVISIONER_PROV_OUTPUT_EVT
     */
    struct ble_mesh_provisioner_prov_output_evt_param {
        esp_ble_mesh_oob_method_t method;       /*!< Method of device Input OOB Authentication */
        esp_ble_mesh_input_action_t action;     /*!< Action of device Input OOB Authentication */
        uint8_t size;                           /*!< Size of device Input OOB Authentication */
        uint8_t link_idx;                       /*!< Index of the provisioning link */
        /** Union of output OOB */
        union {
            char string[8];                     /*!< String output by the Provisioner */
            uint32_t number;                    /*!< Number output by the Provisioner */
        };
    } provisioner_prov_output;                  /*!< Event parameter of ESP_BLE_MESH_PROVISIONER_PROV_OUTPUT_EVT */

```
Output action in input event? 

I cant find any example how to use OOB events, sorry for being so lame.

Thanks

chegewara, 2025-02-16:
Hi @forx157 
Is there any way to select on provisioner OOB method if node can use few methods, like input, output and static OOB?
At the moment only 1 event is fired - `ESP_BLE_MESH_PROVISIONER_PROV_INPUT_EVT`.

I mean the way it is possible in nrf mesh app.

I also cant find the way to handle static OOB when node is setup with:
```
provision.oob_type = 1;
```

Thanks

forx157, 2025-02-17:
Regarding the naming issue of input and output, it is indeed very easy to cause misunderstandings here. In the future, we will provide more detailed explanations in the comments. 

In the event ESP_BLE_MESH_PROVISIONER_PROV_INPUT_EVT, the input is from the perspective of the provisioner, indicating that the provisioner is currently performing an input action. In esp_ble_mesh_output_action_t, the output is from the perspective of the unprov device, indicating the behavior of the unprov device at the moment.vice versa.

chegewara, 2025-02-17:
With help of nrf mesh and a lot of tries i finally figured it out and i have working OOB in node and provisioner.

The only problem i have is there is no event to handle static OOB in provisioner.
In node code it work as far as i can tell with short test using nrf mesh app as provisioner.

Thanks

forx157, 2025-02-17:
Currently, the provisioner cannot choose among multiple OOB methods supported by the unprov device. However, you can describe in detail the application scenarios for making such choices, and I can discuss it internally.

Regarding the static OOB issue, you can configure the static OOB in `esp_ble_mesh_prov_t`. The static OOB cannot be changed during the operation of the provisioner and can only be modified after deinitialization and reinitialization.

chegewara, 2025-02-17:
There is no particular scenario i can think of, just trying to build some ble-mesh component and i am comparing features on esp32 with nrf mesh.

> Regarding the static OOB issue, you can configure the static OOB in `esp_ble_mesh_prov_t`. The static OOB cannot be changed during the operation of the provisioner and can only be modified after deinitialization and reinitialization.

What if i have different nodes from different companies and each one have different static OOB?

What i tested so far is this code in node
```
        provision.oob_type = 1;
        provision.static_val_len = 4;
        static const uint8_t buf[] = {1,2,3,4};
        provision.static_val = buf;
``` 
With nrf mesh i can provisioning it. I dont know how to do it with esp32.

Thanks

forx157, 2025-02-17:
Static OOB is indeed “static” and should not be changed during the lifecycle of the provisioner. Regarding the scenario you mentioned, where our provisioner is used to provisioning devices from other companies that support both static OOB and other methods, our provisioner currently defaults to static OOB. This can lead to provisioning failures due to differences in static OOB values. This is indeed a problem. Currently, the only solution is for the unprovisioned device to disable static OOB to proceed with provisioning. If we could support the selection of multiple methods, this issue could be resolved more conveniently. I will propose this suggestion internally. Thank you very much for raising this question.

chegewara, 2025-02-17:
It not something important i think, but thank you for patience and considering it something that needs to be resolved.

Thanks

forx157, 2025-03-07:
Hi， @chegewara 

Would it be convenient to close the issue if there are no other questions on this topic? Feel free to open it again if there are any other subsequent queries as well.

chegewara, 2025-03-07:
Hi,
Yes we can close it.
I just left it open because I thought we are waiting to implement features related to static oob.

Thanks

closed 2025-06-10 by chegewara