espressif/esp-idf #15407: Consider changing default flash size from 2M to 4M for ESP32-C3, S3 and possibly others

mnigbor, 2025-02-16:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Please consider changing the default flash size from 2MB to 4MB for example and test programs built for the ESP32-C3, S3 and possibly other ESP32 devices with on-chip JTAG support.  This will improve the chance that debugging example programs works out-of-the-box on these platforms with a fresh installation. 

Here's how I encountered the issue leading to this recommendation:

I was attempting to debug the hello_world example program on a Xaio ESP32-C3 using the ESP-IDF command-line on Windows 11. I installed ESP-IDF 5.4 using the online installer and picked defaults throughout. The installation went smoothly and I was able to build, flash, and monitor the hello_world application without encountering any major issues. 

Unfortunately, attempts to debug the application did not go smoothly at all.  I encountered two basic issues:

1) OpenOCD would not run because it could not connect to the ESP32-C3. This was fixed by _downgrading_ the USB driver for the USB JTAG/serial debug unit (Interface 2) from 10.XXX to 6.xxx.  [You can read more about it here](https://stackoverflow.com/questions/79440857/unable-to-debug-esp32-c3-using-vs-code-esp-idf-extension-and-internal-jtag-adapt). It would be nice if you or the OpenOCD people could fix this, but it's not the focus of this issue.
2) GDB would not connect to OpenOCD because OpenOCD could not read required information from the binary on the ESP32-C3 because it's stack size is less than 4M. Changing the stack size from 2M to 4M (using menuconfig) fixed it.  [You can read more about it here](https://github.com/espressif/openocd-esp32/issues/315#issuecomment-2661587948). This works around a [known issue with OpenOCD](https://github.com/espressif/openocd-esp32/issues/315#issuecomment-1959457950).  My suggestion is about this issue.

As things stand today, you cannot debug hello_world or many other test and example applications supplied in the ESP-IDF on many newer, popular ESP32 devices unless you make these two relatively obscure fixes. I think this represents a significant impediment to ESP-IDF use.

Icarus113, 2025-02-17:
Hi @mnigbor , for this one

> GDB would not connect to OpenOCD because OpenOCD could not read required information from the binary on the ESP32-C3 because it's stack size is less than 4M. Changing the stack size from 2M to 4M (using menuconfig) fixed it. https://github.com/espressif/openocd-esp32/issues/315#issuecomment-2661587948. This works around a https://github.com/espressif/openocd-esp32/issues/315#issuecomment-1959457950. My suggestion is about this issue.

If you are referring to the [`ESPTOOLPY_FLASHSIZE`](https://github.com/espressif/esp-idf/blob/master/components/esptool_py/Kconfig.projbuild#L114), we should not change the default value of this one. 

Because this one actually decides how big the flash size can be used. We have many products and the least flash size is 2MB, see [here](https://products.espressif.com/#/product-selector?names=&filter={%22Series%22:[%22ESP32-C3(including%20ESP8685)%22]}). If by default the size is set to 4MB, then some of the checks inside the ESP-IDF drivers may consider for example 0x3C00000 (3MB) flash offset as an valid one and try to do something to the address, which will lead to undefined behaviours.

mnigbor, 2025-02-18:
It turns out I was chasing a red herring. Debugging the ESP32-C3 works fine with default settings if you fix the USB driver and pull D8 (GPIO8) high. Leaving it floating was causing inconsistent debugging behavior (mostly not working).  I believe it just happened to work when I experimented with changes to flash size.

Thanks to [erhankur](https://github.com/erhankur)https://github.com/espressif/openocd-esp32/issues/358#issuecomment-2664280688 for putting me on the right track.

I'm closing the issue.

closed 2025-02-18 by mnigbor