espressif/esp-idf #15409: Cannot run the mipi_isp_dsi example on ESP32-P4 without DSI

lamraoui, 2025-02-17:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi,

I'm trying to run the _mipi_isp_dsi_ [1] example without DSI on a ESP32-P4 dev board from Olimex [2] and with a OV5647 camera.
Basically I just want to record the video, without using any display.
Unfortunatly the callback _s_camera_get_finished_trans_ is never called.

Here is a trace:

```
D (1872) app_start: Starting scheduler on CPU1
D (1872) intr_alloc: Connected src 54 to int 1 (cpu 1)
D (1872) heap_init: New heap initialised at 0x4ff3afc0
D (1872) heap_init: New heap initialised at 0x4ff40000
I (1872) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations
D (1872) esp_psram: Allocating block of size 32768 bytes
D (1872) intr_alloc: Connected src 48 to int 3 (cpu 0)
I (1872) main_task: Calling app_main()
I (1872) mipi_isp_dsi: App started
D (1902) mipi_isp_dsi: CONFIG_EXAMPLE_MIPI_CSI_DISP_HRES: 800, CONFIG_EXAMPLE_MIPI_DSI_DISP_VRES: 1280, bits per pixel: 16
D (1902) mipi_isp_dsi: frame_buffer_size: 2048000
D (1902) mipi_isp_dsi: frame_buffer: 0x48000a40
D (1902) i2c.common: new bus(0) at 0x481f4d18
D (1902) i2c.common: bus clock source frequency: 40000000hz
D (1902) intr_alloc: Connected src 44 to int 4 (cpu 0)
D (1902) sccb_i2c: new io_i2c: 0x481f4e5c
I (1902) ov5647: Detected Camera sensor PID=0x5647
I (1902) sensor_init: fmt[0].name:MIPI_2lane_24Minput_RAW8_800x1280_50fps
I (1902) sensor_init: fmt[1].name:MIPI_2lane_24Minput_RAW8_800x640_50fps
I (1902) sensor_init: fmt[2].name:MIPI_2lane_24Minput_RAW8_800x800_50fps
I (1902) sensor_init: fmt[3].name:MIPI_2lane_24Minput_RAW10_1920x1080_30fps
I (1902) sensor_init: fmt[4].name:MIPI_2lane_24Minput_RAW10_1280x960_binning_45fps
D (1912) ov5647: count=4
D (1942) ov5647: count=88
D (1952) ov5647: Stream=0
D (1952) ov5647: light freq=0x3c
I (1962) sensor_init: Format in use:MIPI_2lane_24Minput_RAW8_800x800_50fps
D (1962) ov5647: Stream=1
D (1962) CSI: config->queue_items: 1
D (1962) CSI: ctlr->h_res: 0d 800
D (1962) CSI: ctlr->v_res: 0d 800
D (1962) CSI: ctlr->in_bpp: 0d 8
D (1962) CSI: ctlr->out_bpp: 0d 16
D (1962) CSI: ctlr->fb_size_in_bytes=1280000
D (1962) CSI: alignment: 0x40

D (1962) CSI: ctlr->backup_buffer: 0x481f4f40

D (1962) csi_hal: CSI-DPHY lane_rate: 200000000 Hz, hs_freq: 0x22, lane_num: 0x2
D (1962) dw-gdma: new group (0) at 0x4832d774
D (1962) dw-gdma: new channel (0,0) at 0x4832d744
D (1962) CSI: csi_transfer_size: 0d 80000
D (1962) intr_alloc: Connected src 24 to int 4 (cpu 0)
D (1962) dw-gdma: install interrupt service for channel (0,0)
D (1992) CSI: trans.buffer: 0x48000a40, trans.buflen: 2048000
I (1992) mipi_isp_dsi: Main loop
I (1992) mipi_isp_dsi: esp_cam_ctlr_receive
I (1992) mipi_isp_dsi: esp_cam_ctlr_receive
```

Here his the source code where all DSI codes were commented out:

```
[... 159 lines cut ...]
    }
}

static bool s_camera_get_new_vb(esp_cam_ctlr_handle_t handle, esp_cam_ctlr_trans_t *trans, void *user_data)
{
    esp_cam_ctlr_trans_t new_trans = *(esp_cam_ctlr_trans_t *)user_data;
    trans->buffer = new_trans.buffer;
    trans->buflen = new_trans.buflen;

    return false;
}

static bool s_camera_get_finished_trans(esp_cam_ctlr_handle_t handle, esp_cam_ctlr_trans_t *trans, void *user_data)
{
    finished_trans_counter++;
    return false;
}
```

[1] https://github.com/espressif/esp-idf/tree/master/examples/peripherals/camera/mipi_isp_dsi
[2] https://www.olimex.com/Products/IoT/ESP32-P4/ESP32-P4-DevKit/open-source-hardware

**IDF Version.**

ESP-IDF v5.5-dev-1655-gc5865270b5-dirty

Icarus113, 2025-02-18:
Hi @lamraoui , I've tried this with latest [example](https://github.com/espressif/esp-idf/blob/master/examples/peripherals/camera/mipi_isp_dsi/main/mipi_isp_dsi_main.c), using 0v5647, and also commented out the dsi related. I can still get `s_camera_get_finished_trans` called. 

```
W (965) spi_flash: Detected size(16384k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (978) main_task: Started on CPU0
I (988) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (988) main_task: Calling app_main()
W (1018) mipi_isp_dsi: CONFIG_EXAMPLE_MIPI_CSI_DISP_HRES: 800, CONFIG_EXAMPLE_MIPI_DSI_DISP_VRES: 1280, bits per pixel: 16
W (1018) mipi_isp_dsi: frame_buffer_size: 2048000
I (1018) ov5647: Detected Camera sensor PID=0x5647
I (1028) sensor_init: fmt[0].name:MIPI_2lane_24Minput_RAW8_800x1280_50fps
I (1028) sensor_init: fmt[1].name:MIPI_2lane_24Minput_RAW8_800x640_50fps
I (1038) sensor_init: fmt[2].name:MIPI_2lane_24Minput_RAW8_800x800_50fps
I (1048) sensor_init: fmt[3].name:MIPI_2lane_24Minput_RAW10_1920x1080_30fps
I (1048) sensor_init: fmt[4].name:MIPI_2lane_24Minput_RAW10_1280x960_binning_45fps
I (1108) sensor_init: Format in use:MIPI_2lane_24Minput_RAW8_800x800_50fps
s_camera_get_new_vb
s_camera_get_new_vb
s_camera_get_finished_trans
s_camera_get_new_vb
s_camera_get_finished_trans
s_camera_get_new_vb
s_camera_get_finished_trans
```
could you try to check if the callback `s_camera_get_new_vb` is called by adding a `esp_rom_printf("s_camera_get_finished_trans\n");` inside it. As `ESP_ERROR_CHECK(esp_cam_ctlr_receive(cam_handle, &new_trans, ESP_CAM_CTLR_MAX_DELAY));` this line may never return.

lamraoui, 2025-02-18:
Hi @Icarus113 ,

It works!
Thank you @Icarus113 !

closed 2025-02-18 by lamraoui

intikhabalam12, 2025-03-22:
> Hi [@Icarus113](https://github.com/Icarus113) ,
> 
> It works! Thank you [@Icarus113](https://github.com/Icarus113) !

How did you make it works. Kindly explain as I cannot make s_camera_get_new_vb to run. Only s_camera_get_finished_trans is calling repeatedly and it too doesn't end the while loop if I make the while loop conditional instead of always true

lamraoui, 2025-04-01:
Hi @intikhabalam12,

Can you share your source code?

intikhabalam12, 2025-04-02:
@lamraoui Thanks for replying. It worked as I got to understand that the code will not go to the next line once it gets to 
ESP_ERROR_CHECK(esp_cam_ctlr_receive(cam_handle, &new_trans, ESP_CAM_CTLR_MAX_DELAY));.

xiarenxian, 2025-05-28:
> Hi [@Icarus113](https://github.com/Icarus113) ,
> 
> It works! Thank you [@Icarus113](https://github.com/Icarus113) !

So where can I get a stream?I printed the length and the pointer of buff.its no change

brianmichalk1, 2025-06-22:
I am interested in this also.  I am not understanding why the second call to esp_cam_ctlr_receive hangs.
I added debug statements to the callback routines, and they are running.

HonestQiao, 2025-07-21:
I'm also running into the same issue. 
Would anyone be able to share working code that successfully runs after commenting out the MIPI-DSI LCD part?