espressif/esp-idf #15410: SDMMC_example_project

NewNoob2002, 2025-02-17:
IDF version.
v5.3.2

Operating System used.
Linux

How did you build your project?
Command line with idf.py

If you are using Windows, please specify command line type.
None

Development Kit.
ESP32-WROOM-EU

Power Supply used.
Battery

What is the expected behavior?
Should mount the SD Nand and write something into it

What is the actual behavior?
Upon running the example and opening the monitor using the idf.py script, I observe the following problem:

I can't find the define of CONFIG_SOC_SDMMC_USE_GPIO_MATRIX, where i need to configure this.
not use the default pins in ESP-IDF document. I want use IO-19, IO-20 and IO-23

[image]

```
#ifdef CONFIG_EXAMPLE_SDMMC_BUS_WIDTH_4
    slot_config.width = 4;
#else
    slot_config.width = 1;
#endif
    // On chips where the GPIOs used for SD card can be configured, set them in
    // the slot_config structure:
#ifdef CONFIG_SOC_SDMMC_USE_GPIO_MATRIX
    slot_config.clk = CONFIG_EXAMPLE_PIN_CLK;
    slot_config.cmd = CONFIG_EXAMPLE_PIN_CMD;
    slot_config.d0 = CONFIG_EXAMPLE_PIN_D0;
#ifdef CONFIG_EXAMPLE_SDMMC_BUS_WIDTH_4
    slot_config.d1 = CONFIG_EXAMPLE_PIN_D1;
    slot_config.d2 = CONFIG_EXAMPLE_PIN_D2;
    slot_config.d3 = CONFIG_EXAMPLE_PIN_D3;
#endif  // CONFIG_EXAMPLE_SDMMC_BUS_WIDTH_4
#endif  // CONFIG_SOC_SDMMC_USE_GPIO_MATRIX
```

igrr, 2025-02-17:
> Upon running the example and opening the monitor using the idf.py script, I observe the following problem:

> I can't find the define of CONFIG_SOC_SDMMC_USE_GPIO_MATRIX, where i need to configure this. not use the default pins in ESP-IDF document. I want use IO-19, IO-20 and IO-23

With an ESP32 chip, you can't.

ESP32's SDMMC peripheral can work with a fixed set of pins from one of the two slots. Since you are using a WROOM module, one of those two slots is occupied by SPI flash. This leaves just one set of pins for SDMMC:

[image]

[image]

(Source: [ESP32 datasheet](https://www.espressif.com/sites/default/files/documentation/esp32_datasheet_en.pdf))

The capability to freely configure SDMMC through GPIO matrix (denoted as `CONFIG_SOC_SDMMC_USE_GPIO_MATRIX`) peripheral pins exists in ESP32-S3 (for slots 0 and 1) and ESP32-P4 (for slot 1).

NewNoob2002, 2025-02-17:
> With an ESP32 chip, you can't.
>
> ESP32's SDMMC peripheral can work with a fixed set of pins from one of the two slots. Since you are using a WROOM module, one of those two slots is occupied by SPI [...]
ok, Thank you, i will check