espressif/esp-idf #15411: Sharp Corner: Easy to select wrong `esp_idf_version` (RDT-1153)

BrianPugh, 2025-02-17:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

A bit of a sharp corner, a github action definition might look like:

```yaml
      matrix:
        idf_ver: ["v5.0", "v5.1", "v5.2", "v5.3", "v5.4", "latest"]
        idf_target: ["esp32"]
        include:
          - idf_ver: "v5.2"
            idf_target: esp32s2
          - idf_ver: "v5.2"
            idf_target: esp32c3
          - idf_ver: "v5.2"
            idf_target: esp32s3
      steps:
          - name: esp-idf build library
            uses: espressif/esp-idf-ci-action@main
            with:
              esp_idf_version: ${{ matrix.idf_ver }}
              target: ${{ matrix.idf_target }}
```

This looks perfectly reasonable. However, it doesn't actually work _exactly_ as expected. One would expect the esp_idf_version `v5.0` to use the latest `v5.0.X` patch release of esp-idf. However, this is passed along as a docker tag, and `v5.0` is a pretty old docker tag that contains the initial release `v5.0.0`. The same goes for all the other versions. What actually work is:

```yaml
              esp_idf_version: release-${{ matrix.idf_ver }}
```

But this isn't exactly correct either, as this contains all the latest changes that might not actually be in a tagged version.

I think the real misleading issue here, is that the docker tag `v5.0` (and similar) should **lead to the latest patch release of that version**. I don't think anyone using the tag `v5.0` is/should rely/expect to get `v5.0.0`. They would expect to get the latest patch release. This goes the same for all bare `v5.X` tags.

kumekay, 2025-02-17:
@BrianPugh I've changed the title to reflect the actual change in terms of ESP-IDF better

dobairoland, 2025-02-24:
What about a special branch which is always fast-forwarded to the latest bugfix release? Tags shouldn't change over time. As a git user I would be pretty mad finding out that tags has been moved.

kumekay, 2025-02-24:
@dobairoland Right, for git tags it's better to keep them. and `stable/vX.Y` should be a branch pointing to the latest release. 
At the same time for docker images it's entirely ok to have tracking tags (connected to the branch) on git side.

BrianPugh, 2025-02-24:
Generally, in most systems, it's standard that semver-like references resolve to the highest version that meets the specified tag prefix:
* `v5.0.6` should resolve to _exactly_ v5.0.6.
* `v5.0` should resolve to the latest v5.0 patch (currently `v5.0.8`).
* `v5` should resolve to the latest v5 release (currently `v5.3.2`). 

The real current issue (in my mind) is that `v5.0` is pegged to `v5.0.0`.

Some examples of systems that work this way:
* poetry, uv, pip (python)
* npm (javascript)
* cargo (rust)
* github actions (though, this is a bit of a nuanced example)
* other docker images (e.g. `python:3.9` resolves to `python:3.9.21`)

This is why I'm pushing for the _slightly_ breaking change, as it would then be consistent with every other version resolver, and more in line with what a user would expect.

To be perfectly clear @dobairoland , I agree that _exact_ tags like `v5.0.8` should never change and be considered immutable.