espressif/esp-idf #15412: Component for JSN SR04M-2 sensor

allacmc, 2025-02-17:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I would like to know if there is a lib for the JSN SR04M-2 sensor. I found a similar lib but I'm not sure if it works exactly with this sensor.

[image]

I found the lib for: https://github.com/ESP32Tutorials/HC-SR04-Ultrasonic-Sensor-with-ESP32-ESP-IDF/tree/main/Ultrasonic_sensor_esp-idf

But I'm not sure if it works exactly with this sensor?

suda-morris, 2025-02-18:
Inspired from this article: https://www.mantech.co.za/Datasheets/Products/JSN-SR04M-200427.pdf Looks like it's as simple as initialize a UART port and then read the data periodically.

Since you also opened a same issue in the https://github.com/ESP32Tutorials/HC-SR04-Ultrasonic-Sensor-with-ESP32-ESP-IDF/issues/1 Can we close this redundant one?

allacmc, 2025-02-18:
It seems simple, but I would like to use a ready-made lib.

Made for esp idf

Alvin1Zhang, 2025-02-28:
Thanks for reporting, feel free to reopen.

closed 2025-02-28 by Alvin1Zhang

allacmc, 2025-02-28:
I solved this with lib ultrassonic.c