espressif/esp-idf #15413: 尽快发布最新的离线ide和idf安装程序

zz6zz666, 2025-02-18:
### Is your feature request related to a problem?

idf下载页面上提供了最新的离线idf安装程序(5.4)，但是包含ide的安装程序仅为idf v5.3，并且ide的最新版本是3.2.0，那个整合ide与idf的里面提供的ide也非最新版本。

### Describe the solution you'd like.

希望尽快整合推出含最新软件的安装程序，方便我们安装部署，因为自己单独安装这些组件总是遇到各种兼容性问题导致反复重装😭😭😭

### Additional context.

希望整合的离线的安装程序总能保持组件为最新版本😭😭😭

zz6zz666, 2025-02-18:
[image]

kolipakakondal, 2025-02-19:
Hello @zz6zz666  Sorry about that, that was a miss on our part. We will update the latest offline IDE installer with Espressif-IDE 3.2.0+ and ESP-IDF v5.4, and we will provide an update here. 

Thanks for bringing it to our notice.

zz6zz666, 2025-02-21:
用idf-v5.4搭配ide使用的时候，代码索引功能一直出问题。用ide-v3.1.0 with idf-v5.3.1安装程序安装完后，C++Indxer没有任何问题。如果使用online安装程序安装完idfv5.4的版本时候，ide打开时候C++Indexer不会默认工作，如果手动打开preference里面的C++Indexer选项，Indexer才能正常工作（ide-v3.1.0 with idf-v5.3.1并不需要如此）。但是这个v5.4每次启动都会报错，并且代码补全、代码跳转等功能时好时坏，经常要反复关闭再启动。报错截图如下

[image]

zz6zz666, 2025-02-21:
另外，就算C++Indexer能够工作，在main文件夹里CMakelists.txt文件添加一些新的头文件和源文件的时候，在其他文件中引用这些头文件以及里面的函数依然会报错，直到整个程序重新编译通过后才能够刷新正常索引。但是build完后如果有报错，这时候要去修改相应文件中代码时，每添加有一两个字符或单词，程序就会卡住加载一遍C++Indexer，卡住的时间经常长达一分钟甚至能多，而加载成功后我实际修改的代码也只是刚刚添加了一两个字符，连一个单词都来不及打完。或者遇到这种情况就干脆选择关闭ide，不得不转而用其他没有索引功能的编辑器去修改代码了。希望能改进索引的刷新机制。