espressif/esp-idf #15415: 在线文档网站在切换芯片target时会导致url中的zh_CN变为zh-CN从而转向404界面

airjinkela, 2025-02-18:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

在版本(version-select)选择release-v5.3及以后的版本，只要更改芯片目标（targets）或者版本（version-select）都会导致url中的zh_CN变为zh-CN从而转向404界面。
只有releases/v*版本和master版本有次问题,

Alvin1Zhang, 2025-02-18:
Thanks for reporting.

Alvin1Zhang, 2025-02-25:
Thanks for reporting and sorry for the slow turnaround, fix is available at https://github.com/espressif/esp-docs/commit/9cc13fabff1e25ad353d06bfd18858305b951159.

closed 2025-02-25 by Alvin1Zhang