espressif/esp-idf #15416: (minor) esp_hid: allow use of BT-SIG assigned vendor_id, not just USB-IF

karlp, 2025-02-18:
### Is your feature request related to a problem?

Using the esp_hid component, the PnP block is filled in based around values provided in the `esp_hid_device_config_t` struct, which has fields for vendor_id and product_id.  However, as of v5.4 at least, the vendor is is hard coded to be a USB-IF id: https://github.com/espressif/esp-idf/blob/v5.4/components/esp_hid/src/ble_hidd.c#L228

It would be nice if you could also use BT-SIG assigned company ids.  Not a very big deal though.

### Describe the solution you'd like.

It might be enough to just have a Kconfig on the component, I don't think it would really warrant new APIs

esp-zhp, 2025-05-08:
@karlp
Thanks again for pointing this out!

We’ve reviewed the code (specifically in `create_info_db()`), and you’re absolutely right — currently, the `vendor_id_source` is hardcoded as `0x02` (USB-IF), even though the Bluetooth specification also allows using `0x01` (BT-SIG company IDs).

As a **temporary workaround**, you can directly patch the line that builds the `pnp_val` array like this:

```c
uint8_t pnp_val[7] = {
    0x01, // Use 0x01 for BT-SIG instead of 0x02 (USB-IF)
    dev->config.vendor_id & 0xFF, (dev->config.vendor_id >> 8) & 0xFF,
    dev->config.product_id & 0xFF, (dev->config.product_id >> 8) & 0xFF,
    dev->config.version & 0xFF, (dev->config.version >> 8) & 0xFF
};
```

However, for a **long-term and robust solution**, we’d recommend extending the `esp_hid_device_config_t` struct to include a new field, e.g., `vendor_id_source`.
This would allow users to explicitly configure whether they want to use a BT-SIG assigned company ID (`0x01`) or a USB-IF vendor ID (`0x02`), without having to modify the component’s internal code.
It also keeps the component flexible and better prepared for the future, in case additional ID sources are introduced.

Do you prefer adding this as a configurable field in the struct (giving per-instance control), or would you rather have a global Kconfig option (applied at build time)?

Looking forward to your thoughts!

karlp, 2025-06-30:
I don't have any updates or preference on what the API should be.  I think kconfig is probably fine, I don't really believe you'd need both in the same project at the same time, and that's less overhead and no API changes.