espressif/esp-idf #15417: ESP8266 register addresses

nearest-river, 2025-02-18:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

Can you help me extract the actual memory addresses from my esp8266 nodemcu..
thanks in advance

```rs
// Memory-mapped register addresses (example values, replace with actual addresses)
const WIFI_MODE_REG: *mut u32 = 0x40212d74 as *mut u32; // WiFi mode register
const WIFI_SSID_REG: *mut u32 = 0x40212d78 as *mut u32; // SSID register
const WIFI_PASSWORD_REG: *mut u32 = 0x40212d7C as *mut u32; // Password register
const WIFI_CONNECT_REG: *mut u32 = 0x40212d80 as *mut u32; // Connection trigger register
const WIFI_STATUS_REG: *const u32 = 0x40212d84 as *const u32; // Connection status register

rs```

igrr, 2025-02-18:
These registers don't exist in the ESP8266, tbh they look made up.

closed 2025-02-18 by igrr

nearest-river, 2025-04-05:
> These registers don't exist in the ESP8266, tbh they look made up.

can you help me out with the ctrl register