espressif/esp-idf #15418: I can't build!

AMAZONGSANG, 2025-02-18:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.3.2-dirty

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

I want to build my project

### What is the actual behavior?

Compilation failure.
However, after the compilation fails, restart the computer and compile again to compile successfully
After the compilation fails, delete the "build" folder and restart the computer to compile again

### Steps to reproduce.

I'm sorry, I don't know.And then one day it was compiled and it was like this

### Build or installation Logs.

```plain
PS F:\ZZJ\zw\Firmware\zw-host> idf.py build
Executing action: all (aliases: build)
Running ninja in directory F:\ZZJ\zw\Firmware\zw-host\build
Executing "ninja all"...
[1/1] C:\WINDOWS\system32\cmd.exe /C "cd /D F:\ZZJ\zw\Firm...:/ZZJ/zw/Firmware/zw-host/build/bootloader/bootloader.bin"
Bootloader binary size 0x5a90 bytes. 0x2570 bytes (29%) free.
[4/8] Generating ld/sections.ld
FAILED: esp-idf/esp_system/ld/sections.ld F:/ZZJ/zw/Firmware/zw-host/build/esp-idf/esp_system/ld/sections.ld
esp-idf\esp_system\CMakeFiles\sections.ld-d65dc84.bat 40d683c3f21cac5e
Traceback (most recent call last):
  File "D:\ESP32\Espressif\python_env\idf5.3_py3.11_env\Lib\site-packages\pyparsing\core.py", line 850, in _parseNoCache
    loc, tokens = self.parseImpl(instring, pre_loc, do_actions)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\ESP32\Espressif\python_env\idf5.3_py3.11_env\Lib\site-packages\pyparsing\core.py", line 2470, in parseImpl
    if instring[loc] == self.firstMatchChar and instring.startswith(
       ~~~~~~~~^^^^^
IndexError: string index out of range

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "D:\ESP32\Espressif\frameworks\esp-idf-v5.3.2\tools\ldgen\ldgen\entity.py", line 124, in add_sections_info
    results = parser.parseString(first_line, parseAll=True)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\ESP32\Espressif\python_env\idf5.3_py3.11_env\Lib\site-packages\pyparsing\util.py", line 376, in _inner
    return fn(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\ESP32\Espressif\python_env\idf5.3_py3.11_env\Lib\site-packages\pyparsing\core.py", line 1216, in parse_string
    raise exc.with_traceback(None)
pyparsing.exceptions.ParseException: Expected 'In archive'  (at char 0), (line:1, col:1)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "D:\ESP32\Espressif\frameworks\esp-idf-v5.3.2\tools\ldgen\ldgen.py", line 176, in <module>
    main()
  File "D:\ESP32\Espressif\frameworks\esp-idf-v5.3.2\tools\ldgen\ldgen.py", line 133, in main
    sections_infos.add_sections_info(dump)
  File "D:\ESP32\Espressif\frameworks\esp-idf-v5.3.2\tools\ldgen\ldgen\entity.py", line 126, in add_sections_info
    raise ParseException('Parsing sections info for library ' + sections_info_dump.name + ' failed. ' + p.msg)
pyparsing.exceptions.ParseException: Parsing sections info for library F:/ZZJ/zw/Firmware/zw-host/build/esp-idf/esp_common/libesp_common.a failed. Expected 'In archive'  (at char 0), (line:1, col:1)
Batch file failed at line 3 with errorcode 1
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the F:\ZZJ\zw\Firmware\zw-host\build\log\idf_py_stderr_output_9156 and F:\ZZJ\zw\Firmware\zw-host\build\log\idf_py_stdout_output_9156
```

### More Information.

I have downloaded other versions of idf, but none of them worked

fhrbata, 2025-02-18:
Hello @AMAZONGSANG ,

it looks like an issue with the `objdump` output, possibly same as reported in https://github.com/espressif/esp-idf/issues/10440. It would be helpful to know the arguments passed to `ldgen.py`, particularly the `--objdump` argument. You might try running `idf.py -v build` in verbose mode to see how `ldgen.py` is executed. Hopefully, you should see it in the output before the error occurs.

Thank you

AMAZONGSANG, 2025-02-19:
I have learned about the #10440 issue, but unfortunately I don't understand it very well, and it is mac, I am not sure how to solve it on Windows

Here is the log I obtained using the "idf.py -v build" command

```
PS F:\ZZJ\zw\Firmware\zw-host> idf.py -v build
Checking Python dependencies...
Python requirements are satisfied.
Constraint file: D:\ESP32\Espressif\espidf.constraints.v5.3.txt
Requirement files:
 - D:\ESP32\Espressif\frameworks\esp-idf-v5.3.2\tools\requirements\requirements.core.txt
Python being checked: D:\ESP32\Espressif\python_env\idf5.3_py3.11_env\Scripts\python.exe
Checking used Python interpreter...
Executing action: all (aliases: build)
Running ninja in directory F:\ZZJ\zw\Firmware\zw-host\build
Executing "ninja -v all"...
[1/9] esp-idf\esp_system\CMakeFiles\sections.ld-d65dc84.bat 40d683c3f21cac5e
FAILED: esp-idf/esp_system/ld/sections.ld F:/ZZJ/zw/Firmware/zw-host/build/esp-idf/esp_system/ld/sections.ld
esp-idf\esp_system\CMakeFiles\sections.ld-d65dc84.bat 40d683c3f21cac5e
Traceback (most recent call last):
  File "D:\ESP32\Espressif\python_env\idf5.3_py3.11_env\Lib\site-packages\pyparsing\core.py", line 850, in _parseNoCache
    loc, tokens = self.parseImpl(instring, pre_loc, do_actions)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\ESP32\Espressif\python_env\idf5.3_py3.11_env\Lib\site-packages\pyparsing\core.py", line 2470, in parseImpl
    if instring[loc] == self.firstMatchChar and instring.startswith(
       ~~~~~~~~^^^^^
IndexError: string index out of range

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "D:\ESP32\Espressif\frameworks\esp-idf-v5.3.2\tools\ldgen\ldgen\entity.py", line 124, in add_sections_info
    results = parser.parseString(first_line, parseAll=True)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\ESP32\Espressif\python_env\idf5.3_py3.11_env\Lib\site-packages\pyparsing\util.py", line 376, in _inner
    return fn(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\ESP32\Espressif\python_env\idf5.3_py3.11_env\Lib\site-packages\pyparsing\core.py", line 1216, in parse_string
    raise exc.with_traceback(None)
pyparsing.exceptions.ParseException: Expected 'In archive'  (at char 0), (line:1, col:1)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "D:\ESP32\Espressif\frameworks\esp-idf-v5.3.2\tools\ldgen\ldgen.py", line 176, in <module>
    main()
  File "D:\ESP32\Espressif\frameworks\esp-idf-v5.3.2\tools\ldgen\ldgen.py", line 133, in main
    sections_infos.add_sections_info(dump)
  File "D:\ESP32\Espressif\frameworks\esp-idf-v5.3.2\tools\ldgen\ldgen\entity.py", line 126, in add_sections_info
    raise ParseException('Parsing sections info for library ' + sections_info_dump.name + ' failed. ' + p.msg)
pyparsing.exceptions.ParseException: Parsing sections info for library F:/ZZJ/zw/Firmware/zw-host/build/esp-idf/esp_app_format/libesp_app_format.a failed. Expected 'In archive'  (at char 0), (line:1, col:1)
Batch file failed at line 3 with errorcode 1
[2/9] C:\WINDOWS\system32\cmd.exe /C "cd /D F:\ZZJ\zw\Firmware\zw-host\build\bootloader && D:\ESP32\Espressif\tools\cmake\3.30.2\bin\cmake.exe -DSDKCONFIG=F:/ZZJ/zw/Firmware/zw-host/sdkconfig -DIDF_PATH=D:/ESP32/Espressif/frameworks/esp-idf-v5.3.2 -DIDF_TARGET=esp32s3 -DPYTHON_DEPS_CHECKED=1 -DPYTHON=D:/ESP32/Espressif/python_env/idf5.3_py3.11_env/Scripts/python.exe -DEXTRA_COMPONENT_DIRS=D:/ESP32/Espressif/frameworks/esp-idf-v5.3.2/components/bootloader -DPROJECT_SOURCE_DIR=F:/ZZJ/zw/Firmware/zw-host -DIGNORE_EXTRA_COMPONENT= -GNinja -S D:/ESP32/Espressif/frameworks/esp-idf-v5.3.2/components/bootloader/subproject -B F:/ZZJ/zw/Firmware/zw-host/build/bootloader && D:\ESP32\Espressif\tools\cmake\3.30.2\bin\cmake.exe -E touch F:/ZZJ/zw/Firmware/zw-host/build/bootloader-prefix/src/bootloader-stamp/bootloader-configure"
FAILED: bootloader-prefix/src/bootloader-stamp/bootloader-configure F:/ZZJ/zw/Firmware/zw-host/build/bootloader-prefix/src/bootloader-stamp/bootloader-configure
C:\WINDOWS\system32\cmd.exe /C "cd /D F:\ZZJ\zw\Firmware\zw-host\build\bootloader && D:\ESP32\Espressif\tools\cmake\3.30.2\bin\cmake.exe -DSDKCONFIG=F:/ZZJ/zw/Firmware/zw-host/sdkconfig -DIDF_PATH=D:/ESP32/Espressif/frameworks/esp-idf-v5.3.2 -DIDF_TARGET=esp32s3 -DPYTHON_DEPS_CHECKED=1 -DPYTHON=D:/ESP32/Espressif/python_env/idf5.3_py3.11_env/Scripts/python.exe -DEXTRA_COMPONENT_DIRS=D:/ESP32/Espressif/frameworks/esp-idf-v5.3.2/components/bootloader -DPROJECT_SOURCE_DIR=F:/ZZJ/zw/Firmware/zw-host -DIGNORE_EXTRA_COMPONENT= -GNinja -S D:/ESP32/Espressif/frameworks/esp-idf-v5.3.2/components/bootloader/subproject -B F:/ZZJ/zw/Firmware/zw-host/build/bootloader && D:\ESP32\Espressif\tools\cmake\3.30.2\bin\cmake.exe -E touch F:/ZZJ/zw/Firmware/zw-host/build/bootloader-prefix/src/bootloader-stamp/bootloader-configure"
-- git rev-parse returned 'fatal: detected dubious ownership in repository at 'D:/ESP32/Espressif/frameworks/esp-idf-v5.3.2'
'D:/ESP32/Espressif/frameworks/esp-idf-v5.3.2' is owned by:
        BUILTIN/Administrators (S-1-5-32-544)
but the current user is:
        DESKTOP-2AN6RV4/PC (S-1-5-21-138969761-3891255532-4185153907-1001)
To add an exception for this directory, call:

        git config --global --add safe.directory D:/ESP32/Espressif/frameworks/esp-idf-v5.3.2'
[... cut ...]
    raise ParseException('Parsing sections info for library ' + sections_info_dump.name + ' failed. ' + p.msg)
pyparsing.exceptions.ParseException: Parsing sections info for library F:/ZZJ/zw/Firmware/zw-host/build/esp-idf/esp_common/libesp_common.a failed. Expected 'In archive'  (at char 0), (line:1, col:1)
Batch file failed at line 3 with errorcode 1
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the F:\ZZJ\zw\Firmware\zw-host\build\log\idf_py_stderr_output_20152 and F:\ZZJ\zw\Firmware\zw-host\build\log\idf_py_stdout_output_20152
```

AMAZONGSANG, 2025-02-19:
Oh！I think of one thing, after using the Windows11 installation assistant, I waited a few days to compile, is this problem related to this step? If so, how do I restore to the original system environment?

AMAZONGSANG, 2025-02-19:
I'm sorry I ordered it wrong

fhrbata, 2025-02-19:
Hello @AMAZONGSANG ,

> I have learned about the #10440 issue, but unfortunately I don't understand it very well, and it is mac, I am not sure how to solve it on Windows

Yes, it was reported for Mac, but the issue seems quite similar. The problem appears to be that `ldgen.py` cannot parse the output from objdump, although I might have overlooked something.

Thank you for the output. I'm sorry, I overlooked that CMake on Windows might wrap the command in a batch file, possibly due to command line length limitations. Therefore, the full ldgen command should be in the `sections.ld-d65dc84.bat` file. Could you please either attach the file `F:/ZZJ/zw/Firmware/zw-host/build/esp-idf/esp_system/ld/sections.ld esp-idf\esp_system\CMakeFiles\sections.ld-d65dc84.bat` or paste its contents here?

> Oh！I think of one thing, after using the Windows11 installation assistant, I waited a few days to compile, is this problem related to this step? If so, how do I restore to the original system environment?

For Windows, I would recommend using either the online or offline installer as described here https://docs.espressif.com/projects/esp-idf/en/stable/esp32/get-started/windows-setup.html. This should set up everything required for the ESP-IDF environment. I suspect there might be an issue with the ESP-IDF installation or environment.

Thank you

AMAZONGSANG, 2025-02-19:
Hello @fhrbata  ,
thank you for your kindness,Here are my files **sections.ld**
```
/*

 * SPDX-FileCopyrightText: 2021 Espressif Systems (Shanghai) CO LTD

 *

 * SPDX-License-Identifier: Apache-2.0

 */
/*

 * SPDX-FileCopyrightText: 2021-2024 Espressif Systems (Shanghai) CO LTD

 *

 * SPDX-License-Identifier: Apache-2.0

 */
/*

 * Automatically generated file. DO NOT EDIT.

 * Espressif IoT Development Framework (ESP-IDF) 5.3.2 Configuration Header

 */
       
/* List of deprecated options */
/* CPU instruction prefetch padding size for flash mmap scenario */
/*

 * PMP region granularity size

 * Software may determine the PMP granularity by writing zero to pmp0cfg, then writing all ones

 * to pmpaddr0, then reading back pmpaddr0. If G is the index of the least-significant bit set,

 * the PMP granularity is 2^G+2 bytes.

 */
/* CPU instruction prefetch padding size for memory protection scenario */
/* Memory alignment size for PMS */
    /* rtc timer data (s_rtc_timer_retain_mem, see esp_clk.c files). For rtc_timer_data_in_rtc_mem section. */
/* Default entry point */
ENTRY(call_start_cpu0);
SECTIONS
{
  /**

   * RTC fast memory holds RTC wake stub code,

   * including from any source file named rtc_wake_stub*.c

   */
  .rtc.text :
  {
    
 . = ALIGN(4); 
 _rtc_fast_start = ABSOLUTE(.);
    
 . = ALIGN(4); 
 _rtc_text_start = ABSOLUTE(.);
    *(.rtc.entry.literal .rtc.entry.text)
    mapping[rtc_text]
    *rtc_wake_stub*.*(.literal .text .literal.* .text.*)
    *(.rtc_text_end_test)
    /* Padding for possible CPU prefetch + alignment for PMS split lines */
    . += 16;
[... 591 lines cut ...]
    found_sections = sections_db.get_sections(self.parent.name, self.name)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\ESP32\Espressif\frameworks\esp-idf-v5.3.2\tools\ldgen\ldgen\entity.py", line 198, in get_sections
    obj = self._match_obj(archive, obj)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\ESP32\Espressif\frameworks\esp-idf-v5.3.2\tools\ldgen\ldgen\entity.py", line 183, in _match_obj
    objs = self.get_objects(archive)
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\ESP32\Espressif\frameworks\esp-idf-v5.3.2\tools\ldgen\ldgen\entity.py", line 176, in get_objects
    self._process_archive(archive)
  File "D:\ESP32\Espressif\frameworks\esp-idf-v5.3.2\tools\ldgen\ldgen\entity.py", line 164, in _process_archive
    parsed = self._get_infos_from_file(stored)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\ESP32\Espressif\frameworks\esp-idf-v5.3.2\tools\ldgen\ldgen\entity.py", line 154, in _get_infos_from_file
    raise ParseException('Unable to parse section info file ' + info.filename + '. ' + p.msg)
pyparsing.exceptions.ParseException: Unable to parse section info file F:/ZZJ/zw/Firmware/zw-host/build/esp-idf/log/liblog.a. Expected end of text  (at char 0), (line:1, col:1)
Batch file failed at line 3 with errorcode 1

[已退出进程，代码为 1 (0x00000001)]
现在可以使用Ctrl+D关闭此终端，或按 Enter 重新启动。
```

我的环境配置就是根据官网配置来的，但是不能进行编译
My environment configuration is based on the official website configuration, but can not be compiled
 it troubles me too much.

fhrbata, 2025-02-19:
Hello @AMAZONGSANG ,

thank you for the additional information. It seems that you posted the output of the `sections.ld-d65dc84.bat`, not the actual content of the file. I rebooted to windows to take a closer look and tried with the `hello_world` example. For illustration here is how my ` build\esp-idf\esp_system\CMakeFiles\sections.ld-d65dc84.bat` looks like for the `hello_world` example. It shows how the `ldgen.py` is invoked and what `objdump` command is passed to it with the ` --objdump` option.

```bash
C:\Users\FrantisekHrbata\work\esp-idf\examples\get-started\hello_world>cat build\esp-idf\esp_system\CMakeFiles\sections.ld-d65dc84.bat
@echo off
cd /D C:\Users\FrantisekHrbata\work\esp-idf\examples\get-started\hello_world\build\esp-idf\esp_system || (set FAIL_LINE=2& goto :ABORT)
C:\Espressif\python_env\idf5.5_py3.11_env\Scripts\python.exe C:/Users/FrantisekHrbata/work/esp-idf/tools/ldgen/ldgen.py --config C:/Users/FrantisekHrbata/work/esp-idf/examples/get-started/hello_world/sdkconfig --fragments-list C:/Users/
FrantisekHrbata/work/esp-idf/components/xtensa/linker.lf;C:/Users/FrantisekHrbata/work/esp-idf/components/esp_driver_gpio/linker.lf;C:/Users/FrantisekHrbata/work/esp-idf/components/esp_pm/linker.lf;C:/Users/FrantisekHrbata/work/esp-idf/
components/esp_mm/linker.lf;C:/Users/FrantisekHrbata/work/esp-idf/components/spi_flash/linker.lf;C:/Users/FrantisekHrbata/work/esp-idf/components/esp_system/linker.lf;C:/Users/FrantisekHrbata/work/esp-idf/components/esp_system/app.lf;C:
/Users/FrantisekHrbata/work/esp-idf/components/esp_common/common.lf;C:/Users/FrantisekHrbata/work/esp-idf/components/esp_common/soc.lf;C:/Users/FrantisekHrbata/work/esp-idf/components/esp_rom/linker.lf;C:/Users/FrantisekHrbata/work/esp-
idf/components/hal/linker.lf;C:/Users/FrantisekHrbata/work/esp-idf/components/log/linker.lf;C:/Users/FrantisekHrbata/work/esp-idf/components/heap/linker.lf;C:/Users/FrantisekHrbata/work/esp-idf/components/soc/linker.lf;C:/Users/Frantise
kHrbata/work/esp-idf/components/esp_hw_support/linker.lf;C:/Users/FrantisekHrbata/work/esp-idf/components/esp_hw_support/dma/linker.lf;C:/Users/FrantisekHrbata/work/esp-idf/components/esp_hw_support/ldo/linker.lf;C:/Users/FrantisekHrbat
a/work/esp-idf/components/esp_hw_support/mspi_timing_tuning/linker.lf;C:/Users/FrantisekHrbata/work/esp-idf/components/freertos/linker_common.lf;C:/Users/FrantisekHrbata/work/esp-idf/components/freertos/linker.lf;C:/Users/FrantisekHrbat
a/work/esp-idf/components/newlib/src/newlib.lf;C:/Users/FrantisekHrbata/work/esp-idf/components/newlib/src/system_libs.lf;C:/Users/FrantisekHrbata/work/esp-idf/components/esp_driver_gptimer/linker.lf;C:/Users/FrantisekHrbata/work/esp-id
f/components/esp_ringbuf/linker.lf;C:/Users/FrantisekHrbata/work/esp-idf/components/esp_driver_uart/linker.lf;C:/Users/FrantisekHrbata/work/esp-idf/components/app_trace/linker.lf;C:/Users/FrantisekHrbata/work/esp-idf/components/esp_even
t/linker.lf;C:/Users/FrantisekHrbata/work/esp-idf/components/esp_phy/linker.lf;C:/Users/FrantisekHrbata/work/esp-idf/components/esp_driver_usb_serial_jtag/linker.lf;C:/Users/FrantisekHrbata/work/esp-idf/components/vfs/linker.lf;C:/Users
/FrantisekHrbata/work/esp-idf/components/lwip/linker.lf;C:/Users/FrantisekHrbata/work/esp-idf/components/esp_netif/linker.lf;C:/Users/FrantisekHrbata/work/esp-idf/components/wpa_supplicant/linker.lf;C:/Users/FrantisekHrbata/work/esp-idf
/components/esp_wifi/linker.lf;C:/Users/FrantisekHrbata/work/esp-idf/components/esp_driver_spi/linker.lf;C:/Users/FrantisekHrbata/work/esp-idf/components/esp_driver_pcnt/linker.lf;C:/Users/FrantisekHrbata/work/esp-idf/components/esp_dri
ver_mcpwm/linker.lf;C:/Users/FrantisekHrbata/work/esp-idf/components/esp_driver_ana_cmpr/linker.lf;C:/Users/FrantisekHrbata/work/esp-idf/components/esp_driver_dac/linker.lf;C:/Users/FrantisekHrbata/work/esp-idf/components/esp_driver_rmt
/linker.lf;C:/Users/FrantisekHrbata/work/esp-idf/components/esp_driver_sdm/linker.lf;C:/Users/FrantisekHrbata/work/esp-idf/components/esp_driver_i2c/linker.lf;C:/Users/FrantisekHrbata/work/esp-idf/components/esp_driver_ledc/linker.lf;C:
/Users/FrantisekHrbata/work/esp-idf/components/esp_driver_parlio/linker.lf;C:/Users/FrantisekHrbata/work/esp-idf/components/driver/twai/linker.lf;C:/Users/FrantisekHrbata/work/esp-idf/components/esp_adc/linker.lf;C:/Users/FrantisekHrbat
a/work/esp-idf/components/esp_driver_isp/linker.lf;C:/Users/FrantisekHrbata/work/esp-idf/components/esp_eth/linker.lf;C:/Users/FrantisekHrbata/work/esp-idf/components/esp_gdbstub/linker.lf;C:/Users/FrantisekHrbata/work/esp-idf/component
s/esp_psram/linker.lf;C:/Users/FrantisekHrbata/work/esp-idf/components/esp_lcd/linker.lf;C:/Users/FrantisekHrbata/work/esp-idf/components/espcoredump/linker.lf;C:/Users/FrantisekHrbata/work/esp-idf/components/ieee802154/linker.lf;C:/Use
rs/FrantisekHrbata/work/esp-idf/components/openthread/linker.lf --input C:/Users/FrantisekHrbata/work/esp-idf/examples/get-started/hello_world/build/esp-idf/esp_system/ld/sections.ld.in --output C:/Users/FrantisekHrbata/work/esp-idf/exa
mples/get-started/hello_world/build/esp-idf/esp_system/ld/sections.ld --kconfig C:/Users/FrantisekHrbata/work/esp-idf/Kconfig --env-file C:/Users/FrantisekHrbata/work/esp-idf/examples/get-started/hello_world/build/config.env --libraries
-file C:/Users/FrantisekHrbata/work/esp-idf/examples/get-started/hello_world/build/ldgen_libraries --objdump C:/Espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/xtensa-esp32-elf-obj [... cut ...]
There also seem to be another problem with the esp-idf repository ownership. Following is also from one of outputs you posted.

> fatal: detected dubious ownership in repository at 'D:/ESP32/Espressif/frameworks/esp-idf-v5.3.2'
'D:/ESP32/Espressif/frameworks/esp-idf-v5.3.2' is owned by:
        BUILTIN/Administrators (S-1-5-32-544)
but the current user is:
        DESKTOP-2AN6RV4/PC (S-1-5-21-138969761-3891255532-4185153907-1001)
To add an exception for this directory, call:
>
>        git config --global --add safe.directory D:/ESP32/Espressif/frameworks/esp-idf-v5.3.2 

Thank you

AMAZONGSANG, 2025-02-19:
Hello @fhrbata ,
Thank you for your correction，I think this is what you want
```
PS F:\ZZJ\zw\Firmware\zw-host> cat build\esp-idf\esp_system\CMakeFiles\sections.ld-d65dc84.bat
@echo off
cd /D F:\ZZJ\zw\Firmware\zw-host\build\esp-idf\esp_system || (set FAIL_LINE=2& goto :ABORT)
D:\ESP32\Espressif\python_env\idf5.3_py3.11_env\Scripts\python.exe D:/ESP32/Espressif/frameworks/esp-idf-v5.3.2/tools/ldgen/ldgen.py --config F:/ZZJ/zw/Firmware/zw-host/sdkconfig --fragments-list D:/ESP32/Espressif/frameworks/esp-idf-v5.3.2/components/xtensa/linker.lf;D:/ESP32/Espressif/frameworks/esp-idf-v5.3.2/components/esp_driver_gpio/linker.lf;D:/ESP32/Espressif/frameworks/esp-idf-v5.3.2/components/esp_pm/linker.lf;D:/ESP32/Espressif/frameworks/esp-idf-v5.3.2/components/esp_mm/linker.lf;D:/ESP32/Espressif/frameworks/esp-idf-v5.3.2/components/spi_flash/linker.lf;D:/ESP32/Espressif/frameworks/esp-idf-v5.3.2/components/esp_system/linker.lf;D:/ESP32/Espressif/frameworks/esp-idf-v5.3.2/components/esp_system/app.lf;D:/ESP32/Espressif/frameworks/esp-idf-v5.3.2/components/esp_common/common.lf;D:/ESP32/Espressif/frameworks/esp-idf-v5.3.2/components/esp_common/soc.lf;D:/ESP32/Espressif/frameworks/esp-idf-v5.3.2/components/esp_rom/linker.lf;D:/ESP32/Espressif/frameworks/esp-idf-v5.3.2/components/hal/linker.lf;D:/ESP32/Espressif/frameworks/esp-idf-v5.3.2/components/log/linker.lf;D:/ESP32/Espressif/frameworks/esp-idf-v5.3.2/components/heap/linker.lf;D:/ESP32/Espressif/frameworks/esp-idf-v5.3.2/components/soc/linker.lf;D:/ESP32/Espressif/frameworks/esp-idf-v5.3.2/components/esp_hw_support/linker.lf;D:/ESP32/Espressif/frameworks/esp-idf-v5.3.2/components/esp_hw_support/dma/linker.lf;D:/ESP32/Espressif/frameworks/esp-idf-v5.3.2/components/esp_hw_support/ldo/linker.lf;D:/ESP32/Espressif/frameworks/esp-idf-v5.3.2/components/freertos/linker_common.lf;D:/ESP32/Espressif/frameworks/esp-idf-v5.3.2/components/freertos/linker.lf;D:/ESP32/Espressif/frameworks/esp-idf-v5.3.2/components/newlib/newlib.lf;D:/ESP32/Espressif/frameworks/esp-idf-v5.3.2/components/newlib/system_libs.lf;D:/ESP32/Espressif/frameworks/esp-idf-v5.3.2/components/esp_driver_gptimer/linker.lf;D:/ESP32/Espressif/frameworks/esp-idf-v5.3.2/components/esp_ringbuf/linker.lf;D:/ESP32/Espressif/frameworks/esp-idf-v5.3.2/components/esp_driver_uart/linker.lf;D:/ESP32/Espressif/frameworks/esp-idf-v5.3.2/components/app_trace/linker.lf;D:/ESP32/Espressif/frameworks/esp-idf-v5.3.2/components/esp_event/linker.lf;D:/ESP32/Espressif/frameworks/esp-idf-v5.3.2/components/esp_driver_pcnt/linker.lf;D:/ESP32/Espressif/frameworks/esp-idf-v5.3.2/components/esp_driver_spi/linker.lf;D:/ESP32/Espressif/frameworks/esp-idf-v5.3.2/components/esp_driver_mcpwm/linker.lf;D:/ESP32/Espressif/frameworks/esp-idf-v5.3.2/components/esp_driver_ana_cmpr/linker.lf;D:/ESP32/Espressif/frameworks/esp-idf-v5.3.2/components/esp_driver_dac/linker.lf;D:/ESP32/Espressif/frameworks/esp-idf-v5.3.2/components/esp_driver_rmt/linker.lf;D:/ESP32/Espressif/frameworks/esp-idf-v5.3.2/components/esp_driver_sdm/linker.lf;D:/ESP32/Espressif/frameworks/esp-idf-v5.3.2/components/esp_driver_i2c/linker.lf;D:/ESP32/Espressif/frameworks/esp-idf-v5.3.2/components/esp_driver_ledc/linker.lf;D:/ESP32/Espressif/frameworks/esp-idf-v5.3.2/components/esp_driver_parlio/linker.lf;D:/ESP32/Espressif/frameworks/esp-idf-v5.3.2/components/driver/twai/linker.lf;D:/ESP32/Espressif/frameworks/esp-idf-v5.3.2/components/esp_phy/linker.lf;D:/ESP32/Espressif/frameworks/esp-idf-v5.3.2/components/vfs/linker.lf;D:/ESP32/Espressif/frameworks/esp-idf-v5.3.2/components/lwip/linker.lf;D:/ESP32/Espressif/frameworks/esp-idf-v5.3.2/components/esp_netif/linker.lf;D:/ESP32/Espressif/frameworks/esp-idf-v5.3.2/components/wpa_supplicant/linker.lf;D:/ESP32/Espressif/frameworks/esp-idf-v5.3.2/components/esp_wifi/linker.lf;D:/ESP32/Espressif/frameworks/esp-idf-v5.3.2/components/esp_adc/linker.lf;D:/ESP32/Espressif/frameworks/esp-idf-v5.3.2/components/esp_eth/linker.lf;D:/ESP32/Espressif/frameworks/esp-idf-v5.3.2/components/esp_gdbstub/linker.lf;D:/ESP32/Espressif/frameworks/esp-idf-v5.3.2/components/esp_psram/linker.lf;D:/ESP32/Espressif/frameworks/esp-idf-v5.3.2/components/esp_lcd/linker.lf;D:/ESP32/Espressif/frameworks/esp-idf-v5.3.2/components/espcoredump/linker.lf;D:/ESP32/Espressif/frameworks/esp-idf-v5.3.2/components/ieee802154/linker.lf;D:/ESP32/Espressif/frameworks/esp-idf-v5.3.2/components/openthread/linker.lf --input F:/ZZJ/zw/Firmware/zw-host/build/esp-idf/esp_system/ld/sections.ld.in --output F:/ZZJ/zw/Firmware/zw-host/build/esp-idf/esp_system/ld/sections.ld --kconfig D:/ESP32/Espressif/frameworks/esp-idf-v5.3.2/Kconfig --env-file F:/ZZJ/zw/Firmware/zw-host/build/config.env --libraries-file F:/ZZJ/zw/Firmware/zw-host/build/ldgen_libraries --objdump D:/ESP32/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/xtensa-esp32s3-elf-objdump.exe || (set FAIL_LINE=3& goto :ABORT)
goto :EOF

:ABORT
set ERROR_CODE=%ERRORLEVEL%
echo Batch file failed at line %FAIL_LINE% with errorcode %ERRORLEVEL%
exit /b %ERROR_CODE%
```
I installed the fourth package that appears on the following page, which is 5.3.2 -Offline Installer
`https://[dl.espressif.com.cn/dl/esp-idf/?idf=4.4](https://dl.espressif.com.cn/dl/esp-idf/?idf=4.4)`

After I execute this this instruction
```
D:\ESP32\Espressif\frameworks\esp-idf-v5.3.2\install.ps1
D:\ESP32\Espressif\frameworks\esp-idf-v5.3.2\export.ps1
```
The following log appears
```
[... cut ...]
For removing old versions of idf-driver, idf-python-wheels use command 'python.exe D:\ESP32\Espressif\frameworks\esp-idf-v5.3.2\tools\idf_tools.py uninstall'
For free up even more space, remove installation packages of those tools. Use option 'python.exe D:\ESP32\Espressif\frameworks\esp-idf-v5.3.2\tools\idf_tools.py uninstall --remove-archives'.

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:
    idf.py build
```
But I still get an error when I compile again
At the same time I ran instructions in Git Bash，The build result is the same

fhrbata, 2025-02-19:
Hello @AMAZONGSANG,

thank you for providing the additional information and assistance as we work to understand the issue.

It appears that `ldgen.py` is having trouble parsing the `objdump` output. The error message from your output:
```
pyparsing.exceptions.ParseException: Parsing sections info for library F:/ZZJ/zw/Firmware/zw-host/build/esp-idf/esp_common/libesp_common.a failed. Expected 'In archive'  (at char 0), (line:1, col:1)
```

Could you please run the following command to ensure the binary functions correctly in your environment and share the output?
```
D:/ESP32/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/xtensa-esp32s3-elf-objdump.exe --version
```

Additionally, please try this command. It might help us understand what `objdump` output is passed to `ldgen` for parsing:
```
D:/ESP32/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/xtensa-esp32s3-elf-objdump.exe -h F:/ZZJ/zw/Firmware/zw-host/build/esp-idf/esp_common/libesp_common.a
```

Thank you.

AMAZONGSANG, 2025-02-19:
hello @fhrbata,
Thank you for taking time out of your busy schedule to reply me
Here are the results I got using two commands
```
PS F:\ZZJ\zw\Firmware\zw-host> D:/ESP32/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/xtensa-esp32s3-elf-objdump.exe --version
GNU objdump (crosstool-NG esp-13.2.0_20240530) 2.41
Copyright (C) 2023 Free Software Foundation, Inc.
This program is free software; you may redistribute it under the terms of
the GNU General Public License version 3 or (at your option) any later version.
This program has absolutely no warranty.
PS F:\ZZJ\zw\Firmware\zw-host> D:/ESP32/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/xtensa-esp32s3-elf-objdump.exe -h F:/ZZJ/zw/Firmware/zw-host/build/esp-idf/esp_common/libesp_common.a
In archive F:/ZZJ/zw/Firmware/zw-host/build/esp-idf/esp_common/libesp_common.a:

esp_err_to_name.c.obj:     file format elf32-xtensa-le

Sections:
Idx Name          Size      VMA       LMA       File off  Algn
  0 .literal.esp_err_to_name 00000008  00000000  00000000  00000034  2**2
                  CONTENTS, ALLOC, LOAD, RELOC, READONLY, CODE
  1 .literal.esp_err_to_name_r 00000018  00000000  00000000  0000003c  2**2
                  CONTENTS, ALLOC, LOAD, RELOC, READONLY, CODE
  2 .text         00000000  00000000  00000000  00000054  2**0
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
  3 .data         00000000  00000000  00000000  00000054  2**0
                  CONTENTS, ALLOC, LOAD, DATA
  4 .bss          00000000  00000000  00000000  00000054  2**0
                  ALLOC
  5 .text.esp_err_to_name 00000027  00000000  00000000  00000054  2**2
                  CONTENTS, ALLOC, LOAD, RELOC, READONLY, CODE
  6 .rodata.esp_err_to_name_r.str1.4 0000000c  00000000  00000000  0000007c  2**2
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
  7 .text.esp_err_to_name_r 00000056  00000000  00000000  00000088  2**2
                  CONTENTS, ALLOC, LOAD, RELOC, READONLY, CODE
  8 .rodata.esp_unknown_msg 00000006  00000000  00000000  000000e0  2**2
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
  9 .rodata.str1.4 0000180a  00000000  00000000  000000e8  2**2
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
 10 .rodata.esp_err_msg_table 000006d0  00000000  00000000  000018f4  2**2
                  CONTENTS, ALLOC, LOAD, RELOC, READONLY, DATA
 11 .debug_frame  00000040  00000000  00000000  00001fc4  2**2
                  CONTENTS, RELOC, READONLY, DEBUGGING, OCTETS
 12 .debug_info   000003b8  00000000  00000000  00002004  2**0
                  CONTENTS, RELOC, READONLY, DEBUGGING, OCTETS
 13 .debug_abbrev 000001d9  00000000  00000000  000023bc  2**0
                  CONTENTS, READONLY, DEBUGGING, OCTETS
 14 .debug_loc    000000f2  00000000  00000000  00002595  2**0
                  CONTENTS, RELOC, READONLY, DEBUGGING, OCTETS
 15 .debug_aranges 00000028  00000000  00000000  00002687  2**0
                  CONTENTS, RELOC, READONLY, DEBUGGING, OCTETS
 16 .debug_ranges 00000018  00000000  00000000  000026af  2**0
                  CONTENTS, RELOC, READONLY, DEBUGGING, OCTETS
 17 .debug_line   000004ad  00000000  00000000  000026c7  2**0
                  CONTENTS, RELOC, READONLY, DEBUGGING, OCTETS
 18 .debug_str    000004f5  00000000  00000000  00002b74  2**0
                  CONTENTS, READONLY, DEBUGGING, OCTETS
 19 .comment      00000030  00000000  00000000  00003069  2**0
                  CONTENTS, READONLY
 20 .xtensa.info  00000038  00000000  00000000  00003099  2**0
                  CONTENTS, READONLY
 21 .xt.lit       00000010  00000000  00000000  000030d1  2**0
                  CONTENTS, RELOC, READONLY
 22 .xt.prop      000000cc  00000000  00000000  000030e1  2**0
                  CONTENTS, RELOC, READONLY
```

Thank you

fhrbata, 2025-02-20:
Hello @AMAZONGSANG ,

> Thank you for taking time out of your busy schedule to reply me

You're very welcome! Though, unfortunately, I haven't been able to provide anything truly helpful for your issue so far. 😞

> Here are the results I got using two commands

To be honest, I wasn't expecting this output since it looks correct. I was hoping to see something unusual that could help explain the behavior you're experiencing.

Based on your outputs, the failure doesn’t seem to be specific to a single library (libesp_common.a, libesp_app_format.a, liblog.a).

```
pyparsing.exceptions.ParseException: Expected 'In archive'  (at char 0), (line:1, col:1)
pyparsing.exceptions.ParseException: Parsing sections info for library F:/ZZJ/zw/Firmware/zw-host/build/esp-idf/esp_common/libesp_common.a failed. Expected 'In archive'  (at char 0), (line:1, col:1)
```
```
pyparsing.exceptions.ParseException: Expected 'In archive'  (at char 0), (line:1, col:1)
pyparsing.exceptions.ParseException: Parsing sections info for library F:/ZZJ/zw/Firmware/zw-host/build/esp-idf/esp_app_format/libesp_app_format.a failed. Expected 'In archive'  (at char 0), (line:1, col:1)
```
The two snippets above, IIUC, mean that there was no output from the objdump passed to ldgen. Strange thing is that the objdump has not failed. If there was something wrong we would see different error like command not found, invalid argument(non-existing, empty or invalid format of the library). It just for some reason returns empty output or it's somehow discarded in the bat file or python, which I think is unlikely.

The following snipped means that there was some output, but ldgen was not able to parse it. The section parsing failed.
```
pyparsing.exceptions.ParseException: Expected end of text, found 'log'  (at char 8231), (line:138, col:1)
pyparsing.exceptions.ParseException: Unable to parse section info file F:/ZZJ/zw/Firmware/zw-host/build/esp-idf/log/liblog.a. Expected end of text  (at char 0), (line:1, col:1)
```

I tried to make sense of this, but I'm not sure how it could have happened. I guess there has to be something wrong with the installation. It's hard to say.

> Tool doesn't match supported version from list ['esp-13.2.0_20240530']:
>  D:/ESP32/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc.exe

This means that the `xtensa-esp32s3-elf-gcc.exe --version`  returns different version than supported for the activated esp-idf(`esp-13.2.0_20240530` as defined in `tools\tools.json`) . Which is strange, because it is located in the `esp-13.2.0_20240530` directory. It seems there might be some  mismatch with toolchains.

I guess if this was a generic problem with ldgen, there would be more issues reported for this.

I would recommend a fresh ESP-IDF installation to rule out any issues that might be caused by a broken installation.
I installed v5.3.2 using both the online and offline installers on Windows 10.0.19045, and everything worked fine. I followed the default installation process but changed the destination directories for ESP-IDF and tools to ensure a completely fresh installation without any potential conflicts with existing setups. After installation, I used the provided PowerShell terminal, where ESP-IDF was activated, and successfully compiled hello_world for the ESP32-S3 target. You might want to try the same approach to see if it helps.

> Oh！I think of one thing, after using the Windows11 installation assistant, I waited a few days to compile, is this problem related to this step? If so, how do I restore to the original system environment?

I'm not sure. I suppose it's possible but unlikely—but I'm not a Windows user and my experience with it is very limited.

Thank you very much for all the information you've provided. You can try a fresh install as described above to see if it helps. Apologies if I missed anything. Perhaps others may have additional insights into what might be wrong.

fhrbata, 2025-02-21:
Few more notes, if you try the fresh install, please make sure you run `idf.py fullclean`. If it fails, delete the `build` directory manually. Maybe you can also try to start the build without ccache with `idf.py --no-ccache build`

AMAZONGSANG, 2025-02-21:
Hello @fhrbata,
Thank you very much for your questions and answers during this time. I'm glad to see you. I reinstalled ESP-IDF three or four times, but it didn't work, and now I've decided to reinstall WIN11 and pray it works

fhrbata, 2025-02-21:
Hello @AMAZONGSANG , thank you very much for all the help and information you have provided. It's much appreciated. Please let me know if the problem persists. Thank you

fhrbata, 2025-03-26:
Hello @AMAZONGSANG , I just want to follow-up on this issue. We got another report, which I believe is the same issue as you described here. It seems that the encryption software https://www.drhchina.com/ may be causing such problems. Have you, by any chance, have this installed on your machine or maybe other encryption software? Thank you