espressif/esp-idf #15420: Unable to combine COREDUMP_RTC_DATA_ATTR and RTC_NOINIT_ATTR

ammaree, 2025-02-18:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

We have been using RTC_NOINIT_ATTR successfully to preserve the contents of an STDOUT buffer in RTC slow RAM across reboots. This buffer allows us to view ESP_LOG messages generated just prior to a reboot or crash, or whilst a console was not connected, at a later stage.

We now would like to add this RTC slow RAM based STDOUT buffer to the COREDUMP image but cannot find a way to do this.

Since RTC_NOINIT_ATTR is primarily used during boot to disable the zero init of this memory area, and COREDUMP_RTC_DATA_ATTR is used at crash time to control memory area(s) included with the coredump info, theoretically we cannot see a reason why these 2 attributes cannot be combined.

Is there any way to achieve the effect of preserving the contents of this single memory block across reboots and also include it in coredumps?

erhankur, 2025-02-24:
@ammaree  Thanks for bringing this up. I am taking this as a feature request since I'm not sure if there's another way to achieve this. Instead of adding new section attribute, we are considering adding an API to allow users to place the desired memory regions into the coredump. I believe it will be helpful for you.

ammaree, 2026-01-19:
Thanks, appreciated

André M. Maree

erhankur, 2026-01-19:
Yes it will be backported to v5.5 and v6.0. 

v6.0 is locked. v5.5 is under review now.

ammaree, 2026-02-03:
On closer inspection this does not solve our requirement.
We require the memory to be:
a) in RTC, contents preserved across reboots; and
b) have NOINIT, for obvious reasons; and
c) be part of COREDUMP, so we can see the last messages before the crash from the buffer.

Any suggestion for how we achieve this?