espressif/esp-idf #15421: ESP32-C6 temperature sensor values are wrong

aspymid, 2025-02-18:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4

### Espressif SoC revision.

ESP32-C6

### Operating System used.

Linux

### How did you build your project?

Eclipse IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP-32-C6-WROM-1

### Power Supply used.

USB

### What is the expected behavior?

I expected to get correct temperature readings

### What is the actual behavior?

The values given by the API are at least 10 Centigrade too small``

### Steps to reproduce.

1. Eclipse with ESP-IDF plugin
2. Create a new IDF project and specify to use the exampe temp_sensor
3. Compile & load
4. Look at the terminal monitor and read the output  values.

Values expected ... a bit more than room temperature (22C)

Example:
I (240) main_task: Calling app_main()
I (240) example: Install temperature sensor, expected temp ranger range: 10~50 ℃
I (250) temperature_sensor: Range [-10°C ~ 80°C], error < 1°C
I (250) example: Enable temperature sensor
I (260) example: Read temperature
I (260) example: Temperature value 13.00 ℃
I (1270) example: Temperature value 13.00 ℃
I (2270) example: Temperature value 13.00 ℃
I (3270) example: Temperature value 13.00 ℃

### Debug Logs.

```plain

```

### More Information.

The same code runs well on an ESP32-S3 and ESP32-C3.

The example code and the documentation does not seem to differ in substance.

The eFuse variable s_deltaT is shown as zero if the driver debug is enabled.

mythbuster5, 2025-02-19:
`s_dealtaT` is 0 sounds the efuse has not been burnt. May you please check the efuse version via `uint32_t version = esp_efuse_rtc_calib_get_ver();` If the efuse burnt, the value should be correct. What I did on mt side is.

1) old C6 without efuse burnt:

```
version is 0
I (275) example: Temperature value 18.00 ℃
```

1) normal C6 without efuse burnt:

```
version is 1
I (254) example: Temperature value 21.40 ℃
I (1264) example: Temperature value 21.40 ℃
```

You may got a early stage C6 for some reason I guess

aspymid, 2025-02-19:
Surprise! The following: 
```
#include "esp_efuse_rtc_calib.h"
	uint32_t efuse_version = esp_efuse_rtc_calib_get_ver();
	ESP_LOGI(TAG, "eFuse version=%u", (unsigned int)efuse_version);
```

**returns value 2**

andylinpersonal, 2025-02-19:
Because in idf v5.4 e37d33cc1c9827fa8a29655827bce4fc4ae7c5f7, the required [function](https://github.com/espressif/esp-idf/blob/e37d33cc1c9827fa8a29655827bce4fc4ae7c5f7/components/efuse/esp32c6/esp_efuse_rtc_calib.c#L133) `esp_efuse_rtc_calib_get_tsens_val` in `efuse` component is still a stub.
The implementation in master branch 1160a86ba0b87b0ea68ad6b91eb10fe9c34ad692 should be okay. You can just copy master's function into your idf v5.4. I've tested it.

aspymid, 2025-02-19:
Yes,  the fix did work perfectly for me!

Thanks!

closed 2025-02-19 by aspymid