espressif/esp-idf #15422: Hard I2c have problem

NewNoob2002, 2025-02-19:
IDF version.
v5.3.2

Operating System used.
Linux

How did you build your project?
Command line with idf.py

If you are using Windows, please specify command line type.
None

Development Kit.
ESP32-WROOM-EU

Power Supply used.
Battery

What is the expected behavior?
should get the register value of I2c device 

My Init function and write function is here:

```
void I2c_Init()
{
    const i2c_master_bus_config_t bus_cfg = {
        .i2c_port = I2C_NUM_0,
        .sda_io_num = (gpio_num_t)BAT_I2C_SDA_Pin,
        .scl_io_num = (gpio_num_t)BAT_I2C_SCL_Pin,
        .clk_source = I2C_CLK_SRC_DEFAULT,
        .glitch_ignore_cnt = 7,
        .flags = {
            .enable_internal_pullup = true
          }
          };
  
    ESP_ERROR_CHECK(i2c_new_master_bus(&bus_cfg, &bus_handle));
  
    const i2c_device_config_t bat_device = {
        .dev_addr_length = I2C_ADDR_BIT_LEN_7,
        .device_address = 0x5c,
        .scl_speed_hz = 100000,
    };
  
    ESP_ERROR_CHECK(i2c_master_bus_add_device(bus_handle, &bat_device, &i2c_Bat_handle));
}

esp_err_t Write_BAT_Reg_Byte(uint8_t reg, const uint8_t value)
{
  uint8_t data[2] = {reg, value};
  esp_err_t error = i2c_master_transmit(i2c_Bat_handle, data, sizeof(data), 1000);
#ifdef _DEBUG_BAT
  if (error == ESP_OK)
  {
    ESP_LOGI("I2c BAT", "send success, reg : %d, value: %d", reg, value);
  }
  return error;
#endif // DEBUG
}
```

What is the actual behavior?
Upon running the example and opening the monitor using the idf.py script, I observe the following problem:

[image]

And i take DSLogicPlus to get sample. it' show below:

[image]

It can send Device Address normal, but can't get NACK, what's the problem.

And SDA and SCL have used a 4.7k ohm resistor for pull-up

[image]

NewNoob2002, 2025-02-19:
But the strange thing is that I successfully obtained the data of the I2C device register last week, but this week this problem occurred

nordlender, 2025-02-24:
1. What device are you trying to communicate with? I've seen some funny "i2c" devices. 
2. What are you connecting to your device with? Jumpers?
3. Do you have common ground between the devices?

NewNoob2002, 2025-02-25:
> 1. What device are you trying to communicate with? I've seen some funny "i2c" devices.
> 2. What are you connecting to your device with? Jumpers?
> 3. Do you have common ground between the devices?

Thanks for your reply, i have fixed this problem, which is because of the capacitance of dc_in

closed 2025-02-26 by mythbuster5