espressif/esp-idf #15423: esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x7780 on unmodified esp-idf demo

BrucePerens, 2025-02-19:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.4

### Espressif SoC revision.

ESP32-WROOM-32E-N4

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board, see https://github.com/VanceVagell/kv4p-ht

### Power Supply used.

USB

### What is the expected behavior?

The test software is the demo at esp-idf/examples/protocols/https_server/simple
The only modification is to set the SSID and WiFi password.

A browser should connect to the demo software without an error on the ESP-32 console.

### What is the actual behavior?

The TLS connection fails several times, and then succeeds. The pertinent lines of the log file are:

E (113675) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x7780
E (113685) esp_https_server: esp_tls_create_server_session failed, 0x7780
E (113685) example: Error event triggered: last_error = ESP_ERR_MBEDTLS_SSL_HANDSHAKE_FAILED, last_tls_err = 30592, tls_flag = 0
E (113685) httpd: httpd_accept_conn: session creation failed
W (113695) httpd: httpd_server: error accepting new connection
I (113705) esp_https_server: performing session handshake
E (114615) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x7780
E (114615) esp_https_server: esp_tls_create_server_session failed, 0x7780
E (114615) example: Error event triggered: last_error = ESP_ERR_MBEDTLS_SSL_HANDSHAKE_FAILED, last_tls_err = 30592, tls_flag = 0
E (114615) httpd: httpd_accept_conn: session creation failed
W (114635) httpd: httpd_server: error accepting new connection
I (114645) esp_https_server: performing session handshake
I (115785) esp_https_server: performing session handshake
W (117475) httpd_uri: httpd_uri: URI '/favicon.ico' not found
W (117475) httpd_txrx: httpd_resp_send_err: 404 Not Found - Nothing matches the given URI

### Steps to reproduce.

1. Build and flash the test software from The test software is the demo at esp-idf/examples/protocols/https_server/simple
2. Set the SSID and password for your test network.
3. Start the console with idf.py monitor
4. Use the Chrome or Firefox browser to 

### Debug Logs.

[... 132 lines cut ...]
E (113685) httpd: httpd_accept_conn: session creation failed
W (113695) httpd: httpd_server: error accepting new connection
I (113705) esp_https_server: performing session handshake
E (114615) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x7780
E (114615) esp_https_server: esp_tls_create_server_session failed, 0x7780
E (114615) example: Error event triggered: last_error = ESP_ERR_MBEDTLS_SSL_HANDSHAKE_FAILED, last_tls_err = 30592, tls_flag = 0
E (114615) httpd: httpd_accept_conn: session creation failed
W (114635) httpd: httpd_server: error accepting new connection
I (114645) esp_https_server: performing session handshake
I (115785) esp_https_server: performing session handshake
W (117475) httpd_uri: httpd_uri: URI '/favicon.ico' not found
W (117475) httpd_txrx: httpd_resp_send_err: 404 Not Found - Nothing matches the given URI
```

### More Information.

It looks like the browser sees a handshake failure and retries until it gets through. The error 0x7780 is from here:
#define MBEDTLS_ERR_SSL_FATAL_ALERT_MESSAGE               -0x7780
This doesn't really look like a clean negotiation of SSL parameters - it shouldn't result in that message if it was.
My attempt to debug it seems to show a failure in reading from the SSL client.
I'd like you to attempt to duplicate this at Espressif - you're a lot better at debugging your own software than I.

    Thanks

    Bruce

hrushikesh430, 2025-02-24:
Hi @BrucePerens, this issue is because of using the self singed certificate on server side, this is documented as well you can refer [README.md](https://github.com/espressif/esp-idf/tree/master/examples/protocols/https_server/simple#certificates). To resolve this issue either you can upload the self signed certificate in your browser like for chrome you can refer [this](https://www.pico.net/kb/how-do-you-get-chrome-to-accept-a-self-signed-certificate/) or use the trusted certificate.

hrushikesh430, 2025-03-10:
Hi @BrucePerens, closing this issue due to delay response. Please feel free to reopen the issue for the further queries.

closed 2025-03-10 by hrushikesh430

BrucePerens, 2025-03-12:
Hi @hrushikesh430,

Of course this issue showed up in my own software before I duplicated it
with the example software. Since this would happen in normal operation, is
there a way for me to suppress the message?

    Thanks

    Bruce