espressif/esp-idf #15424: Firmware crashing during Wi-Fi full calibration procedure

arasan90, 2025-02-19:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4

### Espressif SoC revision.

ESP32C61

### Operating System used.

Linux

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-C61-WROOM-1

### Power Supply used.

USB

### What is the expected behavior?

Wi-Fi performing the full calibration flow and then connecting to Wi-Fi network

### What is the actual behavior?

Firmware crashes during the full calibration procedure

### Steps to reproduce.

1. Use https://github.com/espressif/esp-idf/blob/release/v5.4/examples/wifi/getting_started/station/main/station_example_main.c
2. Hardcode the ssid and password values instead of using kconfig
3. Flash the firmware
4. Monitor the serial output

### Debug Logs.

```plain
ELF file SHA256: 7eec8d498

Rebooting...
ESP-ROM:esp32c61-eco2-20240903
Build:Sep  3 2024
rst:0xc (SW_CPU),boot:0xf (SPI_FAST_FLASH_BOOT)
Core0 Saved PC:0x4080085a
SPI mode:DIO, clock div:1
load:0x40845b80,len:0x176c
load:0x4083c570,len:0xc7c
load:0x4083ea70,len:0x3014
entry 0x4083c57a
I (27) boot: ESP-IDF v5.4-dirty 2nd stage bootloader
I (28) boot: compile time Feb 19 2025 10:19:05
I (29) boot: chip revision: v0.2
I (29) boot: efuse block revision: v0.1
I (31) boot.esp32c61: SPI Speed      : 80MHz
I (35) boot.esp32c61: SPI Mode       : DIO
[... 101 lines cut ...]
408212b0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
408212d0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
408212f0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
40821310: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
40821330: 0x00000000 0x00000054 0x40821338 0x40821338 0x40821338 0x40821338 0x00000000 0x40821350
40821350: 0xffffffff 0x40821350 0x40821350 0x00000000 0x40821364 0xffffffff 0x40821364 0x40821364
40821370: 0x00000000 0x00000001 0x00000000 0x7b00ffff 0x00000000 0x90116da9 0x40821330 0x00000ea0
40821390: 0x6f6d706f 0x00006564 0x2e617473 0x64697373 0x00000000 0x2e617473 0x68747561 0x65646f6d
408213b0: 0x00010000 0x00000000 0x00000004 0x00000002 0x40815940 0x2e617473 0x64697373 0x00000000
408213d0: 0x2e617473 0x68747561 0x65646f6d 0x00000000 0x2e617473 0x00240701 0x00000000 0x00000000
408213f0: 0x00000000 0x40815944 0x2e617473 0x68747561 0x65646f6d 0x00000000 0x2e617473 0x64777370
```

### More Information.

I tried disabling the brownout detection, using an external 5V voltage supplier.
I used the station example code in esp-idf SDK
I tried disabling the stack guard option.

In the end, this was the content of sdkconfig.defaults from latest attempt
CONFIG_ESPTOOLPY_FLASHSIZE_2MB=n
CONFIG_ESPTOOLPY_FLASHSIZE_8MB=y

I also tried with a ESP32-C3-DevkitM-1 development kit (changing the flash size to 4MB) and the same code works as expected on it.
The log reports v5.4-dirty because I changed the export.sh file to make it executable

QingzhaoYin, 2025-02-28:
@arasan90 Hi~, could you please provide the .elf file of this app ?

arasan90, 2025-03-12:
@QingzhaoYin  sorry for the late reply, I was sick and missed your comment.
It does not allow me to load a elf or 7z file.
How can I do that?

QingzhaoYin, 2025-03-12:
@arasan90  Could you please send this file to my email？yinqingzhao@espressif.com

arasan90, 2025-03-12:
@QingzhaoYin  elf file sent, thank you

QingzhaoYin, 2025-03-20:
@arasan90  Hi~, I could not find the email you sent to me. Could you please check and resend the file ? or, you can use a disassembly command `riscv32-esp-elf-objdump -D *.elf > output.txt ` under the build directory  to check where the crash occurred.

arasan90, 2025-03-25:
[output.zip](https://github.com/user-attachments/files/19446593/output.zip)

[tgs.zip](https://github.com/user-attachments/files/19446599/tgs.zip)

@QingzhaoYin  I noticed that zip files are accepted. I have attached both elf file and object dump output file

QingzhaoYin, 2025-03-31:
@arasan90 
Hi~ This issue occurs because the RF calibration data cannot be loaded from flash.

We need more information from you:

Does this issue happen on all ESP32-C61 modules, or just a specific one?
How many devices are powered by the hub at the same time, including the one experiencing the issue?
If you can still reproduce the issue in your current environment, could you package and provide the contents of the ~/esp-idf/examples/wifi/getting_started/station/build directory?
We can use your compiled files to try and reproduce the issue locally.

QingzhaoYin, 2025-04-07:
@arasan90 Or could you provide the bin file? We can download it locally for retesting. Also, could you please provide your code? This issue might require looking at the changes in the code.

arasan90, 2025-04-07:
[binaries.zip](https://github.com/user-attachments/files/19629987/binaries.zip)

In the zip file you will find both elf and bin files.
I used the code in the examples/wifi/getting-started/station folder, without modifications.
Via idf.py I changed the target to the esp32c61 and I added the ssid and password for the target network.

arasan90, 2025-04-30:
I just tested with latest master commit (38628f98b9000ded87827dccbd8a5a1827392cfb) and now it works.
I am able to connect to a network without hard faults.

Alvin1Zhang, 2025-05-06:
Thanks for reporting, glad to hear it works, feel free to reopen.

closed 2025-05-06 by Alvin1Zhang