espressif/esp-idf #15425: I2S macro definition order issue

williamleeag, 2025-02-19:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.5-dev-1942-g1160a86ba0b

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

build successfully as same as the old version v5.4

### What is the actual behavior?

build error,
error: designator order for field 'i2s_std_clk_config_t::ext_clk_freq_hz' does not match declaration order in 'i2s_std_clk_config_t'
   71 |     };
      |     ^
error: designator order for field 'i2s_std_clk_config_t::ext_clk_freq_hz' does not match declaration order in 'i2s_std_clk_config_t'
   88 |     };
      |     ^

### Steps to reproduce.

1. build
2. error found

### Build or installation Logs.

```plain

```

### More Information.

```
typedef struct {
    /* General fields */
    uint32_t                sample_rate_hz;     /*!< I2S sample rate */
    i2s_clock_src_t         clk_src;            /*!< Choose clock source, see `soc_periph_i2s_clk_src_t` for the supported clock sources.
                                                 *   selected `I2S_CLK_SRC_EXTERNAL` (if supports) to enable the external source clock input via MCLK pin,
                                                 */
#if SOC_I2S_HW_VERSION_2
    uint32_t                ext_clk_freq_hz;    /*!< External clock source frequency in Hz, only take effect when `clk_src = I2S_CLK_SRC_EXTERNAL`, otherwise this field will be ignored,
                                                 *   Please make sure the frequency input is equal or greater than BCLK, i.e. `sample_rate_hz * slot_bits * 2`
                                                 */
#endif
    i2s_mclk_multiple_t     mclk_multiple;      /*!< The multiple of MCLK to the sample rate
                                                 *   Default is 256 in the helper macro, it can satisfy most of cases,
                                                 *   but please set this field a multiple of `3` (like 384) when using 24-bit data width,
                                                 *   otherwise the sample rate might be inaccurate
                                                 */
} i2s_std_clk_config_t;

/**
 * @brief I2S default standard clock configuration
 * @note Please set the mclk_multiple to I2S_MCLK_MULTIPLE_384 while using 24 bits data width
 *       Otherwise the sample rate might be imprecise since the BCLK division is not a integer
 * @param rate sample rate
 */
#define I2S_STD_CLK_DEFAULT_CONFIG(rate) { \
    .sample_rate_hz = rate, \
    .clk_src = I2S_CLK_SRC_DEFAULT, \
    .mclk_multiple = I2S_MCLK_MULTIPLE_256, \
    .ext_clk_freq_hz = 0, \
}
#endif
```

The bug is fixed now, solved by 
```
#define I2S_STD_CLK_DEFAULT_CONFIG(rate) { \
    .sample_rate_hz = rate, \
    .clk_src = I2S_CLK_SRC_DEFAULT, \
    //.mclk_multiple = I2S_MCLK_MULTIPLE_256, \
    .ext_clk_freq_hz = 0, \
    .mclk_multiple = I2S_MCLK_MULTIPLE_256, \
}
#endif
```

Icarus113, 2025-02-20:
Link to https://github.com/espressif/esp-idf/issues/15405

williamleeag, 2025-02-20:
Thanks for your update.

closed 2025-02-20 by williamleeag