espressif/esp-idf #15426: Inconsistent ADC Readings Across ESP32-WROOM-32E Modules

anuraagMIPL, 2025-02-19:
Hello,

I am using three different ESP32-WROOM-32E modules and measuring ADC readings on GPIO36 (sensor VP pin) with input voltages ranging from 2.00V to 2.81V. I have verified the input voltage using a DSO, and the hardware input to the pin is identical and accurate across all modules as shown in excel.

However, I have observed significant variations in ADC readings between the three modules, which do not align with the expected formula:

`ADC_Value = (4095 * V) / 3.3`

I have attached a spreadsheet (ADC_Reading.xlsx) containing the raw ADC values from all three modules. According to the ESP32-WROOM-32E datasheet, ADC readings can vary by approximately ±6%, but my measured values exceed this range. This suggests that ADC behavior differs between modules, even under identical conditions.

My Questions:
1. Is there an inherent variation in ADC calibration across different ESP32-WROOM-32E modules?
2. Is there a recommended calibration procedure to normalize ADC readings across multiple ESP32 modules?

If anyone has encountered similar issues or has insights into standardizing ADC readings, I would greatly appreciate your advice! Thanks in advance.

[ADC Reading.xlsx](https://github.com/user-attachments/files/18869203/ADC.Reading.xlsx)

Icarus113, 2025-02-20:
May I know if you're reading the raw results or the calibrated results? See this [example](https://github.com/espressif/esp-idf/blob/master/examples/peripherals/adc/oneshot_read/main/oneshot_read_main.c#L99)

@anuraagMIPL

anuraagMIPL, 2025-02-20:
I am reading the raw ADC results on the ESP32 and simply using the moving average method to minimize rapid fluctuations.

@Icarus113

Icarus113, 2025-02-20:
The link I posted is the calibration API usage which can help you get calibrated results. 
- API link: [`adc_cali_raw_to_voltage`](https://github.com/espressif/esp-idf/blob/master/components/esp_adc/include/esp_adc/adc_cali.h#L54)
- [Example](https://github.com/espressif/esp-idf/blob/master/examples/peripherals/adc/oneshot_read/main/oneshot_read_main.c#L99) link
- An all-in-on API [`adc_oneshot_get_calibrated_result`](https://github.com/espressif/esp-idf/blob/master/components/esp_adc/include/esp_adc/adc_oneshot.h#L148)

@anuraagMIPL

anuraagMIPL, 2025-02-20:
Ok, I will try the above but before that can you also tell me - Why are the ADC readings behaving this way? Is there an inherent variation in ADC calibration across different **ESP32-WROOM-32E** modules? @Icarus113

Icarus113, 2025-02-21:
I think the data listed in the datasheet is based on the calibrated results, instead of raw results. And it's true there will be variation among different chips, even they are ESP32-WROOM-32E modules in a same batch.

anuraagMIPL, 2025-02-21:
Can we consider the Moving Average a form of calibration? I also tried both of the reading RAW and the Moving Average approach in the Arduino-ESP32 framework using `analogRead()` and got identical results! @Icarus113

ginkgm, 2025-03-21:
Hi @anuraagMIPL 

[adc_oneshot_get_calibrated_result](https://github.com/espressif/esp-idf/blob/master/components/esp_adc/include/esp_adc/adc_oneshot.h#L148) is a helper API,  but we also provide [calibration API](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/adc_calibration.html) for you to do the calibration yourself. You can do any kind of filtering (even IIR) as you wish, then do the calibration after that.

For esp32 the calibration is linear, so there is no accuracy difference between average after calibration or before calibration.

closed 2025-05-07 by ginkgm