espressif/esp-idf #15427: ESP32-C6's LP UART cannot generate some low baud rate

andylinpersonal, 2025-02-19:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.5-dev-1860-g0d6099ec53, v5.4-552-ge37d33cc1c

### Espressif SoC revision.

ESP32-C6 (QFN40) (revision v0.0)

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-C6-DevkitC-N8

### Power Supply used.

USB

### What is the expected behavior?

LP UART should be capable of generate correct baud rate under all supported ranges.

### What is the actual behavior?

The [secondary divider](https://github.com/espressif/esp-idf/blob/1160a86ba0b87b0ea68ad6b91eb10fe9c34ad692/components/soc/esp32c6/register/soc/lp_uart_reg.h#L1255) of `LP UART`, `LP_UART_CLK_CONF_REG.CLK_CONF.LP_UART_SCLK_DIV_NUM`, might not work as expected.
This register only appears in the `LL`/`SOC` code and not shown in the TRM.
When the required baud rate is too low and overflow the [main divider](https://github.com/espressif/esp-idf/blob/1160a86ba0b87b0ea68ad6b91eb10fe9c34ad692/components/soc/esp32c6/register/soc/lp_uart_reg.h#L545), `LP_UART_CLKDIV_SYNC_REG`'s [12-bit range](https://github.com/espressif/esp-idf/blob/1160a86ba0b87b0ea68ad6b91eb10fe9c34ad692/components/hal/esp32c6/include/hal/uart_ll.h#L162), the second divider `SCLK_DIV_NUM` will become [non-zero value](https://github.com/espressif/esp-idf/blob/1160a86ba0b87b0ea68ad6b91eb10fe9c34ad692/components/hal/esp32c6/include/hal/uart_ll.h#L171). But the resulting baud rate are `expected_baud_rate * (SCLK_DIV_NUM + 1)`.
The register values of dividers are correct, but the resulting signal is not.

### Steps to reproduce.

1. Configure the `LP UART`'s baud rate to any value outside the `LP_UART_CLKDIV_SYNC_REG`'s range, such as 2400bps.
2. See the output from the `GPIO5`. I use a FTDI's FT232RL as UART-to-USB converter.
3. [sclk_div](https://github.com/espressif/esp-idf/blob/1160a86ba0b87b0ea68ad6b91eb10fe9c34ad692/components/hal/esp32c6/include/hal/uart_ll.h#L162) in this case becomes `2` and `LP_UART_SCLK_DIV_NUM` is `1`, the resulting baud rate becomes `2400 * (LP_UART_SCLK_DIV_NUM + 1) = 4800bps`
4. Try other baud rate, such as `1700bps`, and the corresponding `sclk_div` is 3
5. The resulting baud rate becomes `1700 * 3 = 5100bps`

### Debug Logs.

```plain
None.
```

### More Information.

CMakeLists.txt
```CMake
cmake_minimum_required(VERSION 3.16)

set(CMAKE_CXX_STANDARD 23)
set(CMAKE_CXX_STANDARD_REQUIRED ON)
set(CMAKE_CXX_EXTENSIONS ON)

set(SDKCONFIG ${CMAKE_BINARY_DIR}/sdkconfig)

include($ENV{IDF_PATH}/tools/cmake/project.cmake)

project(test-lpuart)

```

main/CMakeLists.txt
```CMake
idf_component_register(
    SRCS "main.c"
    REQUIRES "ulp" "esp_driver_uart"
)

set(ulp_app_name lpuart)
set(ulp_lp_core_sources "ulp/main.c")
set(ulp_exp_dep_srcs "main.c")
ulp_embed_binary(lpuart "${ulp_lp_core_sources}" "${ulp_exp_dep_srcs}")

```

main/defines.h
```C
#pragma once

#include <soc/gpio_num.h>

#define USE_LP_UART_ON_LP 0
#define USE_LP_UART_ON_HP 1
#define USE_HP_UART 0       // HP UARTs are working as expected.

// #define UART_BAUD_RATE (5000U) // Safe baud rate for LP UART
#define UART_BAUD_RATE (2400U) // Bad baud rate for LP UART -> sclk_div_num = 1
// #define UART_BAUD_RATE (1700U) // Bad baud rate for LP UART -> sclk_div_num = 2

#define UART_TX_PIN GPIO_NUM_5

#if USE_HP_UART
#define UART_PORT   UART_NUM_1
// #define UART_CLK_SRC UART_SCLK_PLL_F80M // 80 MHz Default
// #define UART_CLK_SRC UART_SCLK_RTC      // 15.xxx MHz
// #define UART_CLK_SRC UART_SCLK_XTAL     // 40 MHz
#define UART_CLK_SRC UART_SCLK_DEFAULT
#endif

#if USE_LP_UART_ON_LP || USE_LP_UART_ON_HP
#define UART_PORT   LP_UART_NUM_0
// #define UART_CLK_SRC LP_UART_SCLK_LP_FAST // 15.xxx MHz Default
// #define UART_CLK_SRC LP_UART_SCLK_XTAL_D2 // 20 MHz
#define UART_CLK_SRC LP_UART_SCLK_DEFAULT
#endif

_Static_assert(USE_LP_UART_ON_LP + USE_LP_UART_ON_HP + USE_HP_UART == 1, "Select exact one of the test mode");

```

main/main.c
```C

#include "defines.h"

#include "sdkconfig.h"

#include <driver/i2c.h>
#include <driver/uart.h>
#include <lp_core_uart.h>
#include <string.h>
#include <ulp_lp_core.h>

#include <freertos/FreeRTOS.h>

extern const uint8_t lpuart_bin_start[] asm("_binary_lpuart_bin_start");
extern const uint8_t lpuart_bin_end[] asm("_binary_lpuart_bin_end");

#if USE_LP_UART_ON_LP
static const lp_core_uart_cfg_t lp_uart_cfg = {
    .uart_pin_cfg       = {.tx_io_num  = UART_TX_PIN,
                           .rx_io_num  = GPIO_NUM_NC,
                           .rts_io_num = GPIO_NUM_NC,
                           .cts_io_num = GPIO_NUM_NC},
    .uart_proto_cfg     = {.baud_rate           = UART_BAUD_RATE,
                           .data_bits           = UART_DATA_8_BITS,
                           .parity              = UART_PARITY_DISABLE,
                           .stop_bits           = UART_STOP_BITS_1,
                           .flow_ctrl           = UART_HW_FLOWCTRL_DISABLE,
                           .rx_flow_ctrl_thresh = 0},
    .lp_uart_source_clk = UART_CLK_SRC,
};
#endif

static void ulp_init()
{
	ulp_lp_core_stop();
#if USE_LP_UART_ON_LP
	ESP_ERROR_CHECK(lp_core_uart_init(&lp_uart_cfg));
#endif
	ESP_ERROR_CHECK(ulp_lp_core_load_binary(lpuart_bin_start, (lpuart_bin_end - lpuart_bin_start)));

	/* Start the program */
	ulp_lp_core_cfg_t cfg = {.wakeup_source = ULP_LP_CORE_WAKEUP_SOURCE_LP_TIMER, .lp_timer_sleep_duration_us = 10000};

	ESP_ERROR_CHECK(ulp_lp_core_run(&cfg));
}

static void uart_init()
{
#if USE_HP_UART || USE_LP_UART_ON_HP
	static uart_config_t uart_config = {
	    .baud_rate           = UART_BAUD_RATE,
	    .data_bits           = UART_DATA_8_BITS,
	    .parity              = UART_PARITY_DISABLE,
	    .stop_bits           = UART_STOP_BITS_1,
	    .flow_ctrl           = UART_HW_FLOWCTRL_DISABLE,
	    .rx_flow_ctrl_thresh = 0,
#if USE_HP_UART
	    .source_clk = UART_CLK_SRC,
#endif
#if USE_LP_UART_ON_HP
	    .lp_source_clk = UART_CLK_SRC,
#endif
	    .flags = {.allow_pd = false, .backup_before_sleep = false},
	};
	int intr_alloc_flags = 0;
	ESP_ERROR_CHECK(uart_driver_install(UART_PORT, UART_HW_FIFO_LEN(UART_PORT) * 2, 0, 0, NULL, intr_alloc_flags));
	ESP_ERROR_CHECK(uart_param_config(UART_PORT, &uart_config));
	ESP_ERROR_CHECK(uart_set_pin(UART_PORT, UART_TX_PIN, GPIO_NUM_NC, GPIO_NUM_NC, GPIO_NUM_NC));
#endif
}

void app_main(void)
{
	char linebuf[64];
	snprintf(linebuf, sizeof(linebuf), "[hp] port=%u, hello world!\n", UART_PORT);

	ulp_init();
	uart_init();

	while (true) {
#if USE_HP_UART || USE_LP_UART_ON_HP
		uart_write_bytes(UART_PORT, linebuf, strlen(linebuf));
		uart_wait_tx_done(UART_PORT, portMAX_DELAY);
#endif
		vTaskDelay(pdMS_TO_TICKS(1000));
	}
}

```

main/ulp/main.c
```C
#include "../defines.h"

#include <ulp_lp_core_print.h>
#include <ulp_lp_core_uart.h>
#include <ulp_lp_core_utils.h>

#define CONFIG_USE_LP_UART_ON_LP 1

int main(void)
{
	while (1) {
#if USE_LP_UART_ON_LP
		lp_core_printf("[lp] port=%u, hello world!\n", UART_PORT);
#endif
		ulp_lp_core_delay_us(1000000);
	}
	return 0;
}

```

sdkconfig.defaults
```
CONFIG_ULP_COPROC_ENABLED=y
CONFIG_ULP_COPROC_RESERVE_MEM=15360
CONFIG_ULP_PANIC_OUTPUT_ENABLE=n
CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG=y
```

andylinpersonal, 2025-02-21:
Hi, is there any workaround for this?

songruo, 2025-02-24:
Hi @andylinpersonal, we have confirmed that LP UART port indeed does not have this stage of divider (`LP_UART_CLK_CONF_REG.CLK_CONF.LP_UART_SCLK_DIV_NUM`). Therefore, unfortunately,  it might be impossible to generate a baud rate of 2400 bps or 1700 bps with the LP UART port.

We will fix the IDF to alert the users that some baud rate values are unachievable accordingly.

andylinpersonal, 2025-02-24:
OK, thanks for the reply. 
It'll be better to highlight this limitation in the TRM as well.
BTW, is it possible to use the HP UART port to receive data from LP core while put HP core into light sleep or even deep sleep mode?

songruo, 2025-02-25:
Nope, HP UART cannot work under sleep. The only way I can think of is maybe you can try bit banging since the baud rate is pretty low. You can configure an auto reload LP timer to fire at the baud rate frequency, then read and record the LP IO (rx pin) status in the timer ISR.

closed 2025-03-12 by espressif-bot