espressif/esp-idf #15432: how to use mdns_init() in esp-idf v5.4?

zzzh, 2025-02-20:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

my code is like this:

...
#include "mdns.h"
...

void initialise_mdns(void)
{
	//initialize mDNS
	ESP_ERROR_CHECK( mdns_init() );
	//set mDNS hostname (required if you want to advertise services)
	ESP_ERROR_CHECK( mdns_hostname_set("my-ftp-server") );
	ESP_LOGI(TAG, "mdns hostname set to: [%s]", "my-ftp-server");
}

which works well in v4.4.2
I hope it works well as before

### What is the actual behavior?

but in v5.4:
it says fatal error: mdns.h: No such file or directory
I tried add mdns in REQUIRES of CMakeLists.txt, it says Failed to resolve component 'mdns' required by component 'main': unknown name.
and there's no mdns_init() in any head file.

### Steps to reproduce.

try to use mdns_init(), I'm sure it would fail.

### Build or installation Logs.

```plain
fatal error: mdns.h: No such file or directory
I tried to add mdns in REQUIRES of CMakeLists.txt, it says Failed to resolve component 'mdns' required by component 'main': unknown name.
and there's no mdns_init() in any head file.
```

peter-marcisovsky, 2025-02-20:
Hi @zzzh 

mdns is an external component. You need to include it into your `idf_component.yml` file, not into the cmake. Like [this](https://github.com/espressif/esp-idf/blob/c71d74e2f853b1135c63f47e349f2a08f63f3e01/examples/peripherals/usb/host/uvc/main/idf_component.yml#L5) for example.

zzzh, 2025-02-20:
Hi, @peter-marcisovsky ,

You are right, I run this:
idf.py add-dependency "espressif/mdns^1.7.0"
There's no compile complain for it any more.

Thanks for the help!

closed 2025-02-20 by zzzh