espressif/esp-idf #15437: USB host with external HUB crashes when port is gone

xilard, 2025-02-20:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.4

### Espressif SoC revision.

ESP32-S3

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3 DevKitC-1 v1.1

### Power Supply used.

USB

### What is the expected behavior?

No crash when USB port is gone from the HUB.

### What is the actual behavior?

Output:

```
I (2107) usb: New HID HOST
I (2137) usb: Waiting for HID Device to be connected
W (2517) hid-host: No HID device at USB port 1
E (2987) EXT_HUB: [1:4] Port is gone

assert failed: port_delete ext_port.c:1132 (ext_port->flags.in_pending_list == 0)

Backtrace: 0x40375dfe:0x3fcf4ff0 0x403843f9:0x3fcf5010 0x40389691:0x3fcf5030 0x4205a422:0x3fcf5150 0x42058c9c:0x3fcf5170 0x42059cfa:0x3fcf51a0 0x420583d1:0x3fcf51e0 0x42051643:0x3fcf5210 0x42052b65:0x3fcf5240 0x42053196:0x3fcf5270 0x42051d7c:0x3fcf5290 0x420242f7:0x3fcf52b0 0x40384956:0x3fcf52d0
```

The problem is the `port_set_actions(ext_port, PORT_ACTION_DISABLE)` inserts an item to the pending list.

Original function in **usb/ext_port.c**:

```
static void port_set_actions(ext_port_t *ext_port, uint32_t action_flags)
{
    assert(action_flags != 0); // Sanity check

    // Check if port is not in the pending list
    if (!ext_port->flags.in_pending_list) {
        // Add port to pending list
        ext_port->flags.in_pending_list = 1;
        TAILQ_INSERT_TAIL(&p_ext_port_driver->single_thread.pending_tailq, ext_port, tailq_entry);
    }

    ext_port->action_flags |= action_flags;
    p_ext_port_driver->constant.proc_req_cb(p_ext_port_driver->constant.proc_req_cb_arg);
}
```

Fixed function:

```
static void port_set_actions(ext_port_t *ext_port, uint32_t action_flags)
{
    assert(action_flags != 0); // Sanity check

    // Check if port is not in the pending list
    if ((action_flags != PORT_ACTION_DISABLE) && (!ext_port->flags.in_pending_list)) {
        // Add port to pending list
        ext_port->flags.in_pending_list = 1;
        TAILQ_INSERT_TAIL(&p_ext_port_driver->single_thread.pending_tailq, ext_port, tailq_entry);
    }

    ext_port->action_flags |= action_flags;
    p_ext_port_driver->constant.proc_req_cb(p_ext_port_driver->constant.proc_req_cb_arg);
}
```

After fixed the `port_set_actions` function the port problem still logged out but the mouse/keyboard always connected and works well:

```
I (2577) usb: New HID HOST
I (2607) usb: Waiting for HID Device to be connected
W (2987) hid-host: No HID device at USB port 1
E (3457) EXT_HUB: [1:4] Port is gone
I (5187) usb: HID Device, protocol '2' CONNECTED
HID data: 00040100
HID data: 00040200
HID data: 00030100
HID data: 0000FF00
HID data: 00FF0100
HID data: 00FDFF00
HID data: 00FEFF00
...

### Steps to reproduce.

1. Connect a USB HUB to ESP32-S3 and a USB device (mouse/keyboard) to a HUB port
2. Power ON

### Debug Logs.

```plain

```

roma-jam, 2025-02-20:
Hi @xilard 

thanks for reporting the issue and proposing the fix.
We already have fixed the problem, when the device in the root port detached and we need to handle closing the whole previous device tree, so maybe we already covered this scenario as well.  

To understand better, may I ask, are you detaching the hub from the port with attached device, am I get it right? 
Or it happens in some other case?

xilard, 2025-02-20:
Hi @roma-jam 

All USB connected before power up.
1. ESP-S3 devkit
2. Powered USB HUB
3. USB mouse

Here the log after power up external HUB and devkit. The firmware usualy reports one or more "Port is gone" at startup. After started I unplug and replug USB mouse few times in the HUB.
But need the `port_set_actions()` function fix, otherwise crashes the external hub component. Working well long time without problem after reported CONNECTED.

```
I (2147) usb: New HID HOST
I (2177) usb: Waiting for HID Device to be connected
W (2557) hid-host: No HID device at USB port 1
E (3027) EXT_HUB: [1:4] Port is gone
W (3407) hid-host: No HID device at USB port 2
E (3877) EXT_HUB: [2:4] Port is gone
W (4257) hid-host: No HID device at USB port 3
E (4727) EXT_HUB: [3:4] Port is gone
W (5117) hid-host: No HID device at USB port 4
I (5587) usb: HID Device, protocol '2' CONNECTED
HID data: 00020200
HID data: 00010200
HID data: 00010200
HID data: 00000200
HID data: 00000200
I (10577) usb: HID Device, protocol '2' DISCONNECTED
I (12597) usb: HID Device, protocol '2' CONNECTED
HID data: 00000000
HID data: 00020100
HID data: 00010100
I (16657) usb: HID Device, protocol '2' DISCONNECTED
I (18647) usb: HID Device, protocol '2' CONNECTED
HID data: 00000000
HID data: 00FFFF00
HID data: 00FEFF00
HID data: 00FFFF00
HID data: 00FDFD00
I (21847) usb: HID Device, protocol '2' DISCONNECTED
E (23767) EXT_HUB: [4:4] Port is gone
W (24157) hid-host: No HID device at USB port 8
E (24627) EXT_HUB: [8:4] Port is gone
W (25017) hid-host: No HID device at USB port 9
E (25487) EXT_HUB: [9:4] Port is gone
W (25867) hid-host: No HID device at USB port 10
I (26337) usb: HID Device, protocol '2' CONNECTED
HID data: 00000000
HID data: 00000100
HID data: 00FF0200
HID data: 00FE0000
HID data: 00FEFE00
HID data: 00FBFE00
HID data: 00FE0100
I (28867) usb: HID Device, protocol '2' DISCONNECTED
E (30887) EXT_HUB: [10:4] Port is gone
W (31267) hid-host: No HID device at USB port 12
E (31737) EXT_HUB: [12:4] Port is gone
W (32127) hid-host: No HID device at USB port 13
I (32597) usb: HID Device, protocol '2' CONNECTED
HID data: 00000000
HID data: 00000100
HID data: 00FBFA00
HID data: 00FAFB00
HID data: 00FE0100
```

roma-jam, 2025-02-20:
@xilard 
Thanks for the additional information.

May I ask you to run esp-idf `usb_host_lib` example,  with connected hub and USB mouse and share the output here? 
There will be descriptors of Hub and Mouse.

xilard, 2025-02-20:
Hi @roma-jam 

Ok, I built the `usb_host_lib`, here the output:

```
I (307) USB host lib: USB host library example
I (307) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:2 
I (317) USB host lib: Installing USB Host Library
I (347) CLASS: Registering Client
I (727) CLASS: Opening device at address 1
I (727) CLASS: Getting device information
I (727) CLASS: 	Full speed
I (727) CLASS: 	Parent info:
I (727) CLASS: 		Port: ROOT
I (727) CLASS: 	bConfigurationValue 1
I (737) CLASS: Getting device descriptor
*** Device descriptor ***
bLength 18
bDescriptorType 1
bcdUSB 2.10
bDeviceClass 0x9
bDeviceSubClass 0x0
bDeviceProtocol 0x0
bMaxPacketSize0 64
idVendor 0x5e3
idProduct 0x610
bcdDevice 2.20
iManufacturer 1
iProduct 2
iSerialNumber 0
bNumConfigurations 1
I (757) CLASS: Getting config descriptor
*** Configuration descriptor ***
bLength 9
bDescriptorType 2
wTotalLength 25
bNumInterfaces 1
bConfigurationValue 1
iConfiguration 0
bmAttributes 0xe0
bMaxPower 100mA
	*** Interface descriptor ***
	bLength 9
	bDescriptorType 4
	bInterfaceNumber 0
	bAlternateSetting 0
	bNumEndpoints 1
	bInterfaceClass 0x9
	bInterfaceSubClass 0x0
	bInterfaceProtocol 0x0
	iInterface 0
		*** Endpoint descriptor ***
		bLength 7
		bDescriptorType 5
		bEndpointAddress 0x81	EP 1 IN
		bmAttributes 0x3	INT
		wMaxPacketSize 1
		bInterval 255
I (807) CLASS: Getting Manufacturer string descriptor
GenesysLogic
I (817) CLASS: Getting Product string descriptor
USB2.0 Hub
E (1247) HUB: Connected Low-speed device, not supported
E (1247) EXT_PORT: [1:4] Port disabled, reset attempts=1
W (1247) HUB: Device tree node (parent_port=4): not found
```

Yes, to display mouse descriptor have to enable low speed support.

```
CONFIG_IDF_EXPERIMENTAL_FEATURES=y
CONFIG_USB_HOST_EXT_PORT_SUPPORT_LS=y
```

Output:

```
I (307) USB host lib: USB host library example
I (307) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:2 
I (317) USB host lib: Installing USB Host Library
I (347) CLASS: Registering Client
I (727) CLASS: Opening device at address 1
I (727) CLASS: Getting device information
I (727) CLASS: 	Full speed
I (727) CLASS: 	Parent info:
I (727) CLASS: 		Port: ROOT
I (727) CLASS: 	bConfigurationValue 1
I (737) CLASS: Getting device descriptor
*** Device descriptor ***
bLength 18
bDescriptorType 1
bcdUSB 2.10
bDeviceClass 0x9
bDeviceSubClass 0x0
bDeviceProtocol 0x0
bMaxPacketSize0 64
idVendor 0x5e3
idProduct 0x610
bcdDevice 2.20
iManufacturer 1
iProduct 2
iSerialNumber 0
bNumConfigurations 1
I (757) CLASS: Getting config descriptor
*** Configuration descriptor ***
bLength 9
bDescriptorType 2
wTotalLength 25
bNumInterfaces 1
bConfigurationValue 1
iConfiguration 0
bmAttributes 0xe0
bMaxPower 100mA
	*** Interface descriptor ***
	bLength 9
	bDescriptorType 4
	bInterfaceNumber 0
	bAlternateSetting 0
	bNumEndpoints 1
	bInterfaceClass 0x9
	bInterfaceSubClass 0x0
	bInterfaceProtocol 0x0
	iInterface 0
		*** Endpoint descriptor ***
		bLength 7
		bDescriptorType 5
		bEndpointAddress 0x81	EP 1 IN
		bmAttributes 0x3	INT
		wMaxPacketSize 1
		bInterval 255
I (807) CLASS: Getting Manufacturer string descriptor
GenesysLogic
I (817) CLASS: Getting Product string descriptor
USB2.0 Hub
I (1287) CLASS: Opening device at address 2
I (1287) CLASS: Getting device information
I (1287) CLASS: 	Low speed
I (1287) CLASS: 	Parent info:
I (1287) CLASS: 		Bus addr: 1
I (1297) CLASS: 		Port: 4
I (1297) CLASS: 	bConfigurationValue 1
I (1297) CLASS: Getting device descriptor
*** Device descriptor ***
bLength 18
bDescriptorType 1
bcdUSB 1.10
bDeviceClass 0x0
bDeviceSubClass 0x0
bDeviceProtocol 0x0
bMaxPacketSize0 8
idVendor 0x1c4f
idProduct 0x34
bcdDevice 1.10
iManufacturer 1
iProduct 2
iSerialNumber 0
bNumConfigurations 1
I (1327) CLASS: Getting config descriptor
*** Configuration descriptor ***
bLength 9
bDescriptorType 2
wTotalLength 34
bNumInterfaces 1
bConfigurationValue 1
iConfiguration 0
bmAttributes 0xa0
bMaxPower 100mA
	*** Interface descriptor ***
	bLength 9
	bDescriptorType 4
	bInterfaceNumber 0
	bAlternateSetting 0
	bNumEndpoints 1
	bInterfaceClass 0x3
	bInterfaceSubClass 0x1
	bInterfaceProtocol 0x2
	iInterface 0
		*** Endpoint descriptor ***
		bLength 7
		bDescriptorType 5
		bEndpointAddress 0x81	EP 1 IN
		bmAttributes 0x3	INT
		wMaxPacketSize 4
		bInterval 10
I (1377) CLASS: Getting Manufacturer string descriptor
SIGMACHIP
I (1377) CLASS: Getting Product string descriptor
Usb Mouse]

```

xilard, 2025-02-20:
In addition I installed esp idf v5.4 toolchain on a clean windows in a VM and built `peripherals/usb/host/hid` example with enabled low speed support.

The result a boot loop with external hub component crash.

```
I (312) example: HID Host example
I (312) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:2 
I (352) example: Waiting for HID Device to be connected
W (732) hid-host: No HID device at USB port 1
E (1192) EXT_HUB: [1:4] Port is gone

assert failed: port_delete ext_port.c:1132 (ext_port->flags.in_pending_list == 0)

Backtrace: 0x40375b01:0x3fc99a00 0x40379c8d:0x3fc99a20 0x40380435:0x3fc99a40 0x4201268d:0x3fc99b60 0x42010ea8:0x3fc99b80 0x42011f63:0x3fc99bb0 0x4201059d:0x3fc99bf0 0x4200971b:0x3fc99c20 0x4200ae15:0x3fc99c50 0x4200b442:0x3fc99c80 0x42009f65:0x3fc99ca0 0x42007661:0x3fc99cc0 0x4037a66d:0x3fc99cf0
```

I updated the `ext_port.c` as described in the first post and the example works. It reports some port problem at startup but works.

```
I (308) example: HID Host example
I (308) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:2 
I (348) example: Waiting for HID Device to be connected
W (728) hid-host: No HID device at USB port 1
W (1148) example: To shutdown example, remove all USB devices and press button again.
E (1198) EXT_HUB: [1:4] Port is gone
W (1588) hid-host: No HID device at USB port 2
E (2058) EXT_HUB: [2:4] Port is gone
W (2438) hid-host: No HID device at USB port 3
E (2908) EXT_HUB: [3:4] Port is gone
W (3298) hid-host: No HID device at USB port 4
I (3768) example: HID Device, protocol 'MOUSE' CONNECTED

Mouse
X: 000000	Y: 000000	| | |

```

roma-jam, 2025-02-20:
@xilard 

thank you so much for the additional information and testing. 
There are couple of things are missing, that is why Low-speed device through an external hub is in experimental feature. 

The fix for this and adding the remaining logic will be uploaded during the next synchronization of the public repo, in the week or two.

closed 2025-02-26 by espressif-bot