espressif/esp-idf #15438: Peer.c line 328 assert fail in a blecentral application

scholayil, 2025-02-20:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

Release 5.4

### Espressif SoC revision.

ESP32-C3 (QFN32) (version v0.4)

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom board with ESP32-C3-WROOM-02 module

### Power Supply used.

USB

### What is the expected behavior?

ESP32C3 device acting as ble central is bonded with a Silab BG22 peripheral. On one restart of the ESP32C3 device, the ble central app crashed during device service discovery.

### What is the actual behavior?

The one crash we saw was traced to Peer.c Line 328 assert.

### Steps to reproduce.

Not repeatable, happened randomly 2 times in a month on esp32c3 device startup.

### Debug Logs.

```plain
I (9539) NimBLE: Connection established
I (9539) NimBLE: 

I (9539) NimBLE: Connection secured

I (9549) NimBLE: mtu update event; conn_handle=1 cid=4 mtu=247

I (9699) blecent_gap_event: Pairing Done : 00:00:00:00:01:00
I (9699) blecent_gap_event: Pairing complete with device: 66:78:20:FB:23:34 (Type: 0)

W (9699) blecent_gap_event: idx :0 bleConnectionHandler 1
I (9709) NimBLE: encryption change event; status=0 
I (9709) NimBLE: GATT procedure initiated: discover all services

I (9849) NimBLE: --------------- Got event type 0x22
I (9899) NimBLE: GATT procedure initiated: discover all characteristics; 
I (9899) NimBLE: start_handle=1 end_handle=8

I (10149) blecent_gap_event: Pairing Done : 00:00:00:00:01:00
I (10149) blecent_gap_event: Pairing complete with device: 66:78:20:FB:23:34 (Type: 0)

W (10149) blecent_gap_event: idx :0 bleConnectionHandler 1
I (10159) NimBLE: encryption change event; status=0 
I (10159) NimBLE: GATT procedure initiated: discover all services

assert failed: peer_chr_add peer.c:328 (0)
Core  0 register dump:
MEPC    : 0x403807d4  RA      : 0x4038b096  SP      : 0x3fcd4490  GP      : 0x3fc97c00  
--- 0x403807d4: panic_abort at /home/user1/esp/v5.3.1/esp-idf/components/esp_system/panic.c:463
0x4038b096: __ubsan_include at /home/user1/esp/v5.3.1/esp-idf/components/esp_system/ubsan.c:311

TP      : 0x3fcd4730  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130  
S0/FP   : 0x00000001  S1      : 0x3fcd45da  A0      : 0x3fcd44e8  A1      : 0x3fc9a3dd  
A2      : 0x00000001  A3      : 0x00000029  A4      : 0x00000001  A5      : 0x3fca2000  
A6      : 0x7a797877  A7      : 0x76757473  S2      : 0x0000009d  S3      : 0x3fcd44e8  
S4      : 0x3fcd44e8  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000  
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938  
MSTATUS : 0x00001801  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000  
--- 0x40380001: _vector_table at /home/user1/esp/v5.3.1/esp-idf/components/riscv/vectors_intc.S:54

MHARTID : 0x00000000  

Stack memory:
3fcd4490: 0x00000000 0x00000000 0x3c14ef70 0x40392d60 0x06d7be00 0x00000000 0x0000001b 0x3fc9a3dc
--- 0x40392d60: newlib_include_assert_impl at /home/user1/esp/v5.3.1/esp-idf/components/newlib/assert.c:92

3fcd44b0: 0x00383233 0x00000000 0x00000000 0x00000000 0x3f000000 0x3fc9a0dc 0x3c14ef70 0x3fc9a8d4
3fcd44d0: 0x3c135350 0x3fc9a0ec 0x3fcd44b0 0x3fc9a0f0 0x3c13ac3c 0x3fc9a3dc 0x65737361 0x66207472
3fcd44f0: 0x656c6961 0x70203a64 0x5f726565 0x5f726863 0x20646461 0x72656570 0x333a632e 0x28203832
3fcd4510: 0x00002930 0x3fca3f2c 0x00000001 0x42038756 0x00000001 0xfb3ed300 0x3f84ba20 0x42000000
--- 0x42038756: ble_gatts_bonding_restored at /home/user1/esp/v5.3.1/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_gatts.c:2461

3fcd4530: 0x3fcd4730 0x00000000 0x3fcd456c 0x4038bc12 0x3fcd4690 0x3fca3000 0x3c14f17c 0xffffffff
--- 0x4038bc12: xQueueSemaphoreTake at /home/user1/esp/v5.3.1/esp-idf/components/freertos/FreeRTOS-Kernel/queue.c:1731

3fcd4550: 0x00000001 0x00000000 0x00000001 0x42035e12 0x00000001 0x00000000 0x420311e0 0x00000007
--- 0x42035e12: ble_gap_enc_event at /home/user1/esp/v5.3.1/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_gap.c:7211
0x420311e0: ble_gattc_proc_matches_conn_rx_entry at /home/user1/esp/v5.3.1/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_gattc.c:933

3fcd4570: 0x3fcd45f8 0x00000000 0x3fcd1eb0 0x4038b102 0x00000000 0x00000000 0x3fcd1eb0 0x4038b676
--- 0x4038b102: prvCopyDataToQueue at /home/user1/esp/v5.3.1/esp-idf/components/freertos/FreeRTOS-Kernel/queue.c:2470
0x4038b676: xQueueGenericSend at /home/user1/esp/v5.3.1/esp-idf/components/freertos/FreeRTOS-Kernel/queue.c:966

3fcd4590: 0x00000000 0x3fca3000 0x3fca3854 0x00000000 0x00000000 0x00000000 0x00000000 0xffffffff
3fcd45b0: 0x00000000 0x00000000 0x3fcd45f4 0x00000007 0x00000000 0x3fcd1f08 0x00000000 0x4202bd08
--- 0x4202bd08: peer_chr_disced at /home/user1/esp/v5.3.1/esp-idf/examples/bluetooth/nimble/common/nimble_central_utils/peer.c:366

3fcd45d0: 0x00000000 0x3fcd4644 0x00000005 0x00000007 0x3fca2fc0 0x3fcd4644 0x00000000 0x42030f18
--- 0x42030f18: ble_gattc_disc_all_chrs_rx_adata at /home/user1/esp/v5.3.1/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_gattc.c:2270

3fcd45f0: 0x3fcd4730 0x00030002 0x00000020 0x2a050010 0x00000000 0x00000000 0x00000000 0x00000000
3fcd4610: 0x00000001 0x3fcd4644 0x3fca2fc0 0x420322e6 0x00000001 0x00000000 0x3fca33e4 0x3c14f468
--- 0x420322e6: ble_gattc_rx_read_type_adata at /home/user1/esp/v5.3.1/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_gattc.c:4853

3fcd4630: 0x00000000 0x00000000 0x3fca33e4 0x42032f1e 0x00000000 0x00000002 0x00000005 0x3fccffac
--- 0x42032f1e: ble_att_clt_rx_read_type at /home/user1/esp/v5.3.1/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_att_clt.c:432

3fcd4650: 0x3fccff88 0x00000000 0x3fca3f2c 0x3fca33e4 0x3fca33d4 0x00000001 0x3c14f33c 0x420306fc
--- 0x420306fc: ble_att_rx at /home/user1/esp/v5.3.1/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_att.c:517

3fcd4670: 0x00000000 0x3fcd473c 0x3fcd1eb0 0x0938b800 0x3fccff88 0x00000000 0x3fca3854 0x00000001
3fcd4690: 0x00000000 0x00000000 0x00000000 0x4203054a 0x3fca3000 0x001b2001 0x42030686 0xffffffff
--- 0x4203054a: ble_hs_hci_evt_acl_process at /home/user1/esp/v5.3.1/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_hci_evt.c:1196
0x42030686: ble_att_rx at /home/user1/esp/v5.3.1/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_att.c:486

3fcd46b0: 0x3fca3000 0x3fccf410 0x00000000 0x3fca3000 0x3fca3000 0x3fca2a10 0x3fca3000 0x4202f8e4
--- 0x4202f8e4: ble_hs_process_rx_data_queue at /home/user1/esp/v5.3.1/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs.c:248

3fcd46d0: 0x3fca3000 0x3fca2a10 0x3fca2eb4 0x40381b60 0x00000000 0x00000000 0x00000000 0x00000000
--- 0x40381b60: nimble_port_run at /home/user1/esp/v5.3.1/esp-idf/components/bt/host/nimble/nimble/porting/nimble/src/nimble_port.c:307

3fcd46f0: 0x00000000 0x00000000 0x3c14c000 0x42017002 0x00000000 0x00000000 0x00000000 0x00000000
--- 0x42017002: blecent_host_task at /home/user1/projects/edcm87_hub_esp32/components/blecent/src/ble_central.c:1562

3fcd4710: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fcd4730: 0xa5a5a5a5 0xa5a5a5a5 0x00000150 0x3fcd45e0 0x00000000 0x3fc9b360 0x3fc9b360 0x3fcd473c
3fcd4750: 0x3fc9b358 0x00000004 0x3fcd1c30 0x3fcd1c30 0x3fcd473c 0x00000000 0x00000015 0x3fcd3738
3fcd4770: 0x626d696e 0x685f656c 0x0074736f 0x00000000 0x3fcd4730 0x00000015 0x00000000 0x00000000
3fcd4790: 0x00000000 0x00000000 0x3fca4df8 0x3fca4e60 0x3fca4ec8 0x00000000 0x00000000 0x00000001
3fcd47b0: 0x00000000 0x00000000 0x00000000 0x42004298 0x00000000 0x00000000 0x00000000 0x00000000
--- 0x42004298: esp_cleanup_r at /home/user1/esp/v5.3.1/esp-idf/components/newlib/newlib_init.c:43

3fcd47d0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcd47f0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcd4810: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcd4830: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcd4850: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcd4870: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000020
```

Shreeyash-17, 2025-02-26:
@scholayil 

Possible Reason for the failure :

- Looking at the logs, it seems that the server is sending incorrect information.

- The svc_start_handle provided to peer_chr_add might not match any existing service in the list. A wrong start handle could be used due to incorrect parsing.

Please check this on the server side. However, since the server provides incorrect values, it is causing an assert.
Please provide sdkconfig and debug logs.
To prevent this, I have modified the code and attached a patch file below, which should help avoid the issue.
Additionally, please try with NimBLE examples and verify the behavior.

path to apply patch : esp-idf /
[blecent.txt](https://github.com/user-attachments/files/18983370/blecent.txt)

scholayil, 2025-02-26:
@Shreeyash-17 thanks for the suggestions.

Added the patch file, but not seen any of the ""peer_chr_add: ERROR ...." log that was added in place of assert(0).

So far 40+ rebonded connects (general adv.) or reconnects (directed adv.) have not reproduced the error. Tested under a) powering either side on/off , b) both, or c) remote forced bonding. Will update.

scholayil, 2025-02-27:
@Shreeyash-17 

I am now able to consistently reproduce the error by power cycling the BG22 peripheral at near the device reset chip input voltage level. Here is the log from the ESP32C3 hub.

I (800097) NimBLE: --------------- Got event type 0x22
peer_chr_add: ERROR - Service with start_handle=0x0001 not found!
E (800097) NimBLE: Error: Service discovery failed; status=17 conn_handle=1

More detailed logs attached. I will go ahead close this issue.

Thanks

[when-error.txt](https://github.com/user-attachments/files/19016844/when-error.txt)
[when-no-error.txt](https://github.com/user-attachments/files/19016845/when-no-error.txt)

scholayil, 2025-02-27:
@Shreeyash-17 will the esp-idf be updated with the patch you mentioned above? otherwise I will have to maintain a repo just for this in our private github. Thanks

closed 2025-02-28 by scholayil