espressif/esp-idf #15441: nvs_partition_gen.py to support utf-8

crazydiamondzgy, 2025-02-21:
### Is your feature request related to a problem?

in components\nvs_flash\nvs_partition_generator folder, sample_val.csv is working well
I change some string to utf-8 as following, 
``` csv
# Sample csv file
key,type,encoding,value
storage,namespace,,
u8_key,data,u8,255
i8_key,data,i8,-128
u16_key,data,u16,65535
u32_key,data,u32,4294967295
i32_key,data,i32,-2147483648
str_key,data,string,"中文"
```
then generate the nvs binary file by
``` shell
python C:\home\github\esp-idf\components\nvs_flash\nvs_partition_generator\nvs_partition_gen.py generate C:\home\github\esp-idf\components\nvs_flash\nvs_partition_generator\sample_val.csv nvs.bin 0x3000
```
from the hex view, we can see that the size is error,  actural length of "中文" is 7, but the wrong length in nvs.bin is 3, so I double checked by some simple program to read the key-value item which is failed.

### Describe the solution you'd like.

after all, i reviewed the nvs_partition_gen.py file and did little modification with AI, it seems like 

``` python

    def write_entry(self, key, value, encoding):
        # Encoding-specific handling
        if encoding == 'hex2bin':
            value = value.strip()
            if len(value) % 2 != 0:
                raise InputError('%s: Invalid data length. Should be multiple of 2.' % key)
            value = binascii.a2b_hex(value)
        elif encoding == 'base64':
            value = binascii.a2b_base64(value)
        elif encoding == 'string':
            if type(value) == bytes:
                value = value.decode()
            value += '\0'
            value = value.encode("utf-8")  # added line
```

everything is ok now.
did I need to add a PR to the resposity ?

3reality-wangzhijie, 2025-03-05:
Great job. Chinese comment may also lead to a compiling error.

rrtandler, 2025-03-07:
Hi @crazydiamondzgy ,

For backward compatibility, I would rather consider adding the new encoding type (for example utf8 with meaning utf8 encoded data stored as c-style string in NVS) instead of changing the meaning of existing string encoding. There might be other users of the generator already using ASCII characters >127 for whose the functionality would unexpectedly change.

crazydiamondzgy, 2025-03-12:
see it, good idea, pls add it when you have time

adokitkat, 2025-07-08:
Hello. This should be solved with NVS partition generator version >=0.1.9 where getting the correct length of non-ASCII characters was fixed,

crazydiamondzgy, 2025-07-12:
good job