espressif/esp-idf #15442: Move repeated NVS partition encryped to debug logs

dizcza, 2025-02-21:
This should be a debug log

https://github.com/espressif/esp-idf/blob/c71d74e2f853b1135c63f47e349f2a08f63f3e01/components/nvs_flash/src/nvs_api.cpp#L157

Firstly, I thought the log indicates there was an error to read/write from the enrypted nvs partition and the error was printed silently. I needed to find the source line to investigate whether it was just a notification or more.

rrtandler, 2025-02-24:
Hi @dizcza,
The three IDF examples related to flash/NVS security: 

- examples/security/flash_encryption/README.md
- examples/security/nvs_encryption_hmac/README.md 
- examples/security/security_features_app/README.md

Are showing this information message in their respective log examples. Therefore the existence of the log Info message shouldn't be a surprise, rather confirmation of good flow. I would not consider emitting of this information using LOGI as a bug.

dizcza, 2025-02-24:
I don't want to see the "NVS partition \"%s\" is encrypted" log each time I do a write operation.

dizcza, 2025-02-24:
But I'm not insisting.
Feel free to close.

rrtandler, 2025-02-25:
Hi @dizcza,
Are you using some framework for your application or is the NVS initialisation fully under control of your code? If your application is controlling the moment, when the NVS gets initialised, you can avoid it before every write. Once initialised, it can be reused. You can force the re-initialising of NVS only in case the nvs_get or nvs_set returns error codes. It may not only reduce the unwanted log message but also some cpu time.

rrtandler, 2025-02-25:
@dizcza - Yes, your assumption is correct. The secure init returns at the moment it detects already initialised partition (in your case the default one).  While it, unfortunately, always emits the unwanted LOGI message.  There is one fast solution for you - to keep track of initialisation at the function bsp_nvs_flash_init level i.e. in a static variable and omit the multiple call to the nvs_flash_init.
The long term solution will be the change from LOGI to LOGD in IDF.
I can also recommend to evaluate the esp_err_t result of nvs_set_u16 as it may return result codes valuable for required space estimation or need for re-initialisation. You can encounter errors like ESP_ERR_NVS_NOT_ENOUGH_SPACE, ESP_ERR_NVS_NOT_INITIALIZED or ESP_ERR_INVALID_STATE.

dizcza, 2025-02-25:
All right, thanks for the clarification.

I'll go with your first approach suggestion.

closed 2025-02-25 by dizcza