espressif/esp-idf #15444: Race condition in the i2c driver

gomatteo, 2025-02-21:
Idf version: v5.4-561-gbcb3c32d3a

Hardware: ESP32-D0WD (revision v1.0)

There is a race condition in the i2c driver in case of timeout.
The problem lies here:
https://github.com/espressif/esp-idf/blob/bcb3c32d3af38ce248f23103dab21173edea2d7b/components/esp_driver_i2c/i2c_master.c#L466-L475

If an interrupt fires between line 470 and 471 then `i2c_master->trans_idx` is zeroed before changing `i2c_master->status`.
Inside the ISR the branch `i2c_master->status == I2C_STATUS_READ` is taken, leading to a null pointer dereference on line 650 (`i2c_operation->data` is null).

https://github.com/espressif/esp-idf/blob/bcb3c32d3af38ce248f23103dab21173edea2d7b/components/esp_driver_i2c/i2c_master.c#L647-L650

Here is a backtrace showing the crash

```
#0  0x40087cb4 in i2c_ll_read_rxfifo (hw=0x3ff53000, ptr=0x0, len=3 '\003') at /opt/esp/idf/components/hal/esp32/include/hal/i2c_ll.h:580
#1  i2c_isr_receive_handler (i2c_master=0x3ffd2e2c) at /opt/esp/idf/components/esp_driver_i2c/i2c_master.c:650
#2  0x40087df1 in i2c_master_isr_handler_default (arg=0x3ffd2e2c) at /opt/esp/idf/components/esp_driver_i2c/i2c_master.c:708
#3  0x40083a6c in shared_intr_isr (arg=0x3ffd336c) at /opt/esp/idf/components/esp_hw_support/intr_alloc.c:445
#4  0x40084e88 in _xt_lowint1 () at /opt/esp/idf/components/xtensa/xtensa_vectors.S:1240
#5  0x4011530e in s_i2c_send_commands (i2c_master=0x0, ticks_to_wait=0) at /opt/esp/idf/components/esp_driver_i2c/i2c_master.c:471
#6  0x40115627 in s_i2c_transaction_start (i2c_dev=0x3ffd33b8, xfer_timeout_ms=<optimized out>) at /opt/esp/idf/components/esp_driver_i2c/i2c_master.c:625
#7  0x40115727 in s_i2c_synchronous_transaction (i2c_dev=0x3ffd33b8, i2c_ops=0x3ffd29ec, ops_dim=4, timeout_ms=10) at /opt/esp/idf/components/esp_driver_i2c/i2c_master.c:918
#8  0x4011673f in i2c_master_receive (i2c_dev=0x3ffd33b8, read_buffer=<optimized out>, read_size=4, xfer_timeout_ms=10) at /opt/esp/idf/components/esp_driver_i2c/i2c_master.c:1234
#9  0x400ec57e in HDC1080_ReadTempRhLow (f_Temp=0x3ffb5ed8 <PicoSensori+8>, f_Umidita=0x3ffb5edc <PicoSensori+12>) at /workspaces/pico_2023/main/HdcXX80.c:270
#10 0x400ec8b6 in GestioneHDCxx80 (StatusHDCxx80=0x3ffb5ed0 <PicoSensori>) at /workspaces/pico_2023/main/HdcXX80.c:141
#11 0x400e3c06 in task_hwlow (args=0x0) at /workspaces/pico_2023/main/main.c:1674
#12 0x4008f175 in vPortTaskWrapper (pxCode=0x400e3b2c <task_hwlow>, pvParameters=0x0) at /opt/esp/idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139
```

A possible solution is to set `i2c_master->status` to `I2C_STATUS_TIMEOUT` _before_ zeroing `i2c_master->trans_idx` and `i2c_master->cmd_idx`.

Is there a reason why `i2c_master->status` is not read atomically inside the isr (on line 647)? Wouldn't it be better to use `atomic_load` instead?

closed 2025-03-06 by espressif-bot

JD-MLC, 2025-03-17:
Thank you so much gomatteo!!!! The above fix was not enough to fully fix the problem for me. I think it is still correct, but I also had to modify the code below:

#if !SOC_I2C_STOP_INDEPENDENT
    else {
        i2c_operation_t *i2c_operation = &i2c_master->i2c_trans.ops[i2c_master->read_buf_pos];
        portENTER_CRITICAL_ISR(&i2c_master->base->spinlock);
        i2c_ll_read_rxfifo(hal->dev, i2c_operation->data + i2c_operation->bytes_used, i2c_master->read_len_static);
        i2c_ll_read_rxfifo(hal->dev, i2c_master->i2c_trans.ops[i2c_master->read_buf_pos + 1].data, 1);

adding a simple check between the two i2c_ll_read_rxfifo() 

        i2c_ll_read_rxfifo(hal->dev, i2c_operation->data + i2c_operation->bytes_used, i2c_master->read_len_static);
        // Check if the read buffer is allocated.
	if (i2c_master->i2c_trans.ops[i2c_master->read_buf_pos + 1].data == NULL) {
		// Mark the transaction as error so higher layers can detect the failure.
		i2c_master->status = I2C_STATUS_ACK_ERROR;  // maybe replace with a more appropriate error
		portEXIT_CRITICAL_ISR(&i2c_master->base->spinlock);
		return;
	}
        i2c_ll_read_rxfifo(hal->dev, i2c_master->i2c_trans.ops[i2c_master->read_buf_pos + 1].data, 1);

-----------------
I have a board with a few I2C slaves that have bad I2C state machines, so they mess up the bus a lot and I have to reset it every few hundred transactions using a mix of i2c_master_bus_reset() and i2c_master_bus_rm_device() + i2c_del_master_bus() + setup again. Not sure if that is related, but either way, the I2C slaves suck and constantly have timing issues so it is great for testing I2C code! The ESP32C6 as the I2C master was crashing in i2c_ll_read_rxfifo() because it was passed a NULL pointer. It crashed about once every 20K transactions without any changes, every 30-40K with the previous fix, and with my fix I am testing and now have over 100K transactions without crashing.

AxelLin, 2025-04-05:
> good catch！ Thanks！

v5.4 still does not include this fix.