espressif/esp-idf #15447: OpenEth fails to build for ESP32S3

ShawnHymel, 2025-02-22:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

The [https_mbedtls](https://github.com/espressif/esp-idf/tree/v5.4/examples/protocols/https_mbedtls) example should build.

### What is the actual behavior?

The build fails.

### Steps to reproduce.

1. Go into the [https_mbedtls](https://github.com/espressif/esp-idf/tree/v5.4/examples/protocols/https_mbedtls) example.
2. Set configs to enable OpenEth (`idf.py menuconfig` or `sdkconfig.defaults`):

```
CONFIG_EXAMPLE_CONNECT_WIFI=n
CONFIG_EXAMPLE_USE_OPENETH=y
```

3. Set target to `esp32s2`:

```
idf.py set-target esp32s2
```

4. Build:

```
idf.py build
```

### Build or installation Logs.

```plain
root@17e429e45122:/workspace/apps/https_mbedtls# idf.py build
Executing action: all (aliases: build)
Running ninja in directory /workspace/apps/https_mbedtls/build
Executing "ninja all"...
[6/1010] Generating ../../partition_table/partition-table.bin
Partition table binary generated. Contents:
*******************************************************************************
# ESP-IDF Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,24K,
phy_init,data,phy,0xf000,4K,
factory,app,factory,0x10000,1M,
*******************************************************************************
[746/1010] Performing configure step for 'bootloader'
-- Found Git: /usr/bin/git (found version "2.39.5") 
[... 217 lines cut ...]
/opt/toolchains/esp-idf/components/esp_eth/src/openeth/openeth.h:33:37: error: 'ETS_ETH_MAC_INTR_SOURCE' undeclared (first use in this function); did you mean 'ETS_BT_MAC_INTR_SOURCE'?
   33 | #define OPENETH_INTR_SOURCE         ETS_ETH_MAC_INTR_SOURCE
      |                                     ^~~~~~~~~~~~~~~~~~~~~~~
/opt/toolchains/esp-idf/components/esp_common/include/esp_compiler.h:31:25: note: in definition of macro 'unlikely'
   31 | #define unlikely(x)    (x)
      |                         ^
/opt/toolchains/esp-idf/components/esp_eth/src/openeth/esp_eth_mac_openeth.c:394:5: note: in expansion of macro 'ESP_GOTO_ON_FALSE'
  394 |     ESP_GOTO_ON_FALSE(esp_intr_alloc(OPENETH_INTR_SOURCE, ESP_INTR_FLAG_IRAM, emac_opencores_isr_handler, emac, &(emac->intr_hdl)) == ESP_OK, NULL, out, TAG, "alloc emac interrupt failed");
      |     ^~~~~~~~~~~~~~~~~
/opt/toolchains/esp-idf/components/esp_eth/src/openeth/esp_eth_mac_openeth.c:394:38: note: in expansion of macro 'OPENETH_INTR_SOURCE'
  394 |     ESP_GOTO_ON_FALSE(esp_intr_alloc(OPENETH_INTR_SOURCE, ESP_INTR_FLAG_IRAM, emac_opencores_isr_handler, emac, &(emac->intr_hdl)) == ESP_OK, NULL, out, TAG, "alloc emac interrupt failed");
      |                                      ^~~~~~~~~~~~~~~~~~~
/opt/toolchains/esp-idf/components/esp_eth/src/openeth/openeth.h: In function 'openeth_rx_desc':
/opt/toolchains/esp-idf/components/esp_eth/src/openeth/openeth.h:165:1: error: control reaches end of non-void function [-Werror=return-type]
  165 | }
      | ^
/opt/toolchains/esp-idf/components/esp_eth/src/openeth/openeth.h: In function 'openeth_tx_desc':
/opt/toolchains/esp-idf/components/esp_eth/src/openeth/openeth.h:159:1: error: control reaches end of non-void function [-Werror=return-type]
  159 | }
      | ^
cc1: some warnings being treated as errors
[890/1010] Building C object esp-idf/protobuf-c/CMakeFiles/__idf_protobuf-c.dir/protobuf-c/protobuf-c/protobuf-c.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /workspace/apps/https_mbedtls/build/log/idf_py_stderr_output_1438690 and /workspace/apps/https_mbedtls/build/log/idf_py_stdout_output_1438690
```

glmfe, 2026-03-16:
Hello @ShawnHymel,

I tried to reproduce the issue using ESP-IDF v5.4.3 and the latest master branch, but I couldn't encounter any errors. Could you confirm if this is still occurring on your end?

Also, please ensure you've performed a clean build (idf.py fullclean) before recompiling to rule out any caching issues.

ShawnHymel, 2026-03-26:
I'm still seeing the issue on v5.4.0, but it sounds like it's working on v5.4.3. I've released the course and worked around the problem on v5.4.0, so I'm going to close the issue.

closed 2026-03-26 by ShawnHymel