espressif/esp-idf #15448: Cannot enable MBEDTLS_SSL_PROTO_TLS1_3 when MBEDTLS_DYNAMIC_BUFFER=y

AxelLin, 2025-02-22:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Currently MBEDTLS_SSL_PROTO_TLS1_3 depends on !MBEDTLS_DYNAMIC_BUFFER.
But disabling MBEDTLS_DYNAMIC_BUFFER takes too much memory (about 20K) for ESP32.
So due to the limited memory size on esp32, it's almost not possible to support TLS1.3
for existing products using esp32.

I had asked this issue several months ago: https://github.com/espressif/esp-idf/issues/14126#issuecomment-2395275241
However, there is no update even after @AdityaHPatwardhan replied "will soon update regarding dynamic buffers with TLS 1.3". https://github.com/espressif/esp-idf/issues/14126#issuecomment-2490315817 (It was 3 Months ago).

Can someone help to update the status? @mahavirj

AdityaHPatwardhan, 2025-02-23:
Hi @AxelLin Thanks for raising the issue. Apologies for the delayed response on the earlier thread.
We were working on supporting other features/fixes hence could not prioritise this one. We have started working on this and @hrushikesh430 will update on this as we make progress.

AxelLin, 2025-04-04:
> Hi [@AxelLin](https://github.com/AxelLin) Thanks for raising the issue. Apologies for the delayed response on the earlier 

Still no update at all.

AxelLin, 2025-04-15:
@AdityaHPatwardhan How is the status. Looks like nobody is working on this.

AdityaHPatwardhan, 2025-04-15:
@AxelLin Sorry for the delayed reply, Last time when you pinged, we weren't able to take up the task. But I have good update this time.
We were able to complete this feature and are close to merging it in our internal codebase thanks to @Ashish285,
it should be visible publicly in a couple of weeks.

closed 2025-05-06 by espressif-bot