espressif/esp-idf #15449: When uart set 115200 baud, Use <uart_get_baudrate> get baud, it show error

NewNoob2002, 2025-02-22:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.3.2

### Espressif SoC revision.

ESP32-Down

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-WROOM-EU

### Power Supply used.

USB

### What is the expected behavior?

It should be 115200

### What is the actual behavior?

get the baud rate is 115201, and other rates is right, such 19200, 38400, 56700 etc.

My configure function:

`void HAL::ConfigRadioUart(int baud, uint8_t tx_io, uint8_t rx_io)
{
    uart_config_t uart_config;
    memset(&uart_config, 0, sizeof(uart_config_t));
    uart_config = {
        .baud_rate = baud,
        .data_bits = UART_DATA_8_BITS,
        .parity = UART_PARITY_DISABLE,
        .stop_bits = UART_STOP_BITS_1,
        .flow_ctrl = UART_HW_FLOWCTRL_DISABLE,
    };

    ESP_ERROR_CHECK(uart_param_config(UART_NUM_1, &uart_config));

    ESP_ERROR_CHECK(uart_set_pin(UART_NUM_1, tx_io, rx_io, UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE));

    ESP_ERROR_CHECK(uart_driver_install(UART_NUM_1, 2048, 2048, 10, &uart_radio_queue, 0));
}

ConfigRadioUart(115200, 14, 13);
`

[image]

### Steps to reproduce.

1. just set baud driver

### Debug Logs.

```plain

```

Bruce297, 2025-02-25:
Hi, in calculating the baud rate, it is normal and unavoidable to sometimes have a small error

suda-morris, 2025-02-28:
This function will always reflect the **real baud rate** that is configured in the hardware.  There will be a slight deviation from the actual user-configured value.

We should update the API documentation to tell the truth.

closed 2025-04-02 by espressif-bot