espressif/esp-idf #15450: [SD SPI] Failed to initialize the card witn 0x106, but 64gb - OK

sazanof, 2025-02-22:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

OS: Windows, Linux
IDF: 5.3, 5.4 (test with different)
Board: Wroom boards and custom boards
Log not ok: https://gist.github.com/sazanof/8d0a382d95f3451c15ec6ea746232306
Log with 64gb (success) https://gist.github.com/sazanof/c7a90758752e012375b5907d137f2f43
Topic on esp forum: https://www.esp32.com/viewtopic.php?p=144121

Hello! In my project, I use storing the web interface on a memory card. I couldn't get ESP32 to recognize the card for a very long time. I've tried a lot of options and variants: different ESP32 boards, custom boards, different wires length, pullup resistors, different `max_freq_khz`...

regular microsd 1gb - error (type not supported)
16GB microsdhc memory card - CRC verification error
4gb MICROSDHC U-1 memory card - everything is fine
**64gb Samsung evo microsd U-1 - everything is fine**

[image]

So, cards < 64gb not working. Cards 64gb - works fine!

 I installed Arduino with ESP32 boards and check SD SPI sketch example (https://github.com/espressif/arduino-esp32/tree/master/libraries/SD). **ALL cards WORKS FINE! 1,4,8,16,32 GB.**

Thus, I concluded that I was probably initializing the memory card incorrectly in the code example esp-idf, although I was using an example from your official 5.4 branch (https://github.com/espressif/esp-idf/blob/master/examples/storage/sd_card/sdspi/README.md).

I searched the entire official forum, the entire github - yes, there are similar problems, but they are also floating in nature and there are no solutions to most of the problems. Besides, there's probably a bug somewhere in IDF code or example code, since sd cards work correct through another framework (arduino). 

Can you help to solve this problem? 

More information here https://www.esp32.com/viewtopic.php?p=144121#p144121

igrr, 2025-02-26:
Hi @sazanof,

Thanks for attaching the [log](https://gist.github.com/sazanof/8d0a382d95f3451c15ec6ea746232306) for one of the failing cards. I can confirm that we've seen this issue being reported earlier, although we didn't have the specific card which could reproduce this issue (CMD59 returns "not supported").

Do you get the same error from other non-working cards? If the errors are different, could you please attach or link the logs for those other cards as well? We will then split this issue into one issue per root cause.

igrr, 2025-02-26:
Thank you for the additional logs! I'll update the issue once we have any fixes to share.

igrr, 2025-02-27:
We found a few cards exhibiting similar issues, here are some conclusions so far:

- For the 0x106 ("not supported") error related to sdmmc_send_cmd_crc_on_off: we have a draft patch for this.
- For the 0x109 (CRC) error, this is a similar situation to https://github.com/espressif/esp-idf/issues/14000, currently investigating what is the best way to fix this.
- For the 0x107 ("timeout") error: found one card with the same behavior, still debugging it. The funny thing is that another card of the same model from the same batch works just fine...
- There are also a few cards which fail to initialize with 0x108 or 0x109 errors (data corruption) only on ESP32. On newer chips, such as ESP32-S3, they work fine. This needs further investigation, probably there is some timing difference in SPI mode 0 between these chips.

sazanof, 2025-02-27:
@igrr 

It's great that the problem has been confirmed! I will wait and keep an eye on patches and fixes. If you need anything from me, please let me know!

d4rkmen, 2025-03-19:
> * For the 0x106 ("not supported") error related to sdmmc_send_cmd_crc_on_off: we have a draft patch for this.

show us what you have got :-)

Darmann2, 2025-06-18:
I have the same issue with sdmmc_sd: sdmmc_init_spi_crc: sdmmc_send_cmd_crc_on_off returned 0x106
And I am searching since december for a solution. We have a hardware we build 500 times and we always try to buy the same type of sd-cards - but they are not available some times.
Last year in december we got the first time the 106 error.
Now I bought 10 different sd-cards (PHILIPS) - and they have the same issue!
The only way to solve the problem, is to ignore the 106. And the sd-card is working.

`esp_err_t sdmmc_init_spi_crc(sdmmc_card_t* card)`
`{`
`    /* In SD mode, CRC checks of data transfers are mandatory and performed`
`     * by the hardware. In SPI mode, CRC16 of data transfers is optional and`
`     * needs to be enabled.`
`     */`
`    assert(host_is_spi(card));`
`    esp_err_t err = sdmmc_send_cmd_crc_on_off(card, true);`
`    if (err != ESP_OK) {`
`        ESP_LOGE(TAG, "%s: sdmmc_send_cmd_crc_on_off returned 0x%x", __func__, err);`
_**`        if (err == _0x106)`**_
_**`        	err = ESP_OK;`**_
`        return err;`
`    }`
`    return ESP_OK;`
`}`

If I modify the file esp-idf\components\sdmmc\sdmmc_sd.c (2 italic lines above) then everything works!
So can you please help me, because I don't want to modify everytime the IDF.

akshar001, 2025-09-08:
@igrr any update on 0x107? We're seeing multiple issues related to it.

pavlohamov, 2025-10-07:
UPD: 
device #2 showed interesting behavior compared to dev#1 (same batch)
SD-SPI:  init fails with err 0x106 regardless logs state
SDMMC: init successfull regardless logs state

mhasler-inno, 2025-10-27:
We are facing the same issue with 4 GB cards by Intenso. So far none of them has worked for us without the workaround mentioned above.

Correct me if I'm wrong, but as I understand it the problem seems to be that low-cost consumer cards are typically meant to run in SD mode anyway (where CRC is mandatory), so I'm guessing that many manufactures just shortcut the specification and hard-wire the CRC, hence they don't implement the command in question (since it would be impossible to disable it anyway).

Whatever the reason, since this issue affects many cards and since CRC is optional in SPI mode, it would be reasonable to have this as an option in the IDF, as well. I think the simplest solution would be to add an optional flag to sdmmc_host_t, e.g. SDMMC_HOST_FLAG_NO_SPI_CRC to allow skipping the CRC initialization.

Since we urgently need such an option I'd gladly implement it, provided there is a realistic chance it will be merged anytime soon.

makerinchina-iot, 2025-11-08:
I have the same issue with sdmmc_init_spi_crc: sdmmc_send_cmd_crc_on_off returned 0x106, but run this function the second time, it success:

[image]

masterhard, 2025-11-13:
I am not quite familiar with sd/api nuances, but here is what I read in sdmmc_sd.c:

```
    /* In SD mode, CRC checks of data transfers are mandatory and performed
     * by the hardware. In SPI mode, CRC16 of data transfers is optional and
     * needs to be enabled.
     */
    assert(host_is_spi(card));
    esp_err_t err = sdmmc_send_cmd_crc_on_off(card, true);
```
So if it is optional and it fails, I guess we just need to log a warning, or maybe even an info about crc16 init failure and keep going, aren't we?

Technically, it could be a good idea to make a Kconfig option to make CRC checks mandatory for those who care about CRC working more than about being able to support SD cards that do not have it.