espressif/esp-idf #15451: ESP32-S3 USB peripheral freezes after light sleep at 40MHz

dbeinder, 2025-02-23:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4 release

### Espressif SoC revision.

ESP32-S3 (QFN56) revision 0.2

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

YD-ESP32-23 (ESP32-S3-WROOM-1)

### Power Supply used.

USB

### What is the expected behavior?

The USB peripheral works normally after returning from light sleep

### What is the actual behavior?

With 40MHz CPU frequency, USB-OTG works fine. However, after entering light-sleep and resuming, the USB peripheral is frozen. It doesn't matter if the peripheral was initialized or not. If uninitialized the init function stalls - if it was already initialized, the peripheral just stops handling any USB events.

Briefly switching to 80MHz and back to 40MHz after wakeup "fixes" the issue.

### Steps to reproduce.

Here is a reproducuction of the issue using the `tusb_hid` sample: https://github.com/dbeinder/esp32s3-tusb-hid diff: https://github.com/dbeinder/esp32s3-tusb-hid/commit/4a432236ddcacdad773d30385560fe8dc9f1323a

`tinyusb_driver_install()` will stall in `reset_core()` in dwc2_common.c. See watchdog backtrace.

Uncommenting the switch to 80MHz and back to 40MHz makes the initialization work again.

### Debug Logs.

```plain
I (302) example: Core Clock: 40 MHz
I (302) example: Entering Light Sleep...
I (312) example: Resumed (0) ESP_OK
I (312) example: USB initialization
W (322) tusb_desc: No Device descriptor provided, using default.
I (322) tusb_desc:
┌─────────────────────────────────┐
│  USB Device Descriptor Summary  │
├───────────────────┬─────────────┤
│bDeviceClass       │ 0           │
├───────────────────┼─────────────┤
│bDeviceSubClass    │ 0           │
├───────────────────┼─────────────┤
│bDeviceProtocol    │ 0           │
├───────────────────┼─────────────┤
│bMaxPacketSize0    │ 64          │
├───────────────────┼─────────────┤
│idVendor           │ 0x303a      │
├───────────────────┼─────────────┤
│idProduct          │ 0x4004      │
├───────────────────┼─────────────┤
│bcdDevice          │ 0x100       │
├───────────────────┼─────────────┤
│iManufacturer      │ 0x1         │
├───────────────────┼─────────────┤
│iProduct           │ 0x2         │
├───────────────────┼─────────────┤
│iSerialNumber      │ 0x3         │
├───────────────────┼─────────────┤
│bNumConfigurations │ 0x1         │
└───────────────────┴─────────────┘

E (10282) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (10282) task_wdt:  - IDLE0 (CPU 0)
E (10282) task_wdt: Tasks currently running:
E (10282) task_wdt: CPU 0: main
E (10282) task_wdt: CPU 1: IDLE1
E (10282) task_wdt: Print CPU 0 (current core) backtrace

Backtrace: 0x42010E36:0x3FC94C90 0x4201124C:0x3FC94CB0 0x40377C1E:0x3FC94CE0 0x4200D73F:0x3FC99A90 0x4200D789:0x3FC99AB0 0x4200D8BF:0x3FC99AD0 0x4200D2E6:0x3FC99AF0 0x4200B8ED:0x3FC99B10 0x4200C426:0x3FC99B30 0x4200A898:0x3FC99B60 0x4200A748:0x3FC99BF0 0x4201DCB3:0x3FC99C70 0x4037BD29:0x3FC99CA0
--- 0x42010e36: task_wdt_timeout_handling at C:/Users/David/esp/v5.4/esp-idf/components/esp_system/task_wdt/task_wdt.c:434
0x4201124c: task_wdt_isr at C:/Users/David/esp/v5.4/esp-idf/components/esp_system/task_wdt/task_wdt.c:507
0x40377c1e: _xt_lowint1 at C:/Users/David/esp/v5.4/esp-idf/components/xtensa/xtensa_vectors.S:1240
0x4200d73f: reset_core at C:/proj/esp_debug/tusb_hid/managed_components/espressif__tinyusb/src/portable/synopsys/dwc2/dwc2_common.c:52 (discriminator 1)
0x4200d789: phy_fs_init at C:/proj/esp_debug/tusb_hid/managed_components/espressif__tinyusb/src/portable/synopsys/dwc2/dwc2_common.c:76
0x4200d8bf: dwc2_core_init at C:/proj/esp_debug/tusb_hid/managed_components/espressif__tinyusb/src/portable/synopsys/dwc2/dwc2_common.c:209
0x4200d2e6: dcd_init at C:/proj/esp_debug/tusb_hid/managed_components/espressif__tinyusb/src/portable/synopsys/dwc2/dcd_dwc2.c:403
0x4200b8ed: tud_rhport_init at C:/proj/esp_debug/tusb_hid/managed_components/espressif__tinyusb/src/device/usbd.c:504
0x4200c426: tusb_rhport_init at C:/proj/esp_debug/tusb_hid/managed_components/espressif__tinyusb/src/tusb.c:71
0x4200a898: tinyusb_driver_install at C:/proj/esp_debug/tusb_hid/managed_components/espressif__esp_tinyusb/tinyusb.c:72
0x4200a748: app_main at C:/proj/esp_debug/tusb_hid/main/tusb_hid_example_main.c:198
0x4201dcb3: main_task at C:/Users/David/esp/v5.4/esp-idf/components/freertos/app_startup.c:208
0x4037bd29: vPortTaskWrapper at C:/Users/David/esp/v5.4/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139
```

dbeinder, 2025-10-06:
Any progress on this?

Additional info I've found: ESP32-S3 has DWC2 v4.00a. The specific operation that hangs, is waiting for `GRSTCTL_CSRST` to auto clear. According to the databook, the block should work down to 30MHz AHB clock.

tore-espressif, 2026-08-17:
@dbeinder thank you very much for your investigation! We will check the fix on all remaning USB enabled targets publish a fix, thanks

closed 2026-09-16 by espressif-bot