espressif/esp-idf #15452: ESP32-S3 DeepSleep Touch Wakeup from any pin

DarkZeros, 2025-02-23:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi, I have a WIP prototype on a ESP32-PICO-D4, that uses touch wake from deep sleep.
It has 4 pins/pads (touch buttons), and it wakes from deep sleep under any touch on them.

Then, I simply use `touch_ll_read_trigger_status_mask()` to detect the firing one and perform the 4 different actions.

Due to lack of GPIOs that i needed to extend my design, I decided that ESP32-S3 was the only valid option, since it is the only new generation chip with Touch pad support, as well as more GPIOs.

---------------------------------------------------------

However, I am unable to make this wakeup form deep sleep feature work on the 4 pads, only 1 pad/pin wakes the chip, the last one configured. The rest don´t wake up the chip from deep sleep. Calling the read trigger_status_mask() gives always 0, when this wakeup occurs.

I read here: https://esp32.com/viewtopic.php?t=41970
That there is a way to make it work, however I failed making it work.
I tried this with no sucess: 
`esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_ON);`

This is for me quite a trap, since the 14 touch pads are totally useless for me, if I can´t use them in deep sleep.
My ESP is all the time in deep sleep, until a user input occurs (on any pin!).

Any help to fix it using more power in deep sleep is welcomed, as well as alternatives (I can change to another chip still, since i am prototyping)

L-KAYA, 2025-02-24:
Hi @DarkZeros ! Seems you are still using the legacy touch driver, FYI, the new touch driver is ready on the master branch, please take a try! (see the [programming guide](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/peripherals/cap_touch_sens.html))

About the issue you met, it actually a limitation of the legacy driver. 

As you know the touch hardwares are different on ESP32 and S3, and the S3 Touch has more features, 'only can wake-up by one touch pad' is one of these features, it allows the touch power domain (i.e, `RTC_PERIPH` power domain) shutdown during the deep sleep, which can help to save more power consumption. However, this feature is enabled by default, and not likely to disable by a public API. 

But in the new driver, you can disable it by setting [touch_sleep_config_t::deep_slp_allow_pd](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/peripherals/cap_touch_sens.html#_CPPv4N20touch_sleep_config_t17deep_slp_allow_pdE) to false.

DarkZeros, 2025-02-26:
> Hi [@DarkZeros](https://github.com/DarkZeros) ! Seems you are still using the legacy touch driver, FYI, the new touch driver is ready on the master branch, please take a try! (see the [programming [...]

Hi! Thanks for the guidance, it actually helped me to know the feature was possible and start testing different things.

Right now I am in v5.4, and mostly use "touch_ll" and similar functions to access low level components, and I can do that from deep sleep stubs. I will not move to master and reimplement my code, for reasons....

However, that being said, I had a look at the master code implementation, and found out a way to make it work in my current code! :)

For anyone finding this in the future:
The trick is actually "disable" the sleep touch functionality:
`touch_pad_sleep_channel_enable(pad, false);`
And then enable 
`esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_ON);`

This way the old functionality is restored, any pad enabled will wake up the device, and the `touch_ll_read_trigger_status_mask()` will return the pad that woke the device.

If the touch sleep pad is enabled, the last pad setup will return "0" in the mask, however, the other pads will still work as expected and return a valid mask. But in my use case sometimes it got stuck without touch working, so i recomend disabling it enitrely.

L-KAYA, 2025-02-27:
@DarkZeros Yes, I'm just about to add the workaround that turning on the `ESP_PD_DOMAIN_RTC_PERIPH` power domain during the deep sleep. But you have figured out by yourself! Your step is totally correct!

And the new touch driver will be released in v5.5, if you are not convenient to migrate to v5.5, maybe the new doc can also help you because it contains more details~