espressif/esp-idf #15453: Will tasks with the same priority undergo time-slice rotation? 同优先级任务是否会时间片轮转？

goukisun, 2025-02-23:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

espidf-v5.1.1
From the document, it is seen that FreeRTOS supports time-sliced round-robin scheduling for tasks with the same priority. However, in the actual running of the demo, it will not rotate.
Do I need any special settings?

SoucheSouche, 2025-02-24:
Hi @goukisun,
In ESP-IDF preemption and time-slicing are always enabled, it is not possible to disable one or the other. Can you tell me what demo you are using? There is no example provided in ESP-IDF demonstrating this feature.

goukisun, 2025-02-24:
@SoucheSouche 
Sorry, I made a mistake

closed 2025-02-25 by SoucheSouche