espressif/esp-idf #15457: Pairing device BLE with android and IOS

ElenaG391, 2025-02-24:
Hi all,

I'm trying to pairing a device with android/IOS. The device sends me a PIN:XXX or PIN:XXXX, the device has a limitation with the number of the digits, it send just 3 or 4digits. 
I have to include 000XXX or 00XXXX for pairing, we must have 6digits to connect it.

Could you please let me know how I need to change on the code to accept just 3/4 digits to connect with the bluetooth? is that possible? just we will use the numbers that the device send us: XXX or XXXX. without "0".

is it possible set the 4-digit passkey in ESP-IDF?

thanks

weiyuhannnn, 2025-02-26:
Hello @ElenaG391, Thanks for choosing our Espressif product! May I know which chip and BLE host do you use? This information will help us better identify the issue you reported. Thank you!

ElenaG391, 2025-02-26:
Oh yes! Sorry, I'm using ESP32-MINI-1U and I'm trying to connect iPhone android and IOS.

thanks

esp-zhp, 2025-02-28:
@ElenaG391 
Are you using Classic Bluetooth or BLE? According to the protocol, BLE only supports a 6-digit PIN, while Classic Bluetooth supports a 4-digit PIN. Could you confirm whether you're actually using Classic Bluetooth?

esp-zhp, 2025-02-28:
@ElenaG391 Which host are you using—NimBLE host or Bluedroid host?

ElenaG391, 2025-02-28:
@esp-zhp thanks for your answer. we are using BLE and Nimble host.
so I understand that we can't use 4 digits with BLE and Nimble host, is that correct?

If I change my code to use the classic Bluetooth, can I still use the Bluetooth Approved Logo? I understand that yes, it's Bluetooth as well....

thanks!

rahult-github, 2025-02-28:
Hi @ElenaG391 ,

> so I understand that we can't use 4 digits with BLE and Nimble host, is that correct? 

Yes. This is because the spec mandates 6 digits.  You need to prepend "0s" in case the pin is lesser than 100000 in value 

> If I change my code to use the classic Bluetooth, can I still use the Bluetooth Approved Logo?

 Classic Bluetooth is BR/EDR (and not BLE) . Is your usecase BLE or BR/EDR ?  Anyways, classic bluetooth is also from BTSIG, so it is bluetooth approved.

ElenaG391, 2025-03-03:
I'm using BLE now. Could I start to use classic Bluetooth and we could still use the bluetooth logo, is that correct? 
I think that we will use classic bluetooth because we would like to use just 4 digits. thanks

rahult-github, 2025-03-03:
Hi @ElenaG391  ,  The use case of BLE and BR/EDR are different. However, if you are ok to use classic bluetooth that satisfies your requirement, then you can choose so. Please note nimble stack is BLE only stack . If you plan to use classic bluetooth, you will need to use bluedroid stack.

rahult-github, 2025-03-07:
Closing issue. Feel free to reopen in case of further queries.

closed 2025-03-07 by rahult-github