espressif/esp-idf #15458: Message queued with esp_mqtt_client_enqueue() are delayed

dwmw2, 2025-02-24:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.5

### Espressif SoC revision.

ESP32-S3

### Operating System used.

Linux

### How did you build your project?

Other (please specify in More Information)

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

Using ESPHome with the `idf_send_async` option to make it use `esp_mqtt_client_enqueue()`, I expected it to send my queued messages promptly.

(I set this option because I didn't like it when `esp_mqtt_client_publish()` blocks for 20 seconds when the network is down).

### What is the actual behavior?

Each time around the loop in `esp_mqtt_task()`, it attempts to send precisely *one* message, then goes to sleep for another second before sending the next, leading to a large backlog and lost messages.

### Steps to reproduce.

Set `idf_send_async` in an ESPHome configuration and watch how long it takes for MQTT messages to be sent.

### Debug Logs.

```plain

```

### More Information.

This patch helps a bit, by flushing the whole queue every time. But there's still up to a second (MQTT_POLL_READ_TIMEOUT_MS) before each message is sent. Can we *wake* the task when a message is queued?

```diff
--- components/mqtt/esp-mqtt/mqtt_client.c	2025-02-24 15:40:44.868028678 +0000
+++ components/mqtt/esp-mqtt/mqtt_client.c~	2024-10-31 17:03:14.000000000 +0000
@@ -1662,13 +1662,16 @@ static void esp_mqtt_task(void *pv)
 
 
             // resend all non-transmitted messages first
-            outbox_item_handle_t item = outbox_dequeue(client->outbox, QUEUED, NULL);
+            outbox_item_handle_t item;
+	    do {
+	    item = outbox_dequeue(client->outbox, QUEUED, NULL);
             if (item) {
                 if (mqtt_resend_queued(client, item) == ESP_OK) {
                     if (client->mqtt_state.pending_msg_type == MQTT_MSG_TYPE_PUBLISH && client->mqtt_state.pending_publish_qos == 0) {
                         // delete all qos0 publish messages once we process them
                         if (outbox_delete_item(client->outbox, item) != ESP_OK) {
                             ESP_LOGE(TAG, "Failed to remove queued qos0 message from the outbox");
+			    break;
                         }
                     }
                     if (client->mqtt_state.pending_publish_qos > 0) {
@@ -1679,7 +1682,7 @@ static void esp_mqtt_task(void *pv)
                         }
 #endif
                     }
-                }
+                } else break;
                 // resend other "transmitted" messages after 1s
             } else if (has_timed_out(last_retransmit, client->config->message_retransmit_timeout)) {
                 last_retransmit = platform_tick_get_ms();
@@ -1691,9 +1694,10 @@ static void esp_mqtt_task(void *pv)
                             esp_mqtt5_increment_packet_counter(client);
                         }
 #endif
-                    }
+                    } else break;
                 }
-            }
+            } else break;
+	    } while (item /* && this thread doesn't need to voluntarily yield */);
 
             if (process_keepalive(client) != ESP_OK) {
                 break;

```

dwmw2, 2025-02-25:
Perhaps `max_poll_timeout()` should return 10ms instead of `max_timeout` if there are any `QUEUED` messages in the outbox?

dwmw2, 2025-02-25:
This helps to flush the already-queued messages without waiting a whole second between each one, but it doesn't solve the problem that it's a whole second before the *first* message is sent, after calling `esp_mqtt_client_publish()`. I can't see a way in FreeRTOS to wake a thread that's waiting in `esp_transport_poll_read()` / `select()`.

Using `esp_mqtt_client_publish()` to attempt to publish it directly and fall *back* to queuing doesn't seem to be an option because that can block for a *long* time. If there was a way to use `esp_mqtt_client_publish()` with a much shorter network timeout (and without aborting the connection if it does time out) then that might work?

```diff
--- mqtt_client.c.orig	2025-02-25 10:41:20.280544113 +0000
+++ mqtt_client.c	2025-02-25 10:45:39.571513453 +0000
@@ -1577,6 +1577,7 @@ static void esp_mqtt_task(void *pv)
     client->state = MQTT_STATE_INIT;
     xEventGroupClearBits(client->status_bits, STOPPED_BIT);
     while (client->run) {
+        uint32_t next_timeout = MQTT_POLL_READ_TIMEOUT_MS;
         MQTT_API_LOCK(client);
         run_event_loop(client);
         // delete long pending messages
@@ -1665,6 +1666,7 @@ static void esp_mqtt_task(void *pv)
             outbox_item_handle_t item = outbox_dequeue(client->outbox, QUEUED, NULL);
             if (item) {
                 if (mqtt_resend_queued(client, item) == ESP_OK) {
+                    next_timeout = 10;
                     if (client->mqtt_state.pending_msg_type == MQTT_MSG_TYPE_PUBLISH && client->mqtt_state.pending_publish_qos == 0) {
                         // delete all qos0 publish messages once we process them
                         if (outbox_delete_item(client->outbox, item) != ESP_OK) {
@@ -1693,7 +1695,7 @@ static void esp_mqtt_task(void *pv)
 #endif
                     }
                 }
-            }
+            } // else reduce next_timeout if it *will* time out before the max. */
 
             if (process_keepalive(client) != ESP_OK) {
                 break;
@@ -1704,7 +1706,7 @@ static void esp_mqtt_task(void *pv)
                 ESP_LOGD(TAG, "Refreshing the connection...");
                 esp_mqtt_abort_connection(client);
                 client->state = MQTT_STATE_INIT;
-            }
+            } // else reduce next_timeout if it *will* time out before the max. */
 
             break;
         case MQTT_STATE_WAIT_RECONNECT:
@@ -1733,7 +1735,7 @@ static void esp_mqtt_task(void *pv)
         }
         MQTT_API_UNLOCK(client);
         if (MQTT_STATE_CONNECTED == client->state) {
-            if (esp_transport_poll_read(client->transport, max_poll_timeout(client, MQTT_POLL_READ_TIMEOUT_MS)) < 0) {
+            if (esp_transport_poll_read(client->transport, max_poll_timeout(client, next_timeout)) < 0) {
                 ESP_LOGE(TAG, "Poll read error: %d, aborting connection", errno);
                 esp_mqtt_abort_connection(client);
             }

```

euripedesrocha, 2025-03-17:
Hi @dwmw2 thanks for reporting the issue. 

Do adjust CONFIG_MQTT_POLL_READ_TIMEOUT_MS from Kconfig help improve the system behavior for your scenario?

For the async scenarios we have `esp_mqtt_client_enqueue`, that unfortunately can also block in poor network scenarios. But It will only happen if the client is running the loop. If blocked in waiting for read path the enqueue function will be possible to use. 

We don't process more than one message per loop iteration to avoid holding the client mutex for too long.  I'm assuming that you are publishing very often, could you describe your system so I can understand if we have some workaround for your context or if we need to add something to the component. 

Could you describe the scenario you get the 20 s block with network disconnected?

euripedesrocha, 2025-03-18:
@dwmw2 we are aware of the `esp_mqtt_client_enqueue` blocking problem, and I believe you are also aware of it from the mention from your side on that issue. While that would be fixed later, It will probably not be a complete solution for your case given that the dispatching from the outbox queue will always have some delay, and that is by design. I'm trying to find a workaround for your issue and identify some improvements for such scenarios without compromising the common requirements and library operation.

One possible change is to enable the flush of the queue if enabled by the user. So add a new entry on the config struct to enable flushing the queue when needed. Similar to the solution you suggested, but controlled to keep current behavior. 

As you pointed, we need to balance the load of the client with the retries and accommodate scenarios with poor network connections.

dwmw2, 2025-03-18:
Thanks, @euripedesrocha.

Yes, I'm aware of the `esp_mqtt_client_enqueue` blocking problem, but it seems that #13078 still has the wrong title. Please could you fix it to say _"esp_mqtt_client_enqueue sometimes blocks for 10+ seconds"_ (not milliseconds).

Despatching from the outbox queue doesn't *have* to have a significant delay. What if the thread waited not just in `esp_transport_poll_read()` as it does at the moment, but could also be woken directly by `esp_mqtt_client_enqueue()` when a message has been added to the queue?

I'm thinking of a design where there's a separate queue (or perhaps even a [Queue](https://www.freertos.org/Documentation/02-Kernel/04-API-references/06-Queues/00-QueueManagement))_ between `esp_mqtt_client_enqueue()` and `esp_mqtt_task()`, so the former doesn't *have* to take the main MQTT API lock and thus doesn't have to block for multiple seconds. All it would do is put a new item on the queue and wake the `esp_mqtt_task()` to consume it.

Then the loop in `esp_mqtt_task()` would take the item off the queue and transfer it into its own outbox. Couple that with some variant of my hack above so that `esp_mqtt_task()` doesn't wait a whole second between sending each item from the outbox, but actually sends them a bit faster than that.

That should actually be fairly simple. I'd hack up a proof of concept but I don't know FreeRTOS well enough to implement the part which waits for *either* the transport to be readable, *or* the Queue (or other form of queue/semaphore) to wake up.

euripedesrocha, 2025-03-18:
@dwmw2 The solution you propose is something you can achieve by providing your own outbox implementation and sharing the queue with your system. To have it as a general solution might have some issues like the msg id that needs to be returned when enqueueing the message. 

What makes this a not so trivial task is that the whole library was designed considering that it would be locked for the whole processing. While it makes easier to consider the potential concurrency problems by avoiding them entirely, it makes hard to refactor to accomplish the async nature that `esp_mqtt_client_enqueue` must have. 

Since this is a recurring issue in the library, I'll prioritize the work on this.  As I mention, the original design that we assumed would fix the problem rose other issues in testing and question in review, hence it was abandoned.

dwmw2, 2025-03-18:
> The solution you propose is something you can achieve by providing your own outbox implementation and sharing the queue with your system. 

Hm, I thought I'd looked at that (basically the idea being just to copy `mqtt_outbox.c`, use a *separate* internal lock for the outbox alone, and have a way for the system to quietly insert its *own* messages into the outbox directly).

But things like `outbox_enqueue()` require visibility into data structures like `outbox_message_t`, so even that doesn't work very nicely. And then there's the fact that `mqtt_client_enqueue_publish()` does a whole bunch of stuff to generate the data that even gets passed into `outbox_enqueue()`.

And it *still* doesn't solve the problem mentioned in this ticket, that `esp_mqtt_task()` can't be woken up directly to process the messages from the outbox, and would still take its sweet time sending one message a second.

But yes, this *is* fairly close to the solution I came up with for ESPHome in https://github.com/esphome/esphome/pull/8325 — just put the messages (`{topic, message, qos, retain flag}`) into a queue from the 'real time' loop, and have a separate thread pull them out and call `esp_mqtt_client_publish()` synchronously.

euripedesrocha, 2025-03-19:
>And it still doesn't solve the problem mentioned in this ticket, that esp_mqtt_task() can't be woken up directly to process the messages from the outbox, and would still take its sweet time sending one message a second.

This isn't exactly a problem but a design decision. If you are doing an async operation, we assume that the message can be published later. You have control on the load you intend to have on your system by controlling the network related timeouts and task priority. If you want the message to be published right away, there is `esp_mqtt_client_publish`. 

I understand that in scenarios of a poor network this imposes some blocks on the system, but we always prioritize that the process is completed and messages are sent, and the client is always in a consistent state. 

I can add the option to the config to process all message to be published in the loop, and a PR from your side is welcomed as well. But this needs to be opt in and keep the current behavior as default.

dwmw2, 2025-03-19:
Ah, the perennial 'bug' vs. 'feature' debate. Which philosophically often ends up boiling down to how well-documented the behaviour in question is.

To be clear, the behaviour in question is that messages enqueued with `esp_mqtt_client_enqueue()` are delayed for a period of up to `MQTT_POLL_READ_TIMEOUT_MS`, and the maximum rate of transmission is `1000 / MQTT_POLL_READ_TIMEOUT_MS` messages per second.

But `MQTT_POLL_READ_TIMEOUT_MS` is documented as _"Timeout when polling underlying transport for read"_. And the documentation for `esp_mqtt_client_enqueue()` does say 'later' but makes no mention of a limit of one message per second.

If you want to call it a design decision, I think we have to fix the documentation accordingly, don't we?

> I can add the option to the config to process all message to be published in the loop, and a PR from your side is welcomed as well. But this needs to be opt in and keep the current behavior as default.

Rather than hogging CPU by sending *all* queued messages in the loop, my hack above was just setting the timeout to something like 10ms so that the thread would wake sooner and send the next message in a more reasonable amount of time. Is that something that could be refined and made the default?

The other thing that would still be missing is a way to wake the thread when a message is first enqueued, so that first message isn't delayed for a second.

The main thing I'm trying to avoid is having to have *two* threads, as in my current workaround. Here's another idea which might work... extract the core of the `esp_mqtt_task()` loop (from the `MQTT_API_LOCK()` to the `MQTT_API_UNLOCK()`) into a separate function. The application can (optionally) *disable* the normal thread that runs `esp_mqtt_task()`, and has to invoke that newly exported function regularly instead.

Then the application's own thread can suck messages off its *own* nonblocking queue (like that one I implemented for ESPHome), queue them up with `esp_mqtt_client_enqueue()`, then immediately call into the old `esp_mqtt_task()` core function to process them.

euripedesrocha, 2025-03-19:
@dwmw2 thanks for the suggestions for the improvements. 

There is no debate regarding the async nature of the publishing of `esp_mqtt_client_enqueue` function being a bug. It's async, and it's to be dispatched later. For publishing directly, we have `esp_mqtt_client_publish`. 
If you need shorter interval when processing the queue, adjust the timeouts accordingly. The defaults are what we have for **typical** use cases. 

The proposed solution of processing all unpublished messages on each loop will only hog the CPU depending on the frequency of your published messages. 

I believe we are discussing multiple problems at once:
- `esp_mqtt_client_enqueue` is async and there will always have a time interval, max 1s in default settings for first message. 
- `esp_mqtt_client_enqueue` has a bug that makes it block when it shouldn't, I will work to fix it. 
- Waking up from a different source would require a more complex redesign, which I don't see the case for. Sorry, we need to address the more common cases, if this start to be requested more I will consider how to solve the issue. But I believe that `esp_mqtt_client_publish` is the current solution. I will still spend some time thinking about it, to see if I can find an easy solution that could be adopted.
- Extracting the core loop of the mqtt client is something that we would like to do but the first refactor attempt in that direction wasn't successful due to many concerns about potential concurrency bugs and the test needed. It is still on our future plans to do it. 

In summary:
- I will apply the documentation improvements
- Provide the opt-in flush unpublished to config
- Work on the enqueue fix. 

To fix all it will take us some time and the regular IDF workflow to be available on idf branches. 

I will gladly review any PR from your side, either here on [esp-mqtt](https://github.com/espressif/esp-mqtt)  repository.