espressif/esp-idf #15460: Inquiry Regarding Malfunction of I2S API (i2s_channel_reconfig_std_gpio)

illusionkiller, 2025-02-24:
**Build Environment:** IDF release-5.4, Windows + VSCode

**Test Environment:** M5Stack Core2

**Test Code Snippet:**
`    esp_codec_dev_handle_t codec_dev = bsp_audio_codec_speaker_init();
    i2s_std_gpio_config_t std_gpio_cfg = {
        .mclk = I2S_GPIO_UNUSED, // some codecs may require mclk signal, this example doesn't need it
        .bclk = I2S_GPIO_UNUSED,
        .ws = I2S_GPIO_UNUSED,
        .dout = I2S_GPIO_UNUSED,
        .din = I2S_GPIO_UNUSED,
    };
    if (codec_dev)
    {
        esp_codec_dev_sample_info_t fs = {
            .sample_rate = 44100,
            .channel = 1,
            .bits_per_sample = 16,
        };
        std_gpio_cfg.bclk = GPIO_NUM_12;
        std_gpio_cfg.ws = GPIO_NUM_0;
        std_gpio_cfg.dout = GPIO_NUM_2;
        i2s_channel_disable(bsp_audio_get_codec_tx_channel());
        // ESP_ERROR_CHECK(i2s_channel_reconfig_std_gpio(bsp_audio_get_codec_tx_channel(), &std_gpio_cfg));
        esp_codec_dev_open(codec_dev, &fs);
        esp_codec_dev_set_out_vol(codec_dev, 100);

        extern const unsigned char music[120264];
        unsigned char *buffer = (unsigned char *)calloc(1, 120264);
        memcpy(buffer, music, 120264);
        size_t was_write_length = esp_codec_dev_write(codec_dev, (void *)buffer, 120264);
        ESP_LOGI(TAG, "Write %d bytes to codec!", was_write_length);
        esp_codec_dev_close(codec_dev);
    }`

**Expected Test Result:** The presence or absence of the i2s_channel_reconfig_std_gpio API should not affect the normal audio output.
**Actual Test Result:** When the i2s_channel_reconfig_std_gpio API is uncommented, the audio output becomes abnormal, indicating a potential issue with the API's functionality，and the API does not report any errors.
**Note:** I have not found any unit tests related to the I2S API, so I am unable to perform additional comparative verification. However, I can confirm that the API call itself is implemented correctly.
**Normal and Abnormal Test Logs:**

> D (1677) i2s_common: tx channel is registered on I2S1 successfully
D (1677) i2s_common: rx channel is registered on I2S1 successfully
D (1677) i2s_common: DMA malloc info: dma_desc_num = 6, dma_desc_buf_size = dma_frame_num * slot_num * data_bit_width = 480
D (1687) i2s_std: Clock division info: [sclk] 160000000 Hz [mdiv] 14 [mclk] 11289600 Hz [bdiv] 8 [bclk] 1411200 Hz
I (1697) i2s_std: i2s_std_set_gpio success
D (1707) i2s_std: The tx channel on I2S1 has been initialized to STD mode successfully
D (1717) i2s_common: i2s tx channel enabled
D (1717) i2s_common: DMA malloc info: dma_desc_num = 6, dma_desc_buf_size = dma_frame_num * slot_num * data_bit_width = 480
D (1727) i2s_std: Clock division info: [sclk] 160000000 Hz [mdiv] 14 [mclk] 11289600 Hz [bdiv] 8 [bclk] 1411200 Hz
I (1737) i2s_std: i2s_std_set_gpio success
D (1737) i2s_std: The rx channel on I2S1 has been initialized to STD mode successfully
I (1747) M5Stack: I2S initialized successfully
I (1757) M5Stack: I2S config: sample_rate=44100, data_bit_width=16, slot_mode=1
I (1757) M5Stack: Set REG 0x94 to 0xf4
I (1507) ili9341: LCD panel create success, version: 1.2.0
I (1647) gpio: GPIO[39]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:2
D (1677) i2s_common: tx channel is registered on I2S1 successfully
D (1677) i2s_common: rx channel is registered on I2S1 successfully
D (1677) i2s_common: DMA malloc info: dma_desc_num = 6, dma_desc_buf_size = dma_frame_num * slot_num * data_bit_width = 480
D (1687) i2s_std: Clock division info: [sclk] 160000000 Hz [mdiv] 14 [mclk] 11289600 Hz [bdiv] 8 [bclk] 1411200 Hz
I (1697) i2s_std: i2s_std_set_gpio success
D (1707) i2s_std: The tx channel on I2S1 has been initialized to STD mode successfully
D (1717) i2s_common: i2s tx channel enabled
D (1717) i2s_common: DMA malloc info: dma_desc_num = 6, dma_desc_buf_size = dma_frame_num * slot_num * data_bit_width = 480
D (1727) i2s_std: Clock division info: [sclk] 160000000 Hz [mdiv] 14 [mclk] 11289600 Hz [bdiv] 8 [bclk] 1411200 Hz
I (1737) i2s_std: i2s_std_set_gpio success
D (1737) i2s_std: The rx channel on I2S1 has been initialized to STD mode successfully
I (1747) M5Stack: I2S initialized successfully
I (1757) M5Stack: I2S config: sample_rate=44100, data_bit_width=16, slot_mode=1
I (1757) M5Stack: Set REG 0x94 to 0xf4
I (1647) gpio: GPIO[39]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:2
D (1677) i2s_common: tx channel is registered on I2S1 successfully
D (1677) i2s_common: rx channel is registered on I2S1 successfully
D (1677) i2s_common: DMA malloc info: dma_desc_num = 6, dma_desc_buf_size = dma_frame_num * slot_num * data_bit_width = 480
D (1687) i2s_std: Clock division info: [sclk] 160000000 Hz [mdiv] 14 [mclk] 11289600 Hz [bdiv] 8 [bclk] 1411200 Hz
I (1697) i2s_std: i2s_std_set_gpio success
D (1707) i2s_std: The tx channel on I2S1 has been initialized to STD mode successfully
D (1717) i2s_common: i2s tx channel enabled
D (1717) i2s_common: DMA malloc info: dma_desc_num = 6, dma_desc_buf_size = dma_frame_num * slot_num * data_bit_width = 480
D (1727) i2s_std: Clock division info: [sclk] 160000000 Hz [mdiv] 14 [mclk] 11289600 Hz [bdiv] 8 [bclk] 1411200 Hz
I (1737) i2s_std: i2s_std_set_gpio success
D (1737) i2s_std: The rx channel on I2S1 has been initialized to STD mode successfully
I (1747) M5Stack: I2S initialized successfully
I (1757) M5Stack: I2S config: sample_rate=44100, data_bit_width=16, slot_mode=1
I (1757) M5Stack: Set REG 0x94 to 0xf4
I (1697) i2s_std: i2s_std_set_gpio success
D (1707) i2s_std: The tx channel on I2S1 has been initialized to STD mode successfully
D (1717) i2s_common: i2s tx channel enabled
D (1717) i2s_common: DMA malloc info: dma_desc_num = 6, dma_desc_buf_size = dma_frame_num * slot_num * data_bit_width = 480
D (1727) i2s_std: Clock division info: [sclk] 160000000 Hz [mdiv] 14 [mclk] 11289600 Hz [bdiv] 8 [bclk] 1411200 Hz
I (1737) i2s_std: i2s_std_set_gpio success
D (1737) i2s_std: The rx channel on I2S1 has been initialized to STD mode successfully
I (1747) M5Stack: I2S initialized successfully
I (1757) M5Stack: I2S config: sample_rate=44100, data_bit_width=16, slot_mode=1
I (1757) M5Stack: Set REG 0x94 to 0xf4
D (1727) i2s_std: Clock division info: [sclk] 160000000 Hz [mdiv] 14 [mclk] 11289600 Hz [bdiv] 8 [bclk] 1411200 Hz
I (1737) i2s_std: i2s_std_set_gpio success
D (1737) i2s_std: The rx channel on I2S1 has been initialized to STD mode successfully
I (1747) M5Stack: I2S initialized successfully
I (1757) M5Stack: I2S config: sample_rate=44100, data_bit_width=16, slot_mode=1
I (1757) M5Stack: Set REG 0x94 to 0xf4
I (1757) M5Stack: I2S config: sample_rate=44100, data_bit_width=16, slot_mode=1
I (1757) M5Stack: Set REG 0x94 to 0xf4
D (1767) i2s_common: i2s tx channel disabled
E (1767) i2s_common: i2s_channel_disable(1187): the channel has not been enabled yet
I (1777) I2S_IF: channel mode 0 bits:16/16 channel:2 mask:1
D (1777) i2s_std: Clock division info: [sclk] 160000000 Hz [mdiv] 14 [mclk] 11289600 Hz [bdiv] 8 [bclk] 1411200 Hz
I (1787) I2S_IF: STD Mode 1 bits:16/16 channel:2 sample_rate:44100 mask:1
D (1797) i2s_common: i2s tx channel enabled
I (1797) Adev_Codec: Open codec device OK
I (3227) ui_main: Write 0 bytes to codec!
D (3227) i2s_common: i2s tx channel disabled
I (3227) main_task: Returned from app_main()

L-KAYA, 2025-04-03:
Hi @illusionkiller , thanks for reporting! I'll look into this issue. Could you describe the application scenario? When we have to reconfig the GPIO?

illusionkiller, 2025-05-13:
> Hi [@illusionkiller](https://github.com/illusionkiller) , thanks for reporting! I'll look into this issue. Could you describe the application scenario? When we have to reconfig the GPIO?
Thank you for your attention to this issue.

application scenario:

[image]

I am using the M5 Stack Core2 development board for audio development. There is a conflict in the audio interface of this board, requiring IO reconfiguration when switching between output and input modes. However, the esp_codev_dev abstraction layer does not provide APIs for this purpose, so I resorted to using lower-level I2S APIs. To describe the issue more clearly, I simplified the faulty code.

Resolution:
I have located the unit test code for the underlying I2S APIs. When tested independently without the esp_codev_dev abstraction layer, the I2S APIs work correctly. For example, the following code enables output and input without issues:

```
void Core2AudioCodec::EnableInput(bool enable) {
    i2s_pdm_rx_gpio_config_t pdm_rx_gpio_cfg = {
        .clk = I2S_GPIO_UNUSED,
        .din = I2S_GPIO_UNUSED,
    };
    if (enable == input_enabled_) {
        return;
    }
    if (enable) {
        // esp_codec_dev_sample_info_t fs = {
        //     .bits_per_sample = 16,
        //     .channel = 1, //must be 1
        //     .sample_rate = (uint32_t)input_sample_rate_,
        // };
        pdm_rx_gpio_cfg.clk = mic_sck_;
        pdm_rx_gpio_cfg.din = mic_din_;
        ESP_LOGI(TAG, "EnableInput: sck=%d, din=%d", mic_sck_, mic_din_);
        ESP_ERROR_CHECK(i2s_channel_reconfig_pdm_rx_gpio(rx_handle_, &pdm_rx_gpio_cfg));
        ESP_ERROR_CHECK(i2s_channel_enable(rx_handle_));
        // ESP_ERROR_CHECK(esp_codec_dev_open(input_dev_, &fs));
        // esp_codec_dev_set_in_channel_gain(input_dev_, fs.channel_mask, 40.0);
    } else {
        // ESP_ERROR_CHECK(esp_codec_dev_close(input_dev_));
        ESP_ERROR_CHECK(i2s_channel_disable(rx_handle_));
        pdm_rx_gpio_cfg.clk = I2S_GPIO_UNUSED;
        pdm_rx_gpio_cfg.din = I2S_GPIO_UNUSED;
        ESP_ERROR_CHECK(i2s_channel_reconfig_pdm_rx_gpio(rx_handle_, &pdm_rx_gpio_cfg)); // Stop PDM RX avoid affect tx
    }
    AudioCodec::EnableInput(enable);
}

void Core2AudioCodec::EnableOutput(bool enable) {
    i2s_std_gpio_config_t std_gpio_cfg = {
        .mclk = I2S_GPIO_UNUSED, // some codecs may require mclk signal, this example doesn't need it
        .bclk = I2S_GPIO_UNUSED,
        .ws = I2S_GPIO_UNUSED,
        .dout = I2S_GPIO_UNUSED,
        .din = I2S_GPIO_UNUSED,
    };
    if (enable == output_enabled_) {
        return;
    }
    if (enable) {
        // Play 16bit 1 channel
        // esp_codec_dev_sample_info_t fs = {
        //     .bits_per_sample = 16,
        //     .channel = 1,
        //     // .channel_mask = 0,
        //     .sample_rate = output_sample_rate_,
        //     // .mclk_multiple = 0,
        // };
        std_gpio_cfg.bclk = spk_bclk_;
        std_gpio_cfg.ws = spk_ws_;
        std_gpio_cfg.dout = spk_dout_;
        ESP_LOGI(TAG, "EnableOutput: bclk=%d, ws=%d, dout=%d", spk_bclk_, spk_ws_, spk_dout_);
        ESP_ERROR_CHECK(i2s_channel_reconfig_std_gpio(tx_handle_, &std_gpio_cfg));
        ESP_ERROR_CHECK(i2s_channel_enable(tx_handle_));
        // ESP_ERROR_CHECK(esp_codec_dev_open(output_dev_, &fs));
        // ESP_ERROR_CHECK(esp_codec_dev_set_out_vol(output_dev_, output_volume_));
    } else {
        // ESP_ERROR_CHECK(esp_codec_dev_close(output_dev_));
        ESP_ERROR_CHECK(i2s_channel_disable(tx_handle_));
        std_gpio_cfg.bclk = I2S_GPIO_UNUSED;
        std_gpio_cfg.ws = I2S_GPIO_UNUSED;
        std_gpio_cfg.dout = I2S_GPIO_UNUSED;
        ESP_ERROR_CHECK(i2s_channel_reconfig_std_gpio(tx_handle_, &std_gpio_cfg));
    }
    AudioCodec::EnableOutput(enable);
}
```

L-KAYA, 2025-06-04:
@illusionkiller I might catch some clues. 

## Assumption

The audio was disturbed after GPIO reconfiguration since the clock conflict on that GPIO. 

## Details

For the initial case that using `audio_codec_dev` library, I guess you registered TX and RX channel separately, which make RX and TX channel both working in simplex mode and master role. Which is to say, they will output their own clock to the GPIO, when their clock signals are drew to a same GPIO, two output signal conflict, which make the audio abnormal.

For the second case that using IDF I2S driver, although TX and RX channels are still separate, they can be disabled separately. No signal conflicts will happen because only one channel is outputting and another one is disabled.

## Solution

Register `esp_codec_dev_handle_t` separately for TX and RX channel, so that they can be controlled separately. Disable another one first if you'd like to re-configure the GPIO.

Hope it can help~

illusionkiller, 2025-06-05:
> [@illusionkiller](https://github.com/illusionkiller) I might catch some clues.
>
> ## Assumption
> The audio was disturbed after GPIO reconfiguration since the clock conflict on that GPIO.
>
> ## Details [...]

Thank you for your reply 
You can see that in my code snippet, the commented out part demonstrates the use of the esp_comdec_dev abstraction layer to register TX (export_dex_) and RX (input_dex_) channels separately. When using one of them, the other is disabled, but it still doesn't work in this situation
At present, I am not using the esp_comdec_dev abstraction layer. Instead, I am using the I2S driver API, which works very well
Of course, it's possible that I used the esp_comdec_dev component improperly, but I don't have any examples to refer to

L-KAYA, 2025-06-05:
@illusionkiller Thanks for the feedback! 

If the I2S driver in IDF works well, maybe you can rise an issue in [esp-adf](https://github.com/espressif/esp-adf/tree/master/components/esp_codec_dev) instead. 

This issue will be closed, please feel free to re-open it if you have any further questions related~