espressif/esp-idf #15462: Visual Studio Code Debugger

donaldkohcw, 2025-02-25:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi, 

The debugger Step Over, Step In, and Step Out options are greyed out. Is there a setting to re-enable them? Thanks.

I attached setting.json, launch.json and c_app_properties.json

[c_cpp_properties.json](https://github.com/user-attachments/files/18953254/c_cpp_properties.json)
[launch.json](https://github.com/user-attachments/files/18953255/launch.json)
[settings.json](https://github.com/user-attachments/files/18953253/settings.json)

ESP-IDF v5.3.2 on esp32h2

[image]

Thanks,

brianignacio5, 2025-02-25:
Please share the `ESP-IDF: Doctor Command` output.

Which launch configuration are you using (from launch.json there are 2) ? Do you have a ESP-IDF project opened? What is the output of the debug console and ESP-IDF Output tab ?

donaldkohcw, 2025-02-25:
[report.txt](https://github.com/user-attachments/files/18953759/report.txt)

Thanks for pointing it out. It seems that after I reinitialized the "add VS Code configuration folder," I managed to get it running again.

I have also attached the report.txt. Let me know if there are any settings that I need to configure.

Thanks!