espressif/esp-idf #15463: esp_bredr_tx_power_set() doesnt change all tx power when phone connecter esp32 with classic bluetooth???

jerryc20, 2025-02-25:
[https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/classic_bt/bt_hid_mouse_device](url)

In this sample, I try to set the bluetooth tx power lowest after phone has connected, so that I can disconnect quickly when get away from esp32.

I use esp_bt_controller_disable() and esp_bt_controller_enable(ESP_BT_MODE_CLASSIC_BT) after esp_bredr_tx_power_set(), it looks like that another phone can't discover the esp32 in 3 meters away, but in the same distance, the phone that has connected would not disconnect, the power is still strong. 

[https://docs.espressif.com/projects/esp-idf/zh_CN/stable/esp32/api-reference/bluetooth/controller_vhci.html](url) notes that "The power level will affect the global BR/EDR TX power for operations such as inquiry, page, and connection." 

Could Someone take a try? I dont know if there any wrong with me or bug?

THX!

jcbetter, 2025-02-25:
@jerryc20 

Whrn the example conneted to any device, it will be set to non-discoverable. So it will not be discovered by any other devices.
```
    case ESP_HIDD_OPEN_EVT:
        if (param->open.status == ESP_HIDD_SUCCESS) {
            if (param->open.conn_status == ESP_HIDD_CONN_STATE_CONNECTING) {
                ESP_LOGI(TAG, "connecting...");
            } else if (param->open.conn_status == ESP_HIDD_CONN_STATE_CONNECTED) {
                ESP_LOGI(TAG, "connected to %02x:%02x:%02x:%02x:%02x:%02x", param->open.bd_addr[0],
                         param->open.bd_addr[1], param->open.bd_addr[2], param->open.bd_addr[3], param->open.bd_addr[4],
                         param->open.bd_addr[5]);
                bt_app_task_start_up();
                ESP_LOGI(TAG, "making self non-discoverable and non-connectable.");
                esp_bt_gap_set_scan_mode(ESP_BT_NON_CONNECTABLE, ESP_BT_NON_DISCOVERABLE);   // <=============
            } else {
                ESP_LOGE(TAG, "unknown connection status");
            }
        } else {
            ESP_LOGE(TAG, "open failed!");
        }
        break;
```

If you disable and re-enable the controller, it does not make sense if the connection doesn't disconnect. I just tested it. When I call disable&enable after the device is connected, the connection between esp32 and the mobile phone will be disconnected.

I made the following modifications to the code:

```
    case ESP_HIDD_OPEN_EVT:
        if (param->open.status == ESP_HIDD_SUCCESS) {
            if (param->open.conn_status == ESP_HIDD_CONN_STATE_CONNECTING) {
                ESP_LOGI(TAG, "connecting...");
            } else if (param->open.conn_status == ESP_HIDD_CONN_STATE_CONNECTED) {
                ESP_LOGI(TAG, "connected to %02x:%02x:%02x:%02x:%02x:%02x", param->open.bd_addr[0],
                         param->open.bd_addr[1], param->open.bd_addr[2], param->open.bd_addr[3], param->open.bd_addr[4],
                         param->open.bd_addr[5]);
                bt_app_task_start_up();
                ESP_LOGI(TAG, "making self non-discoverable and non-connectable.");
                esp_bt_gap_set_scan_mode(ESP_BT_NON_CONNECTABLE, ESP_BT_NON_DISCOVERABLE);
                ///////////////////////////
                esp_bredr_tx_power_set(ESP_PWR_LVL_N12, ESP_PWR_LVL_N12);
                esp_err_t err = esp_bt_controller_disable();
                printf("err=%d\n", err);
                err = esp_bt_controller_enable(ESP_BT_MODE_CLASSIC_BT);
                printf("err=%d\n", err);
                ///////////////////////////
            } else {
                ESP_LOGE(TAG, "unknown connection status");
            }
        } else {
            ESP_LOGE(TAG, "open failed!");
        }
        break;
```

jerryc20, 2025-03-12:
> [@jerryc20](https://github.com/jerryc20)
>
> Whrn the example conneted to any device, it will be set to non-discoverable. So it will not be discovered by any other devices.
>
> ```
>     case [...]

thx！

closed 2025-03-12 by jerryc20