espressif/esp-idf #15466: ESP32C3 Timer definition

graeme44, 2025-02-25:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

According to the documentation of the ESP32C3, it should have 4 timers: 

> [The two ESP32-C3 timer groups, with two timers in each, provide the total of four individual timers for use.](https://docs.espressif.com/projects/esp-idf/en/v4.3/esp32c3/api-reference/peripherals/timer.html)

The current definition in `soc_caps.h` defines the 2 timer groups, but only 1 timer per group.

https://github.com/espressif/esp-idf/blob/877057db3d70019d064b593e68a174c8b218cbc6/components/soc/esp32c3/include/soc/soc_caps.h#L364-L370

suda-morris, 2025-02-26:
Hi @graeme44 the code is correct, but the programming guide you read is old (v4.3) and contains some wrong information, sorry about that.

This is the TRM of ESP32C3: https://www.espressif.com/sites/default/files/documentation/esp32-c3_technical_reference_manual_en.pdf#timg

Where is explained how the timers are grouped.

[image]

graeme44, 2025-02-26:
> Hi [@graeme44](https://github.com/graeme44) the code is correct, but the programming guide you read is old (v4.3) and contains some wrong information, sorry about that.
>
> This is the TRM of ESP32C3: [...]

Great, thank you.

closed 2025-02-26 by graeme44