espressif/esp-idf #15467: building issues in VSCode on Linux

ravencarcass, 2025-02-25:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

V5.4

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I would build an example and I would then get the full build as it normally is. After building, I would also like to see the table of memory partitions as usual. 

### What is the actual behavior?

I want to build an example like lights for Matter. I hit the build button and I get the following build problems.
It goes into a constant loop until I cancel the build.  Yesterday it did build and now it doesn't, I find it a bit weird. 

### Steps to reproduce.

1. open the command prompt of Linux
2. write 'cd esp'
3. write 'get_matter'
4. write 'export IDF_CCACHE_ENABLE=1'
5. write 'code' 
6. the example pop automatically open and wait until I see the building logo
7. I press on the building logo and then begins the loop problem.

### Build or installation Logs.

```plain
*  Executing task: ninja  

[0/1] Re-running CMake...
IDF_VERSION: 5.4.0; idf version number: 5.4.0
Sequence of SDKCONFIG files: sdkconfig.defaults
-- Component directory /home/robbe/esp/esp-matter/examples/common/blemesh_platform does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/robbe/esp/esp-matter/examples/common/cmake_common does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/robbe/esp/esp-matter/examples/common/external_platform does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/robbe/esp/esp-matter/examples/common/relinker does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/robbe/esp/esp-matter/examples/common/secondary_ble_adv does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/robbe/esp/esp-matter/examples/common/utils does not contain a CMakeLists.txt file. No component will be added
-- ccache will be used for faster recompilation
-- Building ESP-IDF components for target esp32c6
NOTICE: Skipping optional dependency: espressif/cmake_utilities
NOTICE: Skipping optional dependency: espressif/esp_wifi_remote
NOTICE: Skipping optional dependency: espressif/cmake_utilities
NOTICE: Skipping optional dependency: espressif/esp_wifi_remote
NOTICE: Skipping optional dependency: espressif/cmake_utilities
NOTICE: Skipping optional dependency: espressif/esp_wifi_remote
NOTICE: Skipping optional dependency: espressif/cmake_utilities
NOTICE: Skipping optional dependency: espressif/esp_wifi_remote
NOTICE: Processing 20 dependencies:
NOTICE: [1/20] espressif/button (3.5.0)
NOTICE: [2/20] espressif/cbor (0.6.0~1)
NOTICE: [3/20] espressif/cmake_utilities (0.5.3)
NOTICE: [4/20] espressif/esp-serial-flasher (0.0.11)
[... 74 lines cut ...]
-- Adding linker script /home/robbe/esp/esp-matter/examples/light/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script /home/robbe/esp/esp-matter/examples/light/build/esp-idf/esp_system/ld/sections.ld.in
-- Adding linker script /home/robbe/esp/esp-idf/components/esp_rom/esp32c6/ld/esp32c6.rom.ld
-- Adding linker script /home/robbe/esp/esp-idf/components/esp_rom/esp32c6/ld/esp32c6.rom.api.ld
-- Adding linker script /home/robbe/esp/esp-idf/components/esp_rom/esp32c6/ld/esp32c6.rom.rvfp.ld
-- Adding linker script /home/robbe/esp/esp-idf/components/esp_rom/esp32c6/ld/esp32c6.rom.wdt.ld
-- Adding linker script /home/robbe/esp/esp-idf/components/esp_rom/esp32c6/ld/esp32c6.rom.systimer.ld
-- Adding linker script /home/robbe/esp/esp-idf/components/esp_rom/esp32c6/ld/esp32c6.rom.version.ld
-- Adding linker script /home/robbe/esp/esp-idf/components/esp_rom/esp32c6/ld/esp32c6.rom.phy.ld
-- Adding linker script /home/robbe/esp/esp-idf/components/esp_rom/esp32c6/ld/esp32c6.rom.coexist.ld
-- Adding linker script /home/robbe/esp/esp-idf/components/esp_rom/esp32c6/ld/esp32c6.rom.net80211.ld
-- Adding linker script /home/robbe/esp/esp-idf/components/esp_rom/esp32c6/ld/esp32c6.rom.pp.ld
-- Adding linker script /home/robbe/esp/esp-idf/components/esp_rom/esp32c6/ld/esp32c6.rom.newlib.ld
-- Adding linker script /home/robbe/esp/esp-idf/components/esp_rom/esp32c6/ld/esp32c6.rom.newlib-normal.ld
-- Adding linker script /home/robbe/esp/esp-idf/components/esp_rom/esp32c6/ld/esp32c6.rom.heap.ld
-- Adding linker script /home/robbe/esp/esp-idf/components/soc/esp32c6/ld/esp32c6.peripherals.ld
-- BUTTON: 3.5.0
-- LED_INDICATOR: 0.9.3
ESP Insights Project commit: e2a83f8
-- Components: app_bridge app_reset app_trace app_update bootloader bootloader_support bt chip cmock console cxx driver efuse esp-tls esp_adc esp_app_format esp_bootloader_format esp_coex esp_common esp_driver_ana_cmpr esp_driver_cam esp_driver_dac esp_driver_gpio esp_driver_gptimer esp_driver_i2c esp_driver_i2s esp_driver_isp esp_driver_jpeg esp_driver_ledc esp_driver_mcpwm esp_driver_parlio esp_driver_pcnt esp_driver_ppa esp_driver_rmt esp_driver_sdio esp_driver_sdm esp_driver_sdmmc esp_driver_sdspi esp_driver_spi esp_driver_touch_sens esp_driver_tsens esp_driver_uart esp_driver_usb_serial_jtag esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_lcd esp_local_ctrl esp_matter esp_matter_bridge esp_matter_console esp_matter_controller esp_matter_ota_provider esp_matter_rainmaker esp_mm esp_netif esp_netif_stack esp_partition esp_phy esp_pm esp_psram esp_ringbuf esp_rom esp_security esp_system esp_timer esp_vfs_console esp_wifi espcoredump espressif__button espressif__cbor espressif__cmake_utilities espressif__esp-serial-flasher espressif__esp_bsp_devkit espressif__esp_delta_ota espressif__esp_diag_data_store espressif__esp_diagnostics espressif__esp_encrypted_img espressif__esp_insights espressif__esp_rcp_update espressif__esp_secure_cert_mgr espressif__jsmn espressif__json_generator espressif__json_parser espressif__led_indicator espressif__led_strip espressif__mdns espressif__rmaker_common esptool_py fatfs freertos hal heap http_parser idf_test ieee802154 json log lwip main mbedtls mqtt newlib nvs_flash nvs_sec_provider openthread partition_table protobuf-c protocomm pthread riscv rt sdmmc soc spi_flash spiffs tcp_transport ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant
-- Component paths: /home/robbe/esp/esp-matter/examples/common/app_bridge /home/robbe/esp/esp-matter/examples/common/app_reset /home/robbe/esp/esp-idf/components/app_trace /home/robbe/esp/esp-idf/components/app_update /home/robbe/esp/esp-idf/components/bootloader /home/robbe/esp/esp-idf/components/bootloader_support /home/robbe/esp/esp-idf/components/bt /home/robbe/esp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/components/chip /home/robbe/esp/esp-idf/components/cmock /home/robbe/esp/esp-idf/components/console /home/robbe/esp/esp-idf/components/cxx /home/robbe/esp/esp-idf/components/driver /home/robbe/esp/esp-idf/components/efuse /home/robbe/esp/esp-idf/components/esp-tls /home/robbe/esp/esp-idf/components/esp_adc /home/robbe/esp/esp-idf/components/esp_app_format /home/robbe/esp/esp-idf/components/esp_bootloader_format /home/robbe/esp/esp-idf/components/esp_coex /home/robbe/esp/esp-idf/components/esp_common /home/robbe/esp/esp-idf/components/esp_driver_ana_cmpr /home/robbe/esp/esp-idf/components/esp_driver_cam /home/robbe/esp/esp-idf/components/esp_driver_dac /home/robbe/esp/esp-idf/components/esp_driver_gpio /home/robbe/esp/esp-idf/components/esp_driver_gptimer /home/robbe/esp/esp-idf/components/esp_driver_i2c /home/robbe/esp/esp-idf/components/esp_driver_i2s /home/robbe/esp/esp-idf/components/esp_driver_isp /home/robbe/esp/esp-idf/components/esp_driver_jpeg /home/robbe/esp/esp-idf/components/esp_driver_ledc /home/robbe/esp/esp-idf/components/esp_driver_mcpwm /home/robbe/esp/esp-idf/components/esp_driver_parlio /home/robbe/esp/esp-idf/components/esp_driver_pcnt /home/robbe/esp/esp-idf/components/esp_driver_ppa /home/robbe/esp/esp-idf/components/esp_driver_rmt /home/robbe/esp/esp-idf/components/esp_driver_sdio /home/robbe/esp [... cut ...] nents/spiffs /home/robbe/esp/esp-idf/components/tcp_transport /home/robbe/esp/esp-idf/components/ulp /home/robbe/esp/esp-idf/components/unity /home/robbe/esp/esp-idf/components/usb /home/robbe/esp/esp-idf/components/vfs /home/robbe/esp/esp-idf/components/wear_levelling /home/robbe/esp/esp-idf/components/wifi_provisioning /home/robbe/esp/esp-idf/components/wpa_supplicant
-- Configuring done
-- Generating done
-- Build files have been written to: /home/robbe/esp/esp-matter/examples/light/build
```

ravencarcass, 2025-02-26:
Problem solved with updating the submodule. I dit it with "git submodule update --init --recursive".

Alvin1Zhang, 2025-02-26:
Thanks for reporting and sharing the updates, good to hear that the issue was fixed, feel free to reopen if the issue still happens, thanks.

closed 2025-02-26 by Alvin1Zhang

ravencarcass, 2025-02-26:
The issue still happend. But only with the light example