espressif/esp-idf #15469: esp_wifi_deinit can't release the ram of esp_wifi_init

Wangmale, 2025-02-26:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

4.4.4

### Espressif SoC revision.

esp32

### Operating System used.

Linux

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

custome board

### Power Supply used.

USB

### What is the expected behavior?

E (09:36:00.856) cnt-wifi: Wifi init start Free memory: 136727 bytes 
I (29998) wifi:wifi driver task: 3ffbb3a0, prio:23, stack:6656, core=0
I (30023) wifi:wifi firmware version: 6567a16
I (30023) wifi:wifi certification version: v7.0
I (30023) wifi:config NVS flash: enabled
I (30025) wifi:config nano formating: disabled
I (30029) wifi:Init data frame dynamic rx buffer num: 8
I (30034) wifi:Init management frame dynamic rx buffer num: 8
I (30039) wifi:Init management short buffer num: 32
I (30045) wifi:Init static tx buffer num: 16
I (30049) wifi:Init tx cache buffer num: 32
I (30052) wifi:Init static rx buffer size: 1600
I (30056) wifi:Init static rx buffer num: 4
I (30060) wifi:Init dynamic rx buffer num: 8
I (09:36:00.934) wifi_init: tcpip mbox: 32
I (09:36:00.938) wifi_init: udp mbox: 6
I (09:36:00.943) wifi_init: tcp mbox: 6
I (09:36:00.947) wifi_init: tcp tx win: 5760
I (09:36:00.952) wifi_init: tcp rx win: 5760
I (09:36:00.957) wifi_init: tcp mss: 1440
I (09:36:00.962) wifi_init: WiFi/LWIP prefer SPIRAM
E (09:36:00.986) cnt-wifi: Wifi init end Free memory: 97911 bytes 
esp_wifi_inti(&cfg) use 40k ram.

i use esp_wifi_deinit() can't release this 40k ram

E (09:36:35.699) cnt-mode: Cfg normal start Free memory: 85995 bytes 
W (09:36:35.810) cnt-web: ----------------web stop
I (64942) wifi:flush txq
I (64942) wifi:stop sw txq
I (64943) wifi:lmac stop hw txq
I (64944) wifi:Deinit lldesc rx mblock:4
I (09:36:35.818) cnt-mode: chint_mode_normal
E (09:36:35.821) cnt-mode: Cfg normal end Free memory: 85995 bytes 

### What is the actual behavior?

.

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### Debug Logs.

```plain

```

Wangmale, 2025-02-27:
i use **esp32s3 kit board** and **espidf example** to test this **esp_wifi_init(), can't efficient**.

idf Vesrion: **4.4.4**
chip: **esp32s3**

i just modify the app_main(),source code:
~~~C

void app_main(void)
{
    // Initialize NVS
    esp_err_t ret = nvs_flash_init();
    if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND) {
        ESP_ERROR_CHECK(nvs_flash_erase());
        ret = nvs_flash_init();
    }
    ESP_ERROR_CHECK(ret);

    ESP_LOGI(TAG, "ESP_WIFI_MODE_STA");
    wifi_init_sta();
    vTaskDelay(pdTICKS_TO_MS(10000));
    printf("Before wifi stop Free memory: on chip = %.2f Kbytes,total = %.2f Kbytes %s\n", heap_caps_get_minimum_free_size(MALLOC_CAP_8BIT | MALLOC_CAP_DMA) / 1024.0, esp_get_free_heap_size() / 1024.0, __FILE__);
    esp_wifi_stop();
    printf("Before wifi deinit Free memory: on chip = %.2f Kbytes,total = %.2f Kbytes %s\n", heap_caps_get_minimum_free_size(MALLOC_CAP_8BIT | MALLOC_CAP_DMA) / 1024.0, esp_get_free_heap_size() / 1024.0, __FILE__);
    esp_wifi_deinit();
    printf("After wifi deinit Free memory: on chip = %.2f Kbytes,total = %.2f Kbytes %s\n", heap_caps_get_minimum_free_size(MALLOC_CAP_8BIT | MALLOC_CAP_DMA) / 1024.0, esp_get_free_heap_size() / 1024.0, __FILE__);

    while (1) {
        printf("Loop (1s) print Free memory: on chip = %.2f Kbytes,total = %.2f Kbytes %s\n", heap_caps_get_minimum_free_size(MALLOC_CAP_8BIT | MALLOC_CAP_DMA) / 1024.0, esp_get_free_heap_size() / 1024.0, __FILE__);
        vTaskDelay(pdTICKS_TO_MS(1000));
    }
}

~~~

log:
~~~ bash

I (24) boot: ESP-IDF v4.4.4 2nd stage bootloader
I (25) boot: compile time 01:24:32
I (25) boot: chip revision: v0.1
I (26) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (34) boot.esp32s3: Boot SPI Speed : 80MHz
I (38) boot.esp32s3: SPI Mode       : DIO
I (43) boot.esp32s3: SPI Flash Size : 2MB
I (48) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (72) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (79) boot:  2 factory          factory app      00 00 00010000 00100000
I (87) boot: End of partition table
I (91) boot_comm: chip revision: 1, min. application chip revision: 0
I (98) esp_image: segment 0: paddr=00010020 vaddr=3c070020 size=12fb8h ( 77752) map
I (120) esp_image: segment 1: paddr=00022fe0 vaddr=3fc962f0 size=043c8h ( 17352) load
I (124) esp_image: segment 2: paddr=000273b0 vaddr=40374000 size=08c68h ( 35944) load
I (134) esp_image: segment 3: paddr=00030020 vaddr=42000020 size=6b3b8h (439224) map
I (213) esp_image: segment 4: paddr=0009b3e0 vaddr=4037cc68 size=09684h ( 38532) load
I (229) boot: Loaded app from partition at offset 0x10000
I (229) boot: Disabling RNG early entropy source...
I (241) cpu_start: Pro cpu up.
I (241) cpu_start: Starting app cpu, entry point is 0x40375288
0x40375288: call_start_cpu1 at /opt/esp/idf/components/esp_system/port/cpu_start.c:148

I (0) cpu_start: App cpu up.
I (255) cpu_start: Pro cpu start user code
I (255) cpu_start: cpu freq: 160000000
I (255) cpu_start: Application information:
I (258) cpu_start: Project name:     wifi_station
I (263) cpu_start: App version:      1
[... 51 lines cut ...]
I (2131) wifi:state: auth -> assoc (0)
I (2141) wifi:state: assoc -> run (10)
I (2171) wifi:connected with TEST, aid = 4, channel 5, BW20, bssid = 02:52:67:c0:c1:c2
I (2171) wifi:security: WPA2-PSK, phy: bgn, rssi: -28
I (2181) wifi:pm start, type: 1

I (2181) wifi:set rx beacon pti, rx_bcn_pti: 0, bcn_timeout: 0, mt_pti: 25000, mt_time: 10000
I (2261) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (3181) esp_netif_handlers: sta ip: 192.168.8.114, mask: 255.255.255.0, gw: 192.168.8.1
I (3181) wifi station: got ip:192.168.8.114
I (3181) wifi station: connected to ap SSID:TEST password:12345678
Before wifi stop Free memory: on chip = 276.16 Kbytes,total = 296.77 Kbytes ./main/station_example_main.c
I (1003191) wifi:state: run -> init (0)
I (1003191) wifi:pm stop, total sleep time: 908998467 us / 1001004990 us

I (1003191) wifi:new:<5,0>, old:<5,0>, ap:<255,255>, sta:<5,0>, prof:1
I (1003721) wifi:flush txq
I (1003721) wifi:stop sw txq
I (1003721) wifi:lmac stop hw txq
Before wifi deinit Free memory: on chip = 276.16 Kbytes,total = 298.97 Kbytes ./main/station_example_main.c
I (1003721) wifi:Deinit lldesc rx mblock:10
After wifi deinit Free memory: on chip = 276.16 Kbytes,total = 328.87 Kbytes ./main/station_example_main.c
Loop (10s) print Free memory: on chip = 276.16 Kbytes,total = 328.87 Kbytes ./main/station_example_main.c
Loop (10s) print Free memory: on chip = 276.16 Kbytes,total = 328.91 Kbytes ./main/station_example_main.c
~~~

hansw123, 2025-03-31:
from the log we can see the free heap size is 330.25 Kbytes  and deinit wifi  is 328.87 Kbytes ,   basically consistent.
the 2k different may be nvs_flash or other components cost, so it may be ok

closed 2025-04-17 by Alvin1Zhang