espressif/esp-idf #15475: ESP32-C6 Sometimes startup fail

llisnikowski, 2025-02-26:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4

### Espressif SoC revision.

ESP32-C6

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom board

### Power Supply used.

External 3.3V

### What is the expected behavior?

Always startup.

### What is the actual behavior?

The program sametimes locks up before the `void app_main(void)` function.

The program stop at: `LPPERI.clk_en.rng_ck_en = en;` 
step in `esp/idf/components/hal/esp32c6/include/hal/lp_clkrst_ll.h:24`
in `_lp_clkrst_ll_enable_rng_clock` fuction:
```
__attribute__((always_inline))
static inline void _lp_clkrst_ll_enable_rng_clock(bool en)
{
    LPPERI.clk_en.rng_ck_en = en;
}
```
Backtrace:
- IDF/components/esp_system/port/cpu_start.c:714
- IDF/components/esp_system/port/soc/esp32c6/clk.c:319
- IDF/components/hal/esp32c6/include/hal/lp_clkrst_ll.h:24

### Steps to reproduce.

1. Build empty program
2. Flash
3. Reopen monitor a few times until the program locks up

### Debug Logs.

```plain
--- esp-idf-monitor 1.5.0 on /dev/ttyACM0 115200
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x1 (POWERON),boot:0x7c (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
Valid secure boot key blocks: 0 1 2
RSA-PSS secure boot verification succeeded
load:0x40875890,len:0x2678
load:0x4086c110,len:0xc30
load:0x4086e610,len:0x5014
entry 0x4086c110
I (118) cpu_start: Unicore app
D (118) cpu_start: Pro cpu up
D (126) clk: RTC_SLOW_CLK calibration value: 3824141
I (131) My log: /IDF/components/esp_system/port/cpu_start.c:714
I (132) My log: /IDF/components/esp_system/port/soc/esp32c6/clk.c:319
I (133) My log: /IDF/components/hal/esp32c6/include/hal/lp_clkrst_ll.h:24
```

### More Information.

On `v5.1.3` normal startup

ginkgm, 2025-03-21:
Hi @llisnikowski ,

I doubt whether it's related to your customized board or secure boot.

Have you tested if this issue happens without secure boot or on an official devkit?

Alvin1Zhang, 2025-03-27:
@llisnikowski Thanks for reporting, would you please help share if any updates for the issue? Thanks.

ginkgm, 2025-05-07:
Issue closed. Feel free to reopen if you have further clues.

closed 2025-05-07 by ginkgm