espressif/esp-idf #15480: ESP PoE2 Ethernet sporadically disconnects. -- Ethernet instability.

Decstar77, 2025-02-27:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4

### Espressif SoC revision.

ESP32-D0WD-V3 (revision v3.0)

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32-POE2

### Power Supply used.

USB

### What is the expected behavior?

Ethernet remains connected throughout the program.

### What is the actual behavior?

Disconnects at random points, reconnects after about 30 seconds.

### Steps to reproduce.

1. Start ethernet 
2. Wait some time ( 1/2 hours)
3. Disconnect happens ETHERNET_EVENT_DISCONNECTED
4. Reconnects.

### Debug Logs.

```plain
D (505265) esp.emac: receive len= 60
D (505755) esp.emac: receive len= 60
D (520015) esp.emac: receive len= 333
D (520015) esp.emac: receive len= 110
D (520025) esp.emac: receive len= 313
D (520025) esp.emac: receive len= 90
D (520325) esp.emac: receive len= 110
D (520325) esp.emac: receive len= 90
D (520545) esp.emac: receive len= 590
D (520625) esp.emac: receive len= 110
D (520635) esp.emac: receive len= 451
D (520635) esp.emac: receive len= 390
D (520635) esp.emac: receive len= 90
D (520645) esp.emac: receive len= 431
D (520645) esp.emac: receive len= 176
D (520655) esp.emac: receive len= 370
D (521075) esp.emac: receive len= 60
D (524625) esp.emac: receive len= 62
D (525475) esp.emac: receive len= 60
D (527695) esp.emac: receive len= 60
D (534545) esp.emac: receive len= 590
D (541545) esp.emac: receive len= 590
D (545545) esp.emac: receive len= 590
D (547545) esp.emac: receive len= 590
D (550545) esp.emac: receive len= 590
D (550575) esp.emac: emac stopped
D (550575) event: running post ETH_EVENT:3 with handler 0x400dfb20 and context 0x3ffc2450 on loop 0x3ffc1258
--- 0x400dfb20: eth_action_disconnected at C:/Espressif/frameworks/esp-idf-v5.4/components/esp_eth/src/esp_eth_netif_glue.c:109

D (550575) esp_eth.netif.netif_glue: eth_action_disconnected: 0x3ffc2368, 0x3f406f59, 3, 0x3ffd0050, 0x3ffbff60
D (550585) esp_netif_handlers: esp_netif action disconnected with netif0x3ffc2198 from event_id=3
D (550595) esp_netif_lwip: check: remote, if=0x3ffc2198 fn=0x400e9f58
--- 0x400e9f58: esp_netif_down_api at C:/Espressif/frameworks/esp-idf-v5.4/components/esp_netif/lwip/esp_netif_lwip.c:1752

D (550605) esp_netif_lwip: esp_netif_down_api esp_netif:0x3ffc2198
D (550615) esp_eth: Ethernet link is not up, can't transmit
D (550615) esp_eth: Ethernet link is not up, can't transmit
D (550625) esp_netif_lwip: esp_netif_internal_dhcpc_cb lwip-netif:0x3ffc2224
D (550625) esp_netif_lwip: esp_netif_start_ip_lost_timer esp_netif:0x3ffc2198
D (550635) esp_netif_lwip: if0x3ffc2198 start ip lost tmr: interval=120
D (550645) esp_netif_lwip: esp_netif_start_ip_lost_timer esp_netif:0x3ffc2198
D (550645) esp_netif_lwip: if0x3ffc2198 start ip lost tmr: already started
D (550655) esp_netif_lwip: check: local, if=0x3ffc2198 fn=0x400ea4b8
--- 0x400ea4b8: esp_netif_update_default_netif_lwip at C:/Espressif/frameworks/esp-idf-v5.4/components/esp_netif/lwip/esp_netif_lwip.c:313

D (550665) esp_netif_lwip: esp_netif_update_default_netif_lwip 0x3ffc2198
D (550665) esp_netif_lwip: call api in lwip: ret=0x0, give sem
D (550675) event: running post ETH_EVENT:3 with handler 0x400d8ce0 and context 0x3ffc24c8 on loop 0x3ffc1258
--- 0x400d8ce0: eth_event_handler at C:/Projects/2024/Eranest/SimplyFirmware/src/aqua_scanner_mqtt/main/aqua_scanner_poe.c:349

E (550675) transport_base: poll_read select error 113, errno = Software caused connection abort, fd = 54
I (550685) MAIN: Ethernet Link Down
E (550695) mqtt_client: Poll read error: 119, aborting connection
D (550705) mqtt_client: Reconnect after 10000 ms
D (550715) event: running post MQTT_EVENTS:2 with handler 0x400d8e8c and context 0x3ffc32dc on loop 0x3ffc314c
--- 0x400d8e8c: mqtt_event_handler at C:/Projects/2024/Eranest/SimplyFirmware/src/aqua_scanner_mqtt/main/aqua_scanner_poe.c:471

I (550725) MQTT: MQTT5_EVENT_DISCONNECTED
D (554575) esp.emac: working in 100Mbps
D (554575) esp.emac: working in full duplex
D (554575) esp.emac: Flow control not enabled for the link
D (554575) esp.emac: emac started
D (554585) event: running post ETH_EVENT:2 with handler 0x400dfb58 and context 0x3ffc2420 on loop 0x3ffc1258
--- 0x400dfb58: eth_action_connected at C:/Espressif/frameworks/esp-idf-v5.4/components/esp_eth/src/esp_eth_netif_glue.c:96

D (554595) esp_eth.netif.netif_glue: eth_action_connected: 0x3ffc2368, 0x3f406f59, 2, 0x3ffafc84, 0x3ffbff60
D (554605) esp_netif_handlers: esp_netif action connected with netif0x3ffc2198 from event_id=2
D (554615) esp_netif_lwip: check: remote, if=0x3ffc2198 fn=0x400e9f04
--- 0x400e9f04: esp_netif_up_api at C:/Espressif/frameworks/esp-idf-v5.4/components/esp_netif/lwip/esp_netif_lwip.c:1723

D (554615) esp_netif_lwip: esp_netif_up_api esp_netif:0x3ffc2198
D (554625) esp_netif_lwip: call api in lwip: ret=0x0, give sem
D (554625) esp_netif_lwip: check: remote, if=0x3ffc2198 fn=0x400e9784
--- 0x400e9784: esp_netif_dhcpc_start_api at C:/Espressif/frameworks/esp-idf-v5.4/components/esp_netif/lwip/esp_netif_lwip.c:1515

D (554635) esp_netif_lwip: esp_netif_dhcpc_start_api esp_netif:0x3ffc2198
D (554645) esp_netif_lwip: esp_netif_start_ip_lost_timer esp_netif:0x3ffc2198
D (554645) esp_netif_lwip: if0x3ffc2198 start ip lost tmr: already started
D (554655) esp_netif_lwip: starting dhcp client
D (554665) esp_netif_lwip: call api in lwip: ret=0x0, give sem
D (554665) event: running post ETH_EVENT:2 with handler 0x400d8ce0 and context 0x3ffc24c8 on loop 0x3ffc1258
--- 0x400d8ce0: eth_event_handler at C:/Projects/2024/Eranest/SimplyFirmware/src/aqua_scanner_mqtt/main/aqua_scanner_poe.c:349

I (554675) MAIN: Ethernet Link Up
I (554685) MAIN: Ethernet HW Addr 10:97:bd:e5:05:63
D (554735) esp.emac: receive len= 352
D (555035) esp.emac: receive len= 352
```

### More Information.

I'm using the new ESP PoE2 device and experiencing an issue where the Ethernet link randomly drops, triggering ETHERNET_EVENT_DISCONNECTED. I have the experimental flag CONFIG_ETH_RMII_CLK_OUTPUT_GPIO0 enabled. The debug logs show bursts of data, such as esp.emac: receive len= 590, before the disconnect occurs.
I'm also running an MQTT5 client, and I'm wondering if the Ethernet driver might be running out of memory, causing the connection to drop. I'm not sure how to proceed from here—any advice would be greatly appreciated. Thanks!

`static void eth_init() {
    eth_mac_config_t mac_config = ETH_MAC_DEFAULT_CONFIG();
    eth_phy_config_t phy_config = ETH_PHY_DEFAULT_CONFIG();
    phy_config.phy_addr = 0;
    phy_config.reset_gpio_num = 12;

    gpio_pad_select_gpio(12);
    gpio_set_direction(12,GPIO_MODE_OUTPUT);
    gpio_set_level(12, 1);
    vTaskDelay(pdMS_TO_TICKS(10));

    eth_esp32_emac_config_t esp32_emac_config = ETH_ESP32_EMAC_DEFAULT_CONFIG();
    esp32_emac_config.smi_gpio.mdc_num = 23;
    esp32_emac_config.smi_gpio.mdio_num = 18;

    esp_eth_mac_t * mac = esp_eth_mac_new_esp32( &esp32_emac_config, &mac_config );
    esp_eth_phy_t * phy = esp_eth_phy_new_lan87xx( &phy_config );

    esp_eth_handle_t eth_handle = NULL;
    esp_eth_config_t config = ETH_DEFAULT_CONFIG( mac, phy );
    esp_err_t res = esp_eth_driver_install( &config, &eth_handle );

    if ( res != ESP_OK ) {
        ESP_LOGE( ETH_TAG, "FAILED TO INSTALL DRIVER" );
        esp_restart();
    } else {
        ESP_LOGI( ETH_TAG, "Successfully installed driver" );
    }

    ESP_ERROR_CHECK( esp_netif_init() );
    ESP_ERROR_CHECK( esp_event_loop_create_default() );

    esp_netif_config_t cfg = ESP_NETIF_DEFAULT_ETH();
    esp_netif_t * eth_netif = esp_netif_new( &cfg );
    ESP_ERROR_CHECK( esp_netif_attach( eth_netif, esp_eth_new_netif_glue( eth_handle ) ) );

    ESP_ERROR_CHECK( esp_event_handler_register( ETH_EVENT, ESP_EVENT_ANY_ID, &eth_event_handler, NULL ) );
    ESP_ERROR_CHECK( esp_event_handler_register( IP_EVENT, IP_EVENT_ETH_GOT_IP, &eth_got_ip_event_handler, NULL ) );

    ESP_ERROR_CHECK( esp_eth_start( eth_handle ) );
}`

kostaond, 2025-02-28:
Hi @Decstar77, thanks for the issue report. If I understand it correctly, you use ESP32 to generate (output) REF RMII CLK at GPIO0, am I right? Do you use WiFi or Bluetooth simultaneously with the Ethernet? If so, it's unfortunately hardware problem in ESP32 SoC.
The RMII clock cannot be generated by the internal APLL clock when WiFi or BT used, as it would result in clock instability. See errat a for more information: https://docs.espressif.com/projects/esp-chip-errata/en/latest/esp32/03-errata-description/index.html#clock-esp32-cannot-be-used-as-the-phy-clock-source-if-wi-fi-and-ethernet-are-used-at-the-same-time

kostaond, 2025-03-28:
Closing due to lack of feedback. If you have additional questions, don't hesitate to ask.

closed 2025-03-28 by kostaond